espressif/arduino-esp32 #10129: Chip is unknown ESP32 (revision v1.0)  abort()

goddade, 2024-08-07:
### Board

ESP-WROOM-32D

### Device Description

ESP-WROOM-32D plain module

### Hardware Configuration

pull-up EN, TX & RX.

### Version

latest master (checkout manually)

### IDE Name

PlatformIO

### Operating System

Windows 10

### Flash frequency

40Mhz

### PSRAM enabled

no

### Upload speed

115200

### Description

Strange ESP-WROOM-32D module, the chip is Esp32-D0WD, but flash_id is `unknown ESP32 (revision v1.0)`

```
esptool -p com3 flash_id
esptool.py v4.7.0
Serial port com3
Connecting...
Failed to get PID of a device on com3, using standard reset sequence.
.
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting...
Failed to get PID of a device on com3, using standard reset sequence.

Detecting chip type... ESP32
Chip is unknown ESP32 (revision v1.0)
Features: WiFi, BT, Single Core, 240MHz, VRef calibration in efuse, Coding Scheme 3/4
Crystal is 40MHz
MAC: 54:48:e6:e4:0c:97
Uploading stub...
Running stub...
Stub running...
Manufacturer: 68
Device: 4016
Detected flash size: 4MB
Hard resetting via RTS pin...
```

One of the modules can run the esp32solo1 firmware, but the other cannot.

### Sketch

```cpp
platformio.ini:

[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino
monitor_filters = esp32_exception_decoder
monitor_speed = 115200

main.cpp:

#include <Arduino.h>

void setup() {
  // put your setup code here, to run once:
}

void loop() {
  // put your main code here, to run repeatedly:
}
```

### Debug Message

```plain
ELF file SHA256: 2ac0c63795d8b7a7

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:184
load:0x40078000,len:13228
ho 0 tail 12 room 4
load:0x40080400,len:4
load:0x40080404,len:2924
entry 0x40080570

abort() was called at PC 0x400817c6 on core 0

Backtrace: 0x40082365:0x3ffe3ba0 0x4008660d:0x3ffe3bc0 0x40089375:0x3ffe3be0 0x400817c6:0x3ffe3c50 0x40079312:0x3ffe3c90 |<-CORRUPTED

ELF file SHA256: 2ac0c63795d8b7a7

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:184
load:0x40078000,len:13228
ho 0 tail 12 room 4
load:0x40080400,len:4
load:0x40080404,len:2924
entry 0x40080570

abort() was called at PC 0x400817c6 on core 0

Backtrace: 0x40082365:0x3ffe3ba0 0x4008660d:0x3ffe3bc0 0x40089375:0x3ffe3be0 0x400817c6:0x3ffe3c50 0x40079312:0x3ffe3c90 |<-CORRUPTED
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Jason2866, 2024-08-07:
Esptool detects a Single Core ESP32. Arduino does not support single core esp32.
The reboots you see is normal in this case for an sketch compiled with Arduino.

goddade, 2024-08-08:
The chip's marking is ESP32-D0WD. Is this a fake chip?
[image]
[image]
Only one of them can run single-core firmware.

Jason2866, 2024-08-08:
Whatever it is, it is not a variant usually available for end users. If I where you, I would contact the seller. You did get "something".

closed 2025-03-12 by Jason2866