espressif/arduino-esp32 #10130: BLEServer Memory Leak after device connecting and disconnecting

victorfleischauer, 2024-08-07:
### Board

ESP32-S3-DevKitC-1

### Device Description

ESP32-S3-DevKitC-1 v1.0

### Hardware Configuration

No connections.

### Version

v3.0.4

### IDE Name

PlatformIO

### Operating System

macOS 14.5

### Flash frequency

40Mhz

### PSRAM enabled

no

### Upload speed

460800

### Description

When acting as a BLEServer, after every connection and disconnection the heap memory appears to shrink by 
around 412 bytes (except the first connection, which loses around 1 kilobyte)
This appears to be a memory leak.

### Sketch

```cpp
#include <Arduino.h>
#include <BLEDevice.h>
#include <BLEServer.h>
#include <BLEUtils.h>
#include <BLE2902.h>

BLEServer *pServer = NULL;
BLECharacteristic *pCharacteristic = NULL;

bool deviceConnected = false;
bool oldDeviceConnected = false;
uint32_t value = 0;

// See the following for generating UUIDs:
// https://www.uuidgenerator.net/

#define SERVICE_UUID "4fafc201-1fb5-459e-8fcc-c5c9c331914b"
#define CHARACTERISTIC_UUID "beb5483e-36e1-4688-b7f5-ea07361b26a8"

class MyServerCallbacks : public BLEServerCallbacks
{
  void onConnect(BLEServer *pServer)
  {
    deviceConnected = true;
  };

  void onDisconnect(BLEServer *pServer)
  {
    deviceConnected = false;
  }
};

void setup()
{
  Serial.begin(115200);

  // Create the BLE Device
  BLEDevice::init("ESP32");

  // Create the BLE Server
  pServer = BLEDevice::createServer();
  pServer->setCallbacks(new MyServerCallbacks());

  // Create the BLE Service
  BLEService *pService = pServer->createService(SERVICE_UUID);

  // Create a BLE Characteristic
  pCharacteristic = pService->createCharacteristic(
      CHARACTERISTIC_UUID,
      BLECharacteristic::PROPERTY_READ | BLECharacteristic::PROPERTY_WRITE | BLECharacteristic::PROPERTY_NOTIFY | BLECharacteristic::PROPERTY_INDICATE);

  // Creates BLE Descriptor 0x2902: Client Characteristic Configuration Descriptor (CCCD)
  pCharacteristic->addDescriptor(new BLE2902());

  // Start the service
  pService->start();

  // Start advertising
  BLEAdvertising *pAdvertising = BLEDevice::getAdvertising();
  pAdvertising->addServiceUUID(SERVICE_UUID);
  pAdvertising->setScanResponse(false);
  pAdvertising->setMinPreferred(0x0); // set value to 0x00 to not advertise this parameter
  BLEDevice::startAdvertising();
  Serial.println("Waiting a client connection to notify...");

  delay(2000);

  Serial.printf("Initial heap: %u\n", ESP.getFreeHeap());
}

int disconnectCounter = 0;
void loop()
{
  // disconnecting
  if (!deviceConnected && oldDeviceConnected)
  {
    pServer->startAdvertising(); // restart advertising
    // Serial.println("start advertising");
    oldDeviceConnected = deviceConnected;
    Serial.println("disconnected");
    delay(2000);
    disconnectCounter++;
    Serial.printf("Connections: %d heap: %u\n", disconnectCounter, ESP.getFreeHeap());
  }
  // connecting
  if (deviceConnected && !oldDeviceConnected)
  {
    // do stuff here on connecting
    oldDeviceConnected = deviceConnected;
    Serial.println("connected");
    delay(1000);
  }

  if (!deviceConnected)
  {
    Serial.printf("Connections: %d heap: %u\n", disconnectCounter, ESP.getFreeHeap());
    delay(1000);
  }
}
```

### Debug Message

```plain
entry 0x403c9880
Waiting a client connection to notify...
Initial heap: 255796
Connections: 0 heap: 255796
Connections: 0 heap: 255796
Connections: 0 heap: 255796
connected
disconnected
Connections: 1 heap: 254660
Connections: 1 heap: 254660
Connections: 1 heap: 254660
Connections: 1 heap: 254660
connected
disconnected
Connections: 2 heap: 254248
Connections: 2 heap: 254248
Connections: 2 heap: 254248
Connections: 2 heap: 254248
connected
disconnected
Connections: 3 heap: 253836
Connections: 3 heap: 253836
Connections: 3 heap: 253836
Connections: 3 heap: 253836
Connections: 3 heap: 253836
Connections: 3 heap: 253836
Connections: 3 heap: 253836
Connections: 3 heap: 252168
connected
disconnected
Connections: 4 heap: 253424
Connections: 4 heap: 253424
Connections: 4 heap: 253424
Connections: 4 heap: 253424
connected
disconnected
Connections: 5 heap: 253012
Connections: 5 heap: 253012
Connections: 5 heap: 253012
Connections: 5 heap: 253012
Connections: 5 heap: 253012
Connections: 5 heap: 253012
connected
disconnected
Connections: 6 heap: 252600
Connections: 6 heap: 252600
connected
disconnected
Connections: 7 heap: 252184
Connections: 7 heap: 252184
Connections: 7 heap: 252184
Connections: 7 heap: 252184
Connections: 7 heap: 252184
connected
disconnected
Connections: 8 heap: 251776
Connections: 8 heap: 251776
Connections: 8 heap: 251776
Connections: 8 heap: 251776
Connections: 8 heap: 250104
connected
disconnected
Connections: 9 heap: 251364
Connections: 9 heap: 251364
Connections: 9 heap: 251364
Connections: 9 heap: 251364
connected
disconnected
Connections: 10 heap: 250952
Connections: 10 heap: 250952
Connections: 10 heap: 250952
Connections: 10 heap: 250952
connected
disconnected
Connections: 11 heap: 250532
Connections: 11 heap: 250532
Connections: 11 heap: 250532
Connections: 11 heap: 250532
Connections: 11 heap: 250532
Connections: 11 heap: 250532
Connections: 11 heap: 250532
connected
disconnected
Connections: 12 heap: 250120
Connections: 12 heap: 250120
Connections: 12 heap: 250120
Connections: 12 heap: 250120
Connections: 12 heap: 250120
Connections: 12 heap: 250120
Connections: 12 heap: 250120
connected
disconnected
Connections: 13 heap: 249708
Connections: 13 heap: 249708
Connections: 13 heap: 249708
connected
```

### Other Steps to Reproduce

I have tested this by connecting and disconnecting to the ESP32-S3 using the nRF Connect Android app.

The sketch provided is a modified version of the BLE Notify example.

This bug appears to be the same as the one found by @mirozmrzli in [this discussion](https://github.com/espressif/arduino-esp32/discussions/9834#discussion-6810308)

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Jason2866, 2024-08-07:
The Platformio setup is not good, since it maybe does not use the corresponding Arduino libs how the setup is made.
Better use the Community pioarduino version which does support Arduino core 3.0.4.
Used with:
```
[env:stable]
platform = https://github.com/pioarduino/platform-espressif32/releases/download/51.03.04/platform-espressif32.zip
board = ...
...
```

mirozmrzli, 2024-08-07:
My original observation was done using idf toolchain v4.4 and latest arduino lib for that version, not platformio.

victorfleischauer, 2024-08-07:
> The Platformio setup is not good, since it maybe does not use the corresponding Arduino libs how the setup is made. Better use the Community pioarduino version which does support Arduino core 3.0.4. [...]

Sorry, thanks for informing me, I updated the platformio.ini like so:
```
[env:esp32-s3-devkitc-1]
platform = https://github.com/pioarduino/platform-espressif32/releases/download/51.03.04/platform-espressif32.zip
framework = arduino
board = esp32-s3-devkitc-1
monitor_speed = 115200
```

But the issue still persists:

```
entry 0x403c98ac
Waiting a client connection to notify...
Initial heap: 254276
Connections: 0 heap: 254276
Connections: 0 heap: 254276
connected
disconnected
Connections: 1 heap: 253140
Connections: 1 heap: 253140
connected
disconnected
Connections: 2 heap: 252728
Connections: 2 heap: 252728
Connections: 2 heap: 251060
connected
disconnected
Connections: 3 heap: 252316
Connections: 3 heap: 252316
Connections: 3 heap: 252316
connected
disconnected
Connections: 4 heap: 251904
Connections: 4 heap: 251904
```

victorfleischauer, 2024-08-07:
I have found [this issue](https://github.com/espressif/esp-idf/issues/12124) on the idf repository which from a surface level understanding looks similar to this one here. From what I gathered, the problem there was caused by something in the build environment. I will try to uninstall and reinstall platformio as well as test it on another computer to see if the issue persists.

SuGlider, 2024-08-09:
@victorfleischauer - I tested it using IDF 5.1.4 with Arduino Core 3.0.4 (Arduino IDE Building system).
The free heap space goes down in the initial process of connecting/disconnecting.
After some 15 or so interactions it keeps steady.
The report/output can be seen below:

```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3818,len:0x508
load:0x403c9700,len:0x4
load:0x403c9704,len:0xad0
load:0x403cc700,len:0x29e4
entry 0x403c9880
Waiting a client connection to notify...
Initial heap: 259372
Connections: 0 heap: 259372
Connections: 0 heap: 259372
Connections: 0 heap: 259372
connected
disconnected
Connections: 1 heap: 258236
Connections: 1 heap: 258236
connected
disconnected
Connections: 2 heap: 257824
Connections: 2 heap: 257824
connected
disconnected
Connections: 3 heap: 257412
Connections: 3 heap: 257412
Connections: 3 heap: 257412
connected
disconnected
Connections: 4 heap: 257000
Connections: 4 heap: 257000
Connections: 4 heap: 257000
Connections: 4 heap: 255332
connected
disconnected
Connections: 5 heap: 256588
Connections: 5 heap: 256588
Connections: 5 heap: 256588
connected
disconnected
Connections: 6 heap: 256176
Connections: 6 heap: 256176
Connections: 6 heap: 256176
connected
disconnected
Connections: 7 heap: 255760
Connections: 7 heap: 255760
Connections: 7 heap: 255760
connected
disconnected
Connections: 8 heap: 255352
Connections: 8 heap: 255352
Connections: 8 heap: 255352
connected
disconnected
Connections: 9 heap: 254940
Connections: 9 heap: 254940
Connections: 9 heap: 253268
connected
disconnected
Connections: 10 heap: 254528
Connections: 10 heap: 254528
Connections: 10 heap: 254528
connected
disconnected
Connections: 11 heap: 254108
Connections: 11 heap: 254108
Connections: 11 heap: 252436
connected
disconnected
Connections: 12 heap: 253696
Connections: 12 heap: 253696
Connections: 12 heap: 253696
connected
disconnected
Connections: 13 heap: 253284
Connections: 13 heap: 253284
Connections: 13 heap: 253284
connected
disconnected
Connections: 14 heap: 252872
Connections: 14 heap: 252872
Connections: 14 heap: 252872
connected
disconnected
Connections: 15 heap: 252460
Connections: 15 heap: 252460
Connections: 15 heap: 252460
connected
disconnected
Connections: 16 heap: 252460
Connections: 16 heap: 252460
Connections: 16 heap: 252460
connected
disconnected
Connections: 17 heap: 252460
Connections: 17 heap: 252460
Connections: 17 heap: 252460
connected
disconnected
Connections: 18 heap: 252460
Connections: 18 heap: 252460
Connections: 18 heap: 252460
connected
disconnected
Connections: 19 heap: 252460
Connections: 19 heap: 252460
Connections: 19 heap: 252460
Connections: 19 heap: 252460
connected
disconnected
Connections: 20 heap: 252460
Connections: 20 heap: 252460
Connections: 20 heap: 252460
connected
disconnected
Connections: 21 heap: 250056
connected
disconnected
Connections: 22 heap: 252460
Connections: 22 heap: 252460
connected
disconnected
Connections: 23 heap: 252460
Connections: 23 heap: 252460
Connections: 23 heap: 252460
connected
disconnected
Connections: 24 heap: 252460
Connections: 24 heap: 252460
Connections: 24 heap: 252460
Connections: 24 heap: 252460
connected
disconnected
Connections: 25 heap: 252460
Connections: 25 heap: 252460
Connections: 25 heap: 252460
connected
disconnected
Connections: 26 heap: 250076
connected
disconnected
Connections: 27 heap: 252460
Connections: 27 heap: 252460
Connections: 27 heap: 252460
connected
disconnected
Connections: 28 heap: 251184
Connections: 28 heap: 251184
connected
disconnected
Connections: 29 heap: 250092
connected
disconnected
Connections: 30 heap: 250040
connected
disconnected
Connections: 31 heap: 252460
Connections: 31 heap: 252460
Connections: 31 heap: 252460
connected
disconnected
Connections: 32 heap: 252460
Connections: 32 heap: 252460
Connections: 32 heap: 252460
connected
disconnected
Connections: 33 heap: 252460
Connections: 33 heap: 252460
Connections: 33 heap: 252460
Connections: 33 heap: 252460
connected
disconnected
Connections: 34 heap: 252460
Connections: 34 heap: 252460
Connections: 34 heap: 252460
connected
disconnected
Connections: 35 heap: 252460
Connections: 35 heap: 252460
Connections: 35 heap: 252460
connected
disconnected
Connections: 36 heap: 252460
Connections: 36 heap: 252460
Connections: 36 heap: 252460
connected
disconnected
Connections: 37 heap: 250092
connected
disconnected
Connections: 38 heap: 250076
connected
disconnected
Connections: 39 heap: 250092
connected
disconnected
Connections: 40 heap: 250040
connected
disconnected
Connections: 41 heap: 252460
Connections: 41 heap: 252460
Connections: 41 heap: 252460
connected
disconnected
Connections: 42 heap: 251184
Connections: 42 heap: 251184
connected
disconnected
Connections: 43 heap: 252460
Connections: 43 heap: 252460
connected
disconnected
Connections: 44 heap: 250076
connected
disconnected
Connections: 45 heap: 252460
Connections: 45 heap: 252460
connected
disconnected
Connections: 46 heap: 250076
connected
disconnected
Connections: 47 heap: 252460
Connections: 47 heap: 252460
Connections: 47 heap: 252460
connected
disconnected
Connections: 48 heap: 252460
Connections: 48 heap: 252460
Connections: 48 heap: 252460
connected
disconnected
Connections: 49 heap: 252460
Connections: 49 heap: 252460
Connections: 49 heap: 252460
Connections: 49 heap: 252460
connected
disconnected
Connections: 50 heap: 252460
Connections: 50 heap: 252460
Connections: 50 heap: 252460
Connections: 50 heap: 252460
connected
```

SuGlider, 2024-08-09:
I think that the process is a bit dynamic as it has many tasks being executed along time.
But it seems that there is no memory leak.

Let me know if you agree and we could close this issue.

mirozmrzli, 2024-08-09:
Uhh, memory dropping without being reclaimed is a definition of memory
leak. I also tried nimble which doesn't exhibit this behavior.

SuGlider, 2024-08-09:
The current BLE layer uses Arduino String and some other little parts that allocate small chunks of memory.

I think that this behavior may be a mix with fragmentation problems as well, rather than exclusively a memory leak.

The current BLE library is subject to a future refactoring that will be based on NimBLE.

victorfleischauer, 2024-08-12:
Thanks @SuGlider, interesting to hear that the Arduino BLE library is planed to use NimBLE, I noticed the [NimBLE-Arduino](https://github.com/h2zero/NimBLE-Arduino) does not have this behavior.

I will close this issue then.

closed 2024-08-12 by victorfleischauer