espressif/arduino-esp32 #10131: Is there any update plan for ESP32 P4? I want to experience Arduino running on ESP32 P4 as soon as possible

xuelse, 2024-08-08:
### Related area

Arduino

### Hardware specification

ESP32 P4

### Is your feature request related to a problem?

Is there any update plan for ESP32 P4? I want to experience Arduino running on ESP32 P4 as soon as possible

### Describe the solution you'd like

Is there any update plan for ESP32 P4? I want to experience Arduino running on ESP32 P4 as soon as possible

### Describe alternatives you've considered

Is there any update plan for ESP32 P4? I want to experience Arduino running on ESP32 P4 as soon as possible

### Additional context

Is there any update plan for ESP32 P4? I want to experience Arduino running on ESP32 P4 as soon as possible

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

P-R-O-C-H-Y, 2024-08-08:
@VojtechBartoska PTAL

Jason2866, 2024-08-08:
@xuelse Where have you managed to buy a esp32-p4?

atanisoft, 2024-08-08:
@Jason2866 there were a few ESP32-P4 boards on Espressif's AliExpress store but they have all been purchased. Hopefully they will be restocked soon.

xuelse, 2024-08-09:
> 乐鑫的速卖通店铺上有几块 ESP32-P4 开发板，但都已被购买。希望它们能很快补货。

https://item.taobao.com/item.htm?spm=a1z10.3-c.w4002-8715811646.9.332b9a389M4KAk&id=806603819251&pisk=fBLKOStayAD3Z30L03iGrPZ2PLGMvedUBpRbrTX3Vdp9nQWlZXWhF4pAIzcF-07JeOpPKMbCKzQJwLJlVUXkyW89Hz4nLDR7yabSnx0moBPFzaMDYzfi-J5F9yglOM6syhz5jx0moCU9Jx8-nb5_hhjAZa1CVg6s6_f4OasCPPGON_2CF99W1faUwbgrDy8eX6gC2A9Tdt-d9MB8rOU-obfCvLRdBxMXWqSdhB6TPxqKeJBRQEMiXCRDApfweqHOzd-WF_p_CYjeNhpO9pDYUTYetUBH9DMH6NXpgOdtkoJAJLIXRGF7yw_RBNC6lbqVIFBHFeIsirIl5EjfRhmZKgbdMLTep7UWhgrJoEdpr-XAZkGt6urPA1uNbzZ1FHIPQ1Bm96Ez4g1N6tct6urPA15OnXeL4uS5_&skuId=5484284768477

VojtechBartoska, 2024-09-02:
Hello, closing this in favor of https://github.com/espressif/arduino-esp32/issues/10278.

closed 2024-09-02 by VojtechBartoska