espressif/arduino-esp32 #10132: GDEH0154D67 eink screen working worse after moving arduino to being as a component of esp-idf

Szybet, 2024-08-08:
### Board

Custom, Watchy v2

### Device Description

eink screen connected to ESP32-PICO-D4 (revision v1.1)

### Hardware Configuration

https://github.com/sqfmi/watchy-hardware/blob/v2.0/WatchySchematic.pdf

### Version

v3.0.1

### IDE Name

Platformio

### Operating System

Linux

### Flash frequency

Tried 40Mhz and 80Mhz

### PSRAM enabled

no

### Upload speed

115200

### Description

After I moved arduino core outside fully outside of platformio, as a component of esp idf as described here: https://docs.espressif.com/projects/arduino-esp32/en/latest/esp-idf_component.html the eink screen I'm driving produces horrible ghosting, I can provide a video if that's needed

I tried esp idf 5.1.4 up to 5.1 and arduino core from master up to 3.0.1, the previous setup was working when including arduino core into platformio, as self precompiled arduinoespressif32. I also tried including the configs from the default sdkconfigs I was using to precompile arduinoespressif32 but this didn't changed anything.

Here is the related esp idf issue, but it's more with arduino core now that I investigated
https://github.com/espressif/esp-idf/issues/14324
Also the driver, which was working before without issues:
https://github.com/ZinggJM/GxEPD2

### Sketch

```cpp
https://github.com/Szybet/InkWatchy

Creating a minimized sketch didn't change anything, so no point in providing it, but I tried
```

### Debug Message

```plain
There is no debug message, it's a visual error
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Jason2866, 2024-08-08:
@Szybet As I already mentioned in issue https://github.com/Jason2866/platform-espressif32/issues/78 the way you setup Platformio.ini is not the way as recommended.
Strange issues can happen with this setup. Change this setup!

Szybet, 2024-08-08:
I went through all the weird things in my repo that could have caused it with Jason and after fixing all things it still doesn't work

Szybet, 2024-08-08:
Tested with another board and another screen, the difference there is smaller to the point it doesn't bother me, but the question remains, why does this happen???

Jason2866, 2024-08-08:
@Szybet Please close since there is no Arduino core code involved.

Szybet, 2024-08-08:
Then who should I bother? It's about how arduino core is integrated into esp idf

Jason2866, 2024-08-09:
Don't understand the problem. Arduino as an component of IDF works. Your example does not use this variant either. It is an IDF only project. Furthermore your code runs. The screen flicker you see is for sure not related to Arduino core, since not involved.

Szybet, 2024-08-09:
Depending on how I add arduino core to my project (Which I do use for many thing, the screen driver too):
- as a component, in the `component` dir, the screen flickers badly
- as precompiled binaries via arduinoespressif32 , the screen behaves normally

So it is a arduino core problem?

Jason2866, 2024-08-09:
No, since you wrote with the precompiled Arduino libs it does work. Providing it as a component the Arduino libs are compiled with the settings **you** have done.
You are doing your **own** Arduino build this way. It is not the same as the ready made provided.
You write exactly this. Your self compiled version flickers. The ready made Arduino does not flicker. So it is clearly that it is **no** Arduino issue provided from the version here

Szybet, 2024-08-09:
I use(d) my own precompiled binaries:
https://github.com/Szybet/arduinoespressif32-inkwatchy
https://github.com/Szybet/inkwatchy-lib-builder

The configs in there are unchanged, but I add my own on top of them: 
https://github.com/Szybet/inkwatchy-lib-builder/blob/main/apply_sdkconfig.sh

I obviously tried the other way, applying the configs from that repo to the sdkconfig that the new compilation method uses, but the screen still flickers, nothing changed

So no, I don't have different settings

Szybet, 2024-08-09:
Related arduino forum topic:
https://forum.arduino.cc/t/gxepd2-with-gdeh0154d67-display-ghosting/1290092

Here are the videos, maybe the issue will be clearer after I post them.
Here is the worst result, with the new compiling method:

https://github.com/user-attachments/assets/5bce6c7c-fd7f-4a9a-b0f8-9d3bf5c93ede

Notice it going grey after the reset message, before the watchface? That's the worse part, but the ghosting when going to the menu is also bad.

Here is a video on different hardware and screen, but the newer compiling method:

https://github.com/user-attachments/assets/afdb9540-0458-4acc-a8d0-94f48750f1a8

The problem here is less visible, but it still goes a little grey, so the problem may affect older screens, but those are compatible and weren't an issue before

Finally, an old build where everything is as it should be:

https://github.com/user-attachments/assets/c5552f13-7dda-48e5-85f0-24b808172d18

There is no to grey transition and there is a bit less ghosting after going to the menu

The difference on the other hardware is less visible, but it worked perfectly on the hardware shown on the first video too, after the software change it creates a mess

Jason2866, 2024-08-09:
>The configs in there are unchanged, but I add my own on top of them

No difference?

Szybet, 2024-08-09:
previously, To the default setting I added at the end of the file my own settings
Now, to my settings at the bottom of the file I add the default settings

It doesn't matter...

Jason2866, 2024-08-09:
Closing since issue is not existing with official Arduino version

closed 2024-08-09 by Jason2866