espressif/arduino-esp32 #10135: Implement Zigbee in Arduino

P-R-O-C-H-Y, 2024-08-09:
### Related area

Create an API for Zigbee within Arduino

### Hardware specification

ESP32-C6 and ESP32-H2 as Standalone Nodes. Other SoC can be used as radio co-processor attached to a RPC (802.15.4 radio layer).

### Is your feature request related to a problem?

This is an issue to track the progress of developing Zigbee library, a new API for Arduino.

### Describe the solution you'd like

**Initial tasks:**
- [ ] Add zigbee-sdk to Arduino lib-builder https://github.com/espressif/esp32-arduino-lib-builder/pull/146
- [ ] Add zigbee support #8844
- [ ] Port basic IDF examples to Arduino #9024 #9627

**Zigbee library (wrapper) tasks:**
- [ ] Create Zigbee classes and implement all Zigbee roles
- [ ] Implement Zigbee network scanning
- [ ] Allow multiple endpoints on same Zigbee device (test needed)
- [ ] Implement most HA devices (lights, switches, sensors, etc..)
- [ ] Implement setting Manufacturer and model name
- [ ] Update ported examples to use the Zigbee Library
- [ ] Documentation

more tasks will be added while in progress.

### Describe alternatives you've considered

No other usable Zigbee wrapper around.

### Additional context

Related issues:
#8807 #9156 #9962 #9480

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

elgerg, 2024-08-09:
Hi, love the work so far. I'm so excited to get this running. It looks like it's going to be the next best thing since sliced bread!

Are there any intentions to add power source/battery level into the config?

Many thanks and appreciate the hard work!

P-R-O-C-H-Y, 2024-08-12:
Dev update 12/08/2024 pushed to the dev branch:
- Implemented Factory reset of Zigbee device, in order to connect to new network without reflashing/erasing flash
- Implemented optional setting for Manufacturer and Model names
- Added option to allow endpoint to have multiple endpoint connected -> switch - 2 lights (tested)
- Implemented easy transfer from device it to Device type (0x0000 = ESP_ZB_HA_ON_OFF_SWITCH_DEVICE_ID -> "General On/Off switch".
- Minor sketches update

elgerg, 2024-08-12:
Oh wow, that looks amazing! So easy, I'm going to love this. Thanks!

P-R-O-C-H-Y, 2024-08-12:
> Hi, love the work so far. I'm so excited to get this running. It looks like it's going to be the next best thing since sliced bread!
> 
> Are there any intentions to add power source/battery level into the config?
> 
> If mains powered could it act as a router as well or is that too much for a H2? Maybe ok for the C6..
> 
> Many thanks and appreciate the hard work!

Thank you for your feedback 👍 I really appreciate it.

About the power source/baterry level I will add it on the list as good to have, so I will look after having the main part done :)

P-R-O-C-H-Y, 2024-08-14:
Dev Update 14/08/2024 pushed to dev branch:

- Implemented Color Dimmable Light and Colour Dimmer Switch endpoints, working in RGB colour space (automatically convert to/from XYZ).
- Added examples: Zigbee_Color_Dimmable_Light, Zigbee_Color_Dimmer_Switch
- Minor code updates

elgerg, 2024-08-14:
Nice,
Any chance you can take a look at adding temp sensor and light sensor?
They are the 2 I'm most eager to try, I'm currently using the temp sensor and it works great (even if the code is way difficult to understand).
Thanks!

P-R-O-C-H-Y, 2024-08-14:
> Nice, Any chance you can take a look at adding temp sensor and light sensor? They are the 2 I'm most eager to try, I'm currently using the temp sensor and it works great (even if the code is way difficult to understand). Thanks!

@elgerg The temperature sensor + thermostat are on the list of upcoming tasks.
The light sensor device type is not present [in the list](https://github.com/espressif/esp-zigbee-sdk/blob/7dab668fef089bd8ccf0b137df3ed690bd66fc13/components/esp-zigbee-lib/include/zcl/esp_zigbee_zcl_common.h#L99) in esp32-zigbee-sdk, it should have value 0x106, which is not present.
Can you open a feature request to the esp-zigbee-sdk repository please? Seems that the main part is already [there](https://github.com/espressif/esp-zigbee-sdk/blob/7dab668fef089bd8ccf0b137df3ed690bd66fc13/components/esp-zigbee-lib/include/zcl/esp_zigbee_zcl_illuminance_meas.h#L38) which contains all the clusters and attributes for luminance sensor.

elgerg, 2024-08-14:
Thanks for adding to https://github.com/espressif/esp-zigbee-sdk/issues/401
I'm quite excited about being able to create my own Temp/Lumin Zigbee sensors :D
Thanks for the continued effort!

elgerg, 2024-08-15:
Hi @P-R-O-C-H-Y 

Any chance I can add this to the "Nice to have" list?

In the current Temp sensor version there is an identify cluster (?) added:

```
    esp_zb_cluster_list_add_identify_cluster(cluster_list, esp_zb_identify_cluster_create(&(temperature_sensor->identify_cfg)), ESP_ZB_ZCL_CLUSTER_SERVER_ROLE)

```

Any chance you can add something like that to the list so we can blink an LED when triggered? No idea how to do that at the moment..

Thanks!

P-R-O-C-H-Y, 2024-08-16:
> Hi @P-R-O-C-H-Y
>
> Any chance I can add this to the "Nice to have" list?
>
> In the current Temp sensor version there is an identify cluster (?) added:
>
> ``` [...]

Hi @elgerg,

Can you explain a bit more? I am not sure if I got it correctly. 
What would be "Nice to have"? Option to customise identify cluster? I don't also get the LED blink.

elgerg, 2024-08-16:
Hi @P-R-O-C-H-Y 
I should have been more specific.

If you have a look at this:
[image]

There is an identify button. In theory this should allow for the device to identify itself by blinking an LED or something when the button is pressed so you can figure out which device is which.

This article explains really well on what identify is:
https://medium.com/@omaslyuchenko/hello-zigbee-part-22-identify-cluster-90cf12680306

So what would be good to have is a simple way of using this just like you are doing with the rest of the wrapper.

Does that make a little more sense?

Thanks

P-R-O-C-H-Y, 2024-08-16:
> Hi @P-R-O-C-H-Y I should have been more specific.
>
> If you have a look at this: [...]

Thanks for explaining. That can be really helpful. Adding it to the list :)

elgerg, 2024-08-16:
Excellent. Thanks :)

P-R-O-C-H-Y, 2024-08-16:
Dev Update 16/08/2024 pushed to dev branch:

- Implemented Zigbee network scanning (mostly match WiFi scan API)
- Added Zigbee_Network_Scan example

elgerg, 2024-08-18:
If you're taking requests I'd also like to implement some kind of way of getting the current time from the coordinator to display on a screen. Do you know if it's possible to get the time into some sort of callback?

Thanks :)

P-R-O-C-H-Y, 2024-08-28:
> If you're taking requests I'd also like to implement some kind of way of getting the current time from the coordinator to display on a screen. Do you know if it's possible to get the time into some sort of callback?
> 
> Thanks :)

Do you have any example of this feature?

P-R-O-C-H-Y, 2024-08-28:
Dev Update 28/08/2024 pushed to dev branch:

- Implemented Temperature sensor and Thermostat endpoints.
- Added examples: Zigbee_Temperature_Sensor, Zigbee_Thermostat
- Implemented configure report handler.
- Updated READMEs and description of examples.
- Minor code updates

elgerg, 2024-08-28:
> Dev Update 28/08/2024 pushed to dev branch:
> 
> * Implemented Temperature sensor and Thermostat endpoints.
> * Added examples: Zigbee_Temperature_Sensor, Zigbee_Thermostat
> * Implemented configure report handler.
> * Updated READMEs and description of examples.
> * Minor code updates

This looks awesome!

Do you know when it'll end up in the next release?
Thanks 🙏

Hedda, 2024-09-02:
@P-R-O-C-H-Y Tip; here is a other Arduino library for creating Zigbee Home Automation devices (with links to some examples):

* https://github.com/prairiesnpr/zha_arduino

P-R-O-C-H-Y, 2024-09-02:
> This looks awesome!
>
>
>
> Do you know when it'll end up in the next release?
>
> Thanks 🙏

I am doing the best that the first version can be released soon.

P-R-O-C-H-Y, 2024-09-26:
Dev Update 26/09/2024 - PR ready to be reviewed.

- Refactored methods and classes (exchanged virtual methods for callback functions making it more simpler 
- Implemented Identify (tested with HomeAssistant - SONOFF USB Dongle gateway + ESP with RGB example)
- All examples got updated due to the refactor

P-R-O-C-H-Y, 2024-10-02:
The Zigbee library got merged! Now it's available in both`master` and `release/v3.1` branches.
So in next release 3.0.6 and 3.1.0-rc2 it will be included.

lboue, 2024-10-02:
Great job!

Abdull, 2024-10-02:
awesome to hear!

I'm inexperienced what is required to implement software for Zigbee devices.

Having a look at directory https://github.com/espressif/arduino-esp32/tree/master/libraries/Zigbee/src/ep , do I deduce correctly that each "category" of Zigbee device type would require its own implementation files?

Maybe you can give hints how others could join?

P-R-O-C-H-Y, 2024-10-03:
> awesome to hear!
>
> I'm inexperienced what is required to implement software for Zigbee devices.
>
> Having a look at directory [...]

Hello @Abdull, take a look at the basics of the Zigbee protocol. You can find many sources and videos on the internet ;)
As you noticed, Zigbee device can have 1 or more endpoints, which specify its type (light, sensor, switch, etc.). Each type need to have its own implementation as you found in the library (the link you posted). I am open on any suggestion what EPs to add in next updates and also I would be really happy to see a PR with any new EP support. As a reference you can take a look at the EPs I have added. For more details check the [esp32-zigbee-sdk](https://github.com/espressif/esp-zigbee-sdk) repository and sources there.

elgerg, 2024-10-04:
Hi @P-R-O-C-H-Y 

If you're taking requests these are the sort of things I would like to see:

Human presence detector (PIR etc)
Luminosity/Light sensor
Battery monitoring
Sleepy device

Thanks!

MeisterQ, 2024-10-07:
Id also like to put a feature request:

- TRV support with stuff like external Temp sensor, window open, temperature setpoint, etc (like everything the Shelly TRV supports but only on zigbee)

P-R-O-C-H-Y, 2024-10-07:
> Id also like to put a feature request:
> 
> * TRV support with stuff like external Temp sensor, window open, temperature setpoint, etc (like everything the Shelly TRV supports but only on zigbee)

Hi @MeisterQ, the TRV is just a thermostat. We have support for this type of endpoint. Also we have a temperature sensor endpoint supported as well. Just regular sensor for example for windows, doors, etc. is not yes supported. So if you plan to build your own TRV you can already start prototyping with the stuff we already have :)

Hedda, 2024-10-07:
> If you're taking requests these are the sort of things I would like to see:
> 
> Human presence detector (PIR etc) 
> Luminosity/Light sensor
> Time Client (get the current time from the coordinator)
> Battery monitoring 
> Sleepy device
> 
> Thanks!

FYI, there are more Zigbee examples in the [esp32-zigbee-sdk](https://github.com/espressif/esp-zigbee-sdk) repository which could probably be converted to this new API. See:

* https://github.com/espressif/esp-zigbee-sdk/tree/main/examples

That includes examples for "deep sleep" and "light sleep" (standby) modes for Zigbee End Device:

* https://github.com/espressif/esp-zigbee-sdk/tree/main/examples/esp_zigbee_sleep

As for Battery monitoring, I read note Arduino recently released an open-source Arduino Power Management library for Arduino Pro that could maybe be ported? It includes battery management and charger manager + sleep and standby mode

* https://github.com/arduino-libraries/Arduino_PowerManagement/
  * https://github.com/arduino-libraries/Arduino_PowerManagement/tree/main/examples

More on that here 

* https://www.cnx-software.com/2024/09/26/arduino-power-management-library-arduino-pro-modules/

Time Client looks to be missing, and I agree that Zigbee Time Client is needed to make a standa-alone smart thermostat or HVAC.

MeisterQ, 2024-10-07:
> Hi @MeisterQ, the TRV is just a thermostat. We have support for this type of endpoint. Also we have a temperature sensor endpoint supported as well. Just regular sensor for example for windows, [...]

Thank you. That could be a start. None of the TRVs existing meet my requirements

gdivry, 2024-10-13:
Thanks a lot for this incredible work 
Can I use the esp32c6 as a Zigbee coordinator with the lib ?

dpharris, 2024-10-13:
Could README.md files be possibly added to the arduino-esp32/libraries
/Zigbee/ directory,  to the example directory, and/or the individual
examples?

These would be very helpful to people new to this.  They could also include
actual or links to pictures and videos.

Thanks
David

Hedda, 2024-10-22:
> > Can I use the esp32c6 as a Zigbee coordinator with the lib ?
> 
> As I understand this is currently only to use ESP32-C6/ESP32-H2 as a ”Zigbee End Device” (ZED) or ”Zigbee Router” device (ZR) stand-alone nodes (i.e. standalone devices that can be added to an existing Zigbee network and then controlled via it), not as Zigbee Coordinator or Zigbee Gateway.
> 
> If you want a all-in-one Zigbee Gateway using ESP32 then can I suggest that you look into adding support for [esp-zigbee-sdk](https://github.com/espressif/esp-zigbee-sdk) support to Tasmota's sub-project called Zigbee2Tasmota (Z2T) which is currently only compatible with Silicon Labs and Texas Instruments as Zigbee Coordinator but it might give you ideas:
> 
> * https://tasmota.github.io/docs/Zigbee/
> 
> and
> 
> * https://tasmota.github.io/docs/Zigbee-Internals/

@s-hadinger to go off on a tangent with question above; what exact functions are still missing for this new Zigbee API for Arduino to potentially be useful for a Zigbee Gateway project like Zigbee2Tasmota (Z2T)?

s-hadinger, 2024-10-22:
> @s-hadinger to go off on a tangent with question above; what exact functions are still missing for this new Zigbee API for Arduino to potentially be useful for a Zigbee Gateway project like Zigbee2Tasmota (Z2T)?

I don't know, I have not checked. But there are no plans to port Z2T to H2/C6 due to the predominance of CC2652P and low number of devices using H2/C6. Also for the future, I expect Thread to progressively take over Zigbee. So I think we are well served with Zigbee as of today with both EFR32/CC2652P compatibility

P-R-O-C-H-Y, 2024-11-01:
Another update for the Zigbee library -> PR #10551 

- Support for sleepy device (only deep-sleep is currently possible).
- Support for selecting power source for the device. If battery is selected, the remaining percentage can be updated.
- Support for Humidity cluster, which was added as optional for the Temperature sensor as there is no HA Device type for humidity sensor.
- New example Zigbee_Temp_Hum_Sensor_Sleepy, which demonstrates all the new functionalities
- Mirror fixes and improvements

P-R-O-C-H-Y, 2024-11-01:
I have also did a power consumption test for the sleepy device. Sharing the picture of consumption when sleep was set to 30s, after wakeup, the connection was done, measured data and reported. Then back to sleep.

[image]
You can see the average consumption was around 6mA. The more time the device will sleep the better power efficiency you can get.

tlanfer, 2024-11-15:
This is great work, and i managed to get it to work immediately using a waveshare ESP32-C6 devkit board.
I'm curious about the possiblities with sleep. One of the major reasons i'm using zigbee devices is when they need to run on low power. I have a bunch TRVs that run on battery at last a long time (months), yet respond to commands almost immediately (less than 5 seconds delay). My specific usecase is having a battery powered LED light that i only need to activate every now and then, but should not drain the battery when not in use. Zigbee would be perfect for this.

With deep sleep i'm getting a full reboot, and if i want reasonably quick response times i can only sleep for relatively short times. So i would assume i need some form of light sleep for that? Just wake up enough to check for commands and go back to sleep?

ggtimtom, 2024-11-17:
I want to build two sensors:
A [rain meter](https://www.thingiverse.com/thing:13383) using a hallsensor an a magnet

A digital water meter: In my installed watermeter a little metalsheet is rotating. As a sensor i will use an geartoothsensor.

AndunHH, 2024-11-17:
Thanks for the great work done here. As this whole feature is not jet finished, I'm hesitating to create a new issue. If this here is not the right spot, please give me a hint, where I should put it.

I succesfully used the example for the Switch and the on/off light with the given "light" example on a second seeed xiao esp32-c6 and with an ikea power outlet.

I tried to expand the example, because I don't want to bind a power outlet **manually after every reset**. I already found a [similar ticket for the espressif SDK.](https://github.com/espressif/esp-idf/issues/11904), that tells me, that this is already possible.

I changed the example to not wait in the setup() for a bound device. But then: nothing happens, because _is_bound = false in the follow up routings.
If I bind the first light again, the whole switch starts working and also the formerly bound power outlet is switching as expected. (= the newly bound light and the power outlet are switching together)

Therefore, I assume, that during initialization the state of _is_bound is not correctly read from the memory, right?
I'm was looking at https://github.com/espressif/arduino-esp32/blob/e99f1fdc6d12e07324be6e956318496e6a0bfac3/libraries/Zigbee/src/ZigbeeEP.cpp#L13 but at this point my knowdlege comes to an end. ...

Thanks for any hints how to solve this, or if this already known and planned. 
Or where I should open a individual ticket.

Hedda, 2024-11-19:
@P-R-O-C-H-Y have you thought about maybe splitting Zigbee API for Arduino to make one part hardware-agnostic?

That is, breakout the Zigbee API from arduino-esp32 into its own separate library makes that hardware independent so that it could be reused by other projects with other Zigbee radio libraries, especially if they too have based their Zigbee stack on ZBOSS,  such as example Nordic Semiconductor. With the point being that such a library would attract more Arduino developers that are interested in this Zigbee API but might want to instead make an implementation for a non-ESP32 based radio.

P-R-O-C-H-Y, 2024-11-21:
> This is great work, and i managed to get it to work immediately using a waveshare ESP32-C6 devkit board. I'm curious about the possiblities with sleep. One of the major reasons i'm using zigbee [...]

Hi @tlanfer, thanks for your feedback. 
The light sleep is not possible, as it needs the Power Management enabled and use the FreeRTOS tickless mode. We don't want to enable this settings for whole Arduino core, as it may bring unexpected behavior. If you really need the light sleep, you can use the ESP-IDF and the ESP-ZIGBEE-SDK directly. There is an example on how to use the light sleep :)

P-R-O-C-H-Y, 2024-11-21:
> Thanks for the great work done here. As this whole feature is not jet finished, I'm hesitating to create a new issue. If this here is not the right spot, please give me a hint, where I should put [...]

Hi @AndunHH, thank you for your feedback. I will take a look on this issue with the `_is_bound` if I can see anything wrong there. The TODO comment is there for me, as I might transfer all those variables  to not be a static if possible. 
If you have more info about that issue you have also with an example and steps how to reproduce it, please open a new issue and describe everything there. It will help a lot to have it separated, so we can discuss it there :)

P-R-O-C-H-Y, 2024-11-21:
> @P-R-O-C-H-Y have you thought about maybe splitting Zigbee API for Arduino to make one part hardware-agnostic?
>
> That is, breakout the Zigbee API from arduino-esp32 into its own separate library [...]

Hi @Hedda, I was not thinking about that as I use the ESP-ZIGBEE-SDK underneath, which uses the ZBOSS. The main idea of Arduino APIs is to offer a similar approach to other Arduino interfaces. If users want to use ESP-IDF Zigbee APIs, that is still possible, they can just ignore the Arduino Zigbee APIs. This is valid for most of our Arduino APIs, actually, users can always use the ESP-IDF underneath. We don't have a goal of portability of our libraries, only to keep portability of user applications. But for sure other manufacturers can follow the same APIs, but the Implementation will depend on them.

HexiDev, 2024-12-17:
Hi! I'm not sure. But is there already a wrapper so you can communicate from device to device e.g. a button that toggles another zigbee light without using automations like homeassistant and/or nodered

HexiDev, 2024-12-17:
@P-R-O-C-H-Y Hi! I'm not sure. But is there already a wrapper so you can communicate from device to device e.g. a button that toggles another zigbee light without using automations like homeassistant and/or nodered

P-R-O-C-H-Y, 2024-12-17:
> @P-R-O-C-H-Y Hi! I'm not sure. But is there already a wrapper so you can communicate from device to device e.g. a button that toggles another zigbee light without using automations like homeassistant and/or nodered
> 
> 

Hi @HexiDev, just flash one ESP with on/off light and second ESP with on/off switch examples. The switch will act as a coordinator and create the Zigbee network, light will join the network and bind to the switch. So clicking a button on switch will turn on/off the light :)

HexiDev, 2024-12-17:
But would it be also possible to use an existing network and make 2 zigbee end/router devices talk to eachother? Like in a sense that I can setup a switch as a router that can turn on/off my light that is also a router.

instead of communicating coordinator > end/router device

Hedda, 2024-12-18:
@HexiDev the Zigbee function you are referring to is called Zigbee Binding, and you normally ”bind” two compatible cluster attributes. 

Binding is an on-device feature for Zigbee devices. It provides a mechanism for attaching an endpoint of one Zigbee device to an endpoint of another Zigbee device or to a Zigbee group.

For example, binding a “target destination” Zigbee device like a remote to a Zigbee light bulb, switch or group of light bulbs allows direct control of the “target” device (light, switch, shade) from the “remote” Zigbee device, bypassing the Zigbee Gateway. This means that the remote can control the light bulb or group even when any Zigbee Coordinator and Zigbee Gateway is not active.

Note that not all devices support binding. By default, most any Zigbee Gateway applications binds remotes to the coordinator, so click events are still forwarded to the Zigbee Gateway application for state reporting.

Note that as some remotes can only be bound to a single destination, you might need to unbind the remote from the coordinator before binding it to another device or group.

See how bindings is used in other software:

* https://www.home-assistant.io/integrations/zha#zigbee-binding-and-unbinding

* https://www.zigbee2mqtt.io/guide/usage/binding.html

* https://www.openhab.org/addons/bindings/zigbee/

patrickfnielsen, 2024-12-18:
Examples/support for illuminance/occupancy would be nice, or maybe an example for how support could be implemented if you know the clusters/endpoints/attributes needed.

lsroka76, 2024-12-22:
Does anybody successfully compiled Zigbee Gateway example in Arduino. Especially ot_rcp part?

Farisabdurrasid, 2025-01-08:
@P-R-O-C-H-Y hi, can you help me?, I'm not sure if there is a way to configure esp32c6 as an end device and esp32H2 as a coordinator so they can connect to each other by sending messages via serial monitor. Is it possible and may I know where I can get the code?

P-R-O-C-H-Y, 2025-01-08:
> @P-R-O-C-H-Y hi, can you help me?, I'm not sure if there is a way to configure esp32c6 as an end device and esp32H2 as a coordinator so they can connect to each other by sending messages via serial monitor. Is it possible and may I know where I can get the code?

Hi @Farisabdurrasid. To get this correctly, you want to use Zigbee protocol to send a messages between 2 devices (coordinator <-> end device) via serial monitor? Idk if this is the purpose of the Zigbee library and overall the Zigbee HA devices. What kind of messages it should be? Or can you explain it more deeply? Thanks

Farisabdurrasid, 2025-01-08:
yes like that, without having to use Home Assistant, so these 2 ESP32C6
devices communicate with each other using zigbee.

For the project I am planning to create, the ESP32-C6 (coordinator) will
act as the MCU for the soil moisture sensor. When the soil moisture level
reaches a set point, another ESP32-C6 (end device) will act as the MCU to
change the traffic light's color. The data from the coordinator will be
sent to the end device using Zigbee communication. can it?

Farisabdurrasid, 2025-01-08:
> Hi @Farisabdurrasid. To get this correctly, you want to use Zigbee protocol to send a messages between 2 devices (coordinator <-> end device) via serial monitor? Idk if this is the purpose of the [...]

@P-R-O-C-H-Y yes like that, without having to use Home Assistant, so these 2 ESP32C6 devices communicate with each other using zigbee.

For the project I am planning to create, the ESP32-C6 (coordinator) will act as the MCU for the soil moisture sensor. When the soil moisture level reaches a set point, another ESP32-C6 (end device) will act as the MCU to change the traffic light's color. The data from the coordinator will be sent to the end device using Zigbee communication. can it?

teambvd, 2025-01-23:
> For the project I am planning to create, the ESP32-C6 (coordinator) will act as the MCU for the soil moisture sensor. When the soil moisture level reaches a set point, another ESP32-C6 (end device) [...]

Sounds like what you're looking for is something more like ESP-NOW. Any to any comms isn't really what ZigBee does, and trying to do it by working within the ZigBee protocol is just going to unnecessarily complicate implementing what you're looking for IMO.

Farisabdurrasid, 2025-01-23:
Yes, it makes sense.

Thanks for the knowledge

MegaJack22, 2025-02-05:
Hey PROCHY,
First off, I just wanted to say a huge thanks for everything you’ve done for the community. Your work on this project is absolutely amazing, and it’s clear how much effort and time you’ve put into it. It’s inspiring to see someone so dedicated to making things better for everyone else. Your contributions have helped so many people, including me, to build some really cool stuff.
Now, I had a quick question for you. Do you think it would be possible to implement Zigbee OTA updates through systems like Z2M or similar platforms? This could add even more flexibility to projects built using your library, allowing remote firmware updates without needing physical access to devices. What are your thoughts on that?
Thanks again for all your hard work—it truly makes a difference!

P-R-O-C-H-Y, 2025-02-06:
Hi everyone, can you please try #10946 if the OTA works for you? Thanks

quentains, 2025-02-09:
Wow. I bought an ESP32-H2 8 months ago to try out to build Zigbee sensors. I had to deal with the ESP-IDF documentation for several days before finally giving up. Now I'm back to the news and discovering this new arduino library. Thank you very much for your great work. It will clearly help the most beginners among us to get started! Has the PR about documentation already been created?

P-R-O-C-H-Y, 2025-02-10:
Hi @quentains, the documentation is unfortunately not ready yet. I have a quick draft, but as I am still working on adding new stuff, this is on lower priority now. Most of the functionality is demonstrated in the examples, so you can for now take a look at them :)

[40 more replies]

PR #11525 (2025-07-21): feat(docs): Add Zigbee library API documentation
## Description of Change
This pull request adds documentation for new Zigbee library, expanding the API coverage.

* **Added Zigbee APIs to Table of Contents**: Included a new section in `libraries.rst` that organizes Zigbee-related APIs under their own subsection, improving navigation and discoverability.

## Tests scenarios

## Related links

Closes #10135