espressif/arduino-esp32 #10140: [send_ssl_data():382]: (-80) UNKNOWN ERROR CODE (0050)

samvandenabeele, 2024-08-11:
### Board

XIAO ESP32-S3

### Device Description

SENSE module

### Hardware Configuration

no other hardware

### Version

v3.0.4

### IDE Name

Platformio

### Operating System

Windows 11

### Flash frequency

40Mhz (not sure)

### PSRAM enabled

yes

### Upload speed

115200

### Description

When i try to send an image to my server, i get an error
`Connection successful!
[181711][E][ssl_client.cpp:37] _handle_error(): [send_ssl_data():382]: (-80) UNKNOWN ERROR CODE (0050)`
The issue has been made many times, but it alway's says it has been fixed 2 to 3 years ago

### Sketch

```cpp
WiFiClientSecure client;
WiFi.mode(WIFI_STA);
WiFi.begin(ssid, password); 

if (client.connect(serverName.c_str(), serverPort)) {
    Serial.println("Connection successful!");
    String head = "--RandomNerdTutorials\r\nContent-Disposition: form-data; name=\"file\"; filename=\"esp32-cam.jpg\"\r\nContent-Type: image/jpeg\r\n\r\n";
    String tail = "\r\n--RandomNerdTutorials--\r\n";

    uint32_t imageLen = fb->len;
    uint32_t extraLen = head.length() + tail.length();
    uint32_t totalLen = imageLen + extraLen;
  
    client.print("POST " + serverPath + " HTTP/1.1" + "\r\n");
    client.print("Host: " + serverName + "\r\n");
    client.print("Content-Length: " + String(totalLen) + "\r\n");
    client.print("Content-Type: multipart/form-data; boundary=RandomNerdTutorials\r\n");
    client.println();
    client.print(head);
  
    uint8_t *fbBuf = fb->buf;
    size_t fbLen = fb->len;
    for (size_t n=0; n<fbLen; n=n+1024) {
      if (n+1024 < fbLen) {
        client.write(fbBuf, 1024);
        fbBuf += 1024;
      }
      else if (fbLen%1024>0) {
        size_t remainder = fbLen%1024;
        client.write(fbBuf, remainder);
      }
    }   
    client.print(tail);
```

### Debug Message

```plain
_handle_error(): [send_ssl_data():382]: (-80) UNKNOWN ERROR CODE (0050)
```

### Other Steps to Reproduce

I tried sending the image with Postman, and that works fine, so i don't expect a server-side error

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2024-08-11:
Cryptography has a lot of moving parts that can break, so what was fixed 2 years ago may not be your problem.  Please set core debug level to verbose, upload the code, and post the whole log of the transaction.  I don't see where you are providing a server certificate, so that is the most likely error.

samvandenabeele, 2024-09-07:
I have fixed it. The file was to large and a time-out was issues.

closed 2024-09-07 by samvandenabeele