espressif/arduino-esp32 #10141: WiFi not connecting, but release 2.0.0 works

rbdannenberg, 2024-08-11:
### Board

Sparkfun ESP32 Thing

### Device Description

Just USB to Windows PC for development and serial monitor

### Hardware Configuration

No

### Version

latest master (checkout manually)

### IDE Name

Arduino 1.8.15

### Operating System

Windows 10

### Flash frequency

80 mHz

### PSRAM enabled

yes

### Upload speed

921600

### Description

I'm trying to connect to an old Westell A90-750015-07 WiFi router (disconnected from ethernet or DSL). This works with as late as library version 2.0.0, but after that, including 3.0.3,  I get WL_NO_SSID_AVAIL from WiFi.status(). I can connect to this WiFi network from my Mac without problem, and I've tried this on 2 different Sparkfun ESP32 Thing boards. I'm running identical code and only changing the library. I also tried the old library on Arduino IDE 2.3.2, but I get the error "Invalid FQBN: getting build properties for board esp32:esp32:esp32thing: invalid option 'EraseFlash'. Of course I tried the current library 3.0.3 with the current Arduino IDE 2.3.2, but I get the same WL_NO_SSID_AVAIL problem.

### Sketch

```cpp
#include "WiFi.h"
#include <lwip/sockets.h>
#include <lwip/netdb.h>

const int LED_PIN = 5;

const char *HOSTNAME = "esp";
const char *networkName = "Aylesboro";
const char *networkPswd = "----------";

int udp_recv_sock = -1;
int udp_recv_port = 8007;

void printLine()
{
    Serial.println();
    for (int i = 0; i < 30; i++)
        Serial.print("-");
    Serial.println();
}

void connectToWiFi(const char * ssid, const char * pwd)
{
    int ledState = 0;

    printLine();
    Serial.println("Connecting to WiFi network: " + String(ssid));
    WiFi.begin(ssid, pwd);
    WiFi.setHostname(HOSTNAME);
  
    while (WiFi.status() != WL_CONNECTED) {
        // Blink LED while we're connecting:
        digitalWrite(LED_PIN, ledState);
        ledState = (ledState + 1) % 2; // Flip ledState
        delay(500);
        Serial.print("status: ");
        Serial.print(WiFi.status());
        Serial.print(" WL_CONNECTED ");
        Serial.println(WL_CONNECTED);
    }

    Serial.println();
    Serial.println("WiFi connected!");
    Serial.print("IP address: ");
    Serial.println(WiFi.localIP());
}

void setup() {
    // Initilize hardware:
    Serial.begin(115200);
    Serial.println("this is getip to test getting local IP");
    connectToWiFi(networkName, networkPswd);

    /* code omitted here that opens a socket for UDP */
}

void loop() {
    // put your main code here, to run repeatedly:
}
```

### Debug Message

```plain
Output is "status: 1 WL_CONNECTED 3"

[note: expected output and output using library v2.0.0 is "status: 3 WLCONNECTED 3"
 "status: 1" indicates WL_NO_SSID_AVAIL.]
```

### Other Steps to Reproduce

Behavior seems to be router-dependent. I can connect to my normal home WiFi, but the router definitely works and ESP32 connects to it using v2.0.0 or earlier.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Jason2866, 2024-08-11:
Does the WiFi Router maybe not support WPA/WPA2?

rbdannenberg, 2024-08-11:
I don't think it supports WPA/WPA2. Are you suggesting WEP was removed from the ESP32 WiFi implementation after around v2.0.0? Because 2.0.0 and earlier seem to work fine. My MacOS can connect as well.

Jason2866, 2024-08-11:
Yes, see https://github.com/espressif/arduino-esp32/issues/9559#issuecomment-2075488024

rbdannenberg, 2024-08-12:
Thanks for the quick reply. That solved the problem. I didn't realize the Arduino API and the esp32-arduino API are different (although mostly compatible), nor that there is separate documentation. I probably would not have found this without your help, so thanks again.

SuGlider, 2024-08-12:
Great! Thanks @Jason2866 for the help. Closing the issue.

closed 2024-08-12 by SuGlider