espressif/arduino-esp32 #10142: I2S: Error message: [esp32-hal-periman.c:122] perimanSetPinBus(): Invalid pin: 255  on i2s->setPins()

nielsnl68, 2024-08-11:
### Board

M5Atom Echo.

### Device Description

M5Atom Echo.

### Hardware Configuration

static const int8_t NOT_A_I2SPIN = -1;

const uint8_t I2S_SCK = 19;
const uint8_t I2S_WS = 33;
const uint8_t I2S_DIN = 23;
const uint8_t I2S_DOUT = 22;

Connected to the internal I2S speaker IC.

### Version

v3.0.4

### IDE Name

Arduino IDE

### Operating System

win10

### Flash frequency

80Mhz

### PSRAM enabled

no

### Upload speed

115200

### Description

When i want to set the *setPins* i get the error message 
> [esp32-hal-periman.c:122] perimanSetPinBus(): Invalid pin: 255
I tried to use different way's to set the value to **-1** but the "**perimanSetPinBus()**" see the not used pins as value 255. 

### Sketch

```cpp
static const int8_t NOT_A_I2SPIN = -1;

I2SClass i2s;

void setup() {
    i2s.setPins(I2S_SCK, I2S_WS, I2S_DOUT, NOT_A_I2SPIN, NOT_A_I2SPIN);
    if (!i2s.begin(I2S_MODE_STD, 16000, I2S_DATA_BIT_WIDTH_16BIT, I2S_SLOT_MODE_MONO, I2S_STD_SLOT_RIGHT)) {
        Serial.println("Failed to initialize I2S Speaker bus!");
        return;
    }
}

void loop()
```

### Debug Message

```plain
[ 27240][E][esp32-hal-periman.c:122] perimanSetPinBus(): Invalid pin: 255
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2024-08-11:
I don't have a PICO-D4 to test on, but I am unable to reproduce on standard esp32.  Please set core debug level to verbose and post the log (the stuff in setup anyhow) .
In the future, please post minimal code that can be copy/pasted so we can be running exactly the same code you are.

nielsnl68, 2024-08-11:
> I don't have a PICO-D4 to test on, but I am unable to reproduce on standard esp32. Please set core debug level to verbose and post the log (the stuff in setup anyhow) . In the future, please post minimal code that can be copy/pasted so we can be running exactly the same code you are.

I updated the sketch a little bit. So you can now test it as is. Sadly i tried the script above and i can't reproduce it as well. I tried to follow the steps from my own sketch and that did not resulted in the above error as well.
So there must be something else going one.

The issue is also shown on different devices. It looks as if an *int8_t* is pressed into a *uint8_t* making the -1 into a 255; 

As i said above, i cant reproduce using the script above as well. I'm going to look further until i can reproduce it. When you have any suggestion on how to search for the issue all input is welcome. The error message does not tell which pin has the "255" value.

Anyway thanks for trying to reproduce the error. I will comeback to this soon.

P-R-O-C-H-Y, 2025-01-09:
I am closing this issue as it's solved in the latest versions. Was not able to reproduce with 3.1.1.