espressif/arduino-esp32 #10145: ESP32 S3 I2C issues when working as slave

JorgeMALopes, 2024-08-12:
### Board

ESP32 S3 wroom 1

### Device Description

Just an USB port, and a TI battery monitor.

### Hardware Configuration

Currently only working with just the CPU, but has a TI battery monitor connected to the other I2C port.

### Version

v3.0.3

### IDE Name

Arduino IDE

### Operating System

windows 10

### Flash frequency

80MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

When i connect an ESP32S3 as an I2C slave to an ESP32 or a drone flight controller, as a master, the data comes out wrong.
When i reverse the code and the S3 is the master and the ESP32 is the slave then the data comes out correct.

I made a simple sketch where the slave returns 4 bytes when there is a request, 0x01FF0306 or 0x01FF03FF. The master requests a variable amount of bytes, sometimes more, sometimes less. When the S3 is the slave the first byte received from it is the slave address followed by part of the slave data. When the ESP32 is the slave then only data is returned.

I have asked a friend to lend me his oscilloscope so i can check the data lines for each case.

These are the lines for the S3 as a slave, notice the long time after the request and the repetition of the slave address.
[image]
[image]
[image]

This is the lines for ESP32 as slave, the delay is minimal and there is no repetition of the slave address.
[image]

The most stable case was then the returned bytes where of the same quantity as the requested ones, but sometimes the device address appears in the beginning and the only way to fix it is to reset the S3 which is not a viable solution.

I have tested with wire and wire1 and tested with other connections on the S3.

### Sketch

```cpp
-----------------------------------Slave code:-----------------------------

#include "Wire.h" 						// i2c library

#define I2C_DEV_ADDR 0x30			// i2c adress, needs to be the same as in the lua script

int opcode = 0;							// received operation code
int i2cCommand = 0;						// received i2c command

typedef union Dados_I {  				// union for sending int via i2c
[... 226 lines cut ...]
=========== After Setup Start ============
INTERNAL Memory Info:
------------------------------------------
  Total Size        :   396660 B ( 387.4 KB)
  Free Bytes        :   358416 B ( 350.0 KB)
  Allocated Bytes   :    33052 B (  32.3 KB)
  Minimum Free Bytes:   353320 B ( 345.0 KB)
  Largest Free Block:   327668 B ( 320.0 KB)
------------------------------------------
GPIO Info:
------------------------------------------
  GPIO : BUS_TYPE[bus/unit][chan]
  --------------------------------------  
    14 : I2C_SLAVE_SDA[0]
    19 : USB_DM
    20 : USB_DP
    21 : I2C_SLAVE_SCL[0]
    43 : UART_TX[0]
    44 : UART_RX[0]
============ After Setup End =============
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

JorgeMALopes, 2024-09-09:
[image]
[image]

I got a pc based logic analyzer and an ESP32 S3 devkitc devboard and using the same code the same issue happens.

me-no-dev, 2024-09-09:
The ESP32's Slave works differently than all other chips we have. ESP32 needs data to be PRELOADED before the request is received. Fore reference are [these lines](https://github.com/espressif/arduino-esp32/blob/master/libraries/Wire/examples/WireSlave/WireSlave.ino#L28-L32) in the slave example. That is why there is no delay between the address and the data. All other chips wait for the request to be handled and data written to the fifo, which is then returned.
In the code above, ESP32 returns the previous `Wire.write`.

JorgeMALopes, 2024-09-09:
The issue i am having is with the ESP32-S3 as a slave.
The first byte that the S3 slave sends is the slave address and not the data that was requested and it is trying to write with 
`Wire.write(batata.b, 4);`  in the `onRequest()`.
This happens even when the master is an STM arm based flight controller.

This is also happening if i program the S3 with IDF.

me-no-dev, 2024-09-10:
I tried the code. the only change I made was to write two bytes, because that is how many you read in your master sketch and the result is as expected 
```
requestFrom: 2
0x06, 0x03, // ..
```

JorgeMALopes, 2024-09-10:
Thank you. 
I have noticed that works in this case. The code i put here is a very simplified version of what i am making and in that sometimes the data comes clean and sometimes it comes contaminated with a random byte. What confuses me is that the random byte that appears is the slave address byte. Is this a normal thing?

As a further experiment into this, if you request 5 bytes and the slave writes 4 bytes the response fluctuates:

`requestFrom: 5
0x61, 0x61, 0x06, 0x03, 0x03, // aa...
requestFrom: 5
0x06, 0x03, 0x03, 0x03, 0x03, // .....
requestFrom: 5
0x06, 0x03, 0x03, 0x03, 0x03, // .....
requestFrom: 5
0x06, 0x03, 0x03, 0x03, 0x03, // .....
requestFrom: 5
0x06, 0x03, 0x03, 0x03, 0x03, // .....
requestFrom: 5
0x06, 0x03, 0x03, 0x03, 0x03, // .....
requestFrom: 5
0x06, 0x03, 0x03, 0x03, 0x03, // .....
requestFrom: 5
0x06, 0x03, 0x03, 0x03, 0x03, // .....
requestFrom: 5
0x06, 0x03, 0x03, 0x03, 0x03, // .....
requestFrom: 5
0x61, 0x61, 0x61, 0x61, 0x61, // aaaaa
requestFrom: 5
`

Why is the slave address appearing in the data?

me-no-dev, 2024-09-10:
where is it appearing? the slave address is `0x30` not `0x03`

me-no-dev, 2024-09-10:
Please make sure that you write as many bytes as will be read. That is important, else you are reading random bytes in the fifo

JorgeMALopes, 2024-09-10:
The slave address is 0x30, if you bit shift it left once and put in the read bit it becomes 0x61. This follows when i change the slave address, 0x10 becomes 0x21. This is the data that is poluting the response.  
Here is a capture of the data, notice that the first byte that is received as data is the same as the address byte plus the read bit.
[image]

Also I got it to get that random byte to appear even when i request the same size of data that the slave is writing.

[image]

I did one request for 1 byte, uploaded to the esp32 master and then changed the code and requested 4 bytes and uploaded to esp32 master, now it is locked in this situation. It comes out of this situation when i reset the S3.

Also it is kind of weird that if i place 4 known bytes in the fifo and read 3 bytes the first byte is the device address bit shifted left once plus the read bit.

JorgeMALopes, 2024-09-10:
After some playing around i noticed that if the ESP32 S3 slave does not write to the fifo then what is read by the ESP32 master is the left bit shifted address of the slave + 1. 

[image]

Does this mean that on the previous examples the ESP32 S3 slave is to slow to populate the fifo?

I hope i am not being annoying, am truly trying to figure this out. 
I am trying to avoid having to replace the ESP32 S3 with something else that is more stable. 

Thanks for all the help.

me-no-dev, 2024-09-10:
You must read as many bytes as you wrote. That is all. No other tricks are necessary

JorgeMALopes, 2024-09-10:
I will try that. Thank you.

JorgeMALopes, 2024-09-10:
Same issue happens irregularly.

Master code:
`#include "Wire.h"

#define I2C_DEV_ADDR 0x30

typedef union Dados_I {  				// union for sending int via i2c
  int i;
  byte b[4];
};

typedef union Dados_F {					// union for sending float via i2c
  float f;
  byte b[4];
};

typedef union Dados_Bool { 				// union for sending a boolean via i2c
  bool bo;
  byte b[1];
};

// values that are requested from the smart battery
const int req_health = 1;				// battery health		
const int req_voltage = 2;				// total battery voltage
const int req_cell_count = 3;			// cell count
const int req_rem_capacity_pct = 4;		// remaining capacity in percentage
const int req_cycle_count = 5;			// cycle count
const int req_current = 6;				// battery current
const int req_consumed_mah = 7;			// consumed capacity in mah
const int req_consumed_wh = 8;			// consumed capacity in wh
const int req_temperature = 9;			// battery temperature
const int req_cell_voltage_1 = 11;		// cell voltages for each of the cells
const int req_cell_voltage_2 = 12;
const int req_cell_voltage_3 = 13;
const int req_cell_voltage_4 = 14;
const int req_cell_voltage_5 = 15;
const int req_cell_voltage_6 = 16;
const int req_cell_voltage_7 = 17;
const int req_cell_voltage_8 = 18;
const int req_cell_voltage_9 = 19;
const int req_cell_voltage_10 = 20;
const int req_cell_voltage_11 = 21;
const int req_cell_voltage_12 = 22;
const int req_cell_voltage_13 = 23;
const int req_cell_voltage_14 = 24;
const int req_cell_voltage_15 = 25;
const int req_cell_voltage_16 = 26;
const int req_cell_voltage_17 = 27;
const int req_cell_voltage_18 = 28;
const int req_cell_voltage_19 = 29;
const int req_cell_voltage_20 = 30;
const int req_cell_voltage_21 = 31;
const int req_cell_voltage_22 = 32;
const int req_cell_voltage_23 = 33;
const int req_cell_voltage_24 = 34;

Dados_Bool health;					// local variable where the batttery health is stored
Dados_F voltage;					// local variable where the batttery voltage is stored (V)
Dados_I cell_count;					// local variable where the batttery cell count is stored 
Dados_I rem_capacity_pct;			// local variable where the batttery capacity is stored (%)
Dados_I cycle_count;				// local variable where the batttery cycle count is stored 
Dados_F current;					// local variable where the batttery current count is stored (A)
Dados_F consumed_mah;				// local variable where the batttery consumed mah is stored (mah)
Dados_F consumed_wh;				// local variable where the batttery consumed wh is stored (wh)
Dados_F temperature;				// local variable where the batttery temperature is stored (c)
Dados_I cell_voltage_1;				// local variables where the batttery cell voltages are stored (mv)
Dados_I cell_voltage_2;
Dados_I cell_voltage_3;
Dados_I cell_voltage_4;
[... 430 lines cut ...]
    case req_cell_voltage_24:
      Wire.write(cell_voltage_24.b,4);
      break;

    default:
      break;
  }
}

//void onRequest() { 					// on request function, sends data via i2c depeding on the request
// //Wire.write(batata.b, 4);
// Serial.println("request");
//}
`
What i should be getting is:
[image]

I am getting this:
[image]

Notice the decimal 97 =0x61 when i request the health byte and the 768097 value, if you convert it into hex you get 0x0B, 0xB8 and 0x61, the value that is the slave adress bit shifted left + 1.

There are no tricks, I send the same amount of bytes that are requested.

It fixes itself if i reset the slave. This is not optimal.

JorgeMALopes, 2024-09-10:
I don't know why the code breaks when i paste it into the code tags. I'm sorry for that.

me-no-dev, 2024-09-11:
I have slightly revised your example above. Mostly prints and it's functionally the same. I use the same pins and all.
[image]

me-no-dev, 2024-09-11:
One suggestion for your use case. Use `Wire.slaveWrite(data, len)` instead of `Wire.write(data, len)`. Then you can load the values directly in the `onReceive` callback. That will make communication faster too. Also do not print inside the callbacks. It takes time and slows things down.
```cpp
void onReceive(int len) {     // on receive function, used to update local variables or activate local functions
  i2cCommand = 0;             // resets the command
  opcode = Wire.read();       // reads the operation code
  if (len == 2) {             // if there are 2 bytes
    i2cCommand = Wire.read(); // the second byte is the command
  }
  loadValues();               // load the values here
}

void onRequest() {
}

void loadValues(){
  switch (opcode) {
    case req_health:
      Wire.slaveWrite(health.b, 1);
      break;

    case req_voltage:
      Wire.slaveWrite(voltage.b, 4);
      break;
 ............
```

JorgeMALopes, 2024-09-11:
Wonderful. Thank you.
I will try it now.

me-no-dev, 2024-09-11:
@JorgeMALopes any news?

JorgeMALopes, 2024-09-12:
It worked with the code as is. But stops working when i add the rest.
In the device i am making the ESP32S3 is an interpreter for a BQ769142. The S3 controls and gets the data from the BQ769142, via I2C as a master, does some simple calculations and then feeds it via I2C to a flight controller as slave.

As soon as i add the BQ769142 code to it the communication with the flight controller breaks.
I am going to desolder the S3 from the board and wire up a standard ESP32 and try that. If i have success i will inform. 
For now i think the S3 is not a viable device for an i2c slave.

JorgeMALopes, 2024-09-12:
It worked with the code as is. But stops working when i add the rest.
In the device i am making the ESP32S3 is an interpreter for a BQ769142. The S3 controls and gets the data from the BQ769142, via I2C as a master, does some simple calculations and then feeds it via I2C to a flight controller as slave.

As soon as i add the BQ769142 code to it the communication with the flight controller breaks.
I am going to desolder the S3 from the board and wire up a standard ESP32 and try that. If i have success i will inform. 
For now i think the S3 is not a viable device for an i2c slave.

Thanks for all the help.

JorgeMALopes, 2024-09-12:
Still no success with the regular ESP32. 
I have a question but i am unsure if i should create a new thread elsewhere or ask here so i can figure out how to fix this or if this is fixable.

Can an ESP32 be an i2c master on one bus and obtain data from a bunch of sensors, do some calculations and then have it available as a slave on a separate bus? 

[image]

me-no-dev, 2024-09-13:
yes it can

JorgeMALopes, 2024-09-13:
Ok, going to try and figure that out.
Thanks for everything.

JorgeMALopes, 2024-09-13:
Figured it out. Tried it with a S3, it works.

Modified my main code, did not work with either esp32 or S3. Start hunting down the culprit. Found it. It was trying to write to a position outside of an array. 

There is no bug in the I2C bus on the S3, it just behaves differently when compared to the regular ESP32, but still works fine as long as you ask as many bytes as you write on the other side.

I think i can finish the rest of the code.

Thanks for all the help!!

closed 2025-01-10 by rftafas