espressif/arduino-esp32 #10146: no port arduino ide 1.18.9

BimoSora2, 2024-08-12:
### Board

esp32 dev kitv1

### Device Description

ESP32 Devkit V1

### Hardware Configuration

ex

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE 1.18.9

### Operating System

Linux

### Flash frequency

40 Mhz

### PSRAM enabled

yes

### Upload speed

15520

### Description

I just updated the ESP32 board to 3.0.2, when the update was finished, why couldn't the Arduino Ide 1.x.x find the port but the Arduino Ide 2.x.x could find it?
https://github.com/user-attachments/assets/3ee9ab4f-4376-4657-9d6f-f6e70f7e6e38

### Sketch

```cpp
/*
  Blink without Delay

  Turns on and off a light emitting diode (LED) connected to a digital pin,
  without using the delay() function. This means that other code can run at the
  same time without being interrupted by the LED code.

  The circuit:
  - Use the onboard LED.
  - Note: Most Arduinos have an on-board LED you can control. On the UNO, MEGA
    and ZERO it is attached to digital pin 13, on MKR1000 on pin 6. LED_BUILTIN
    is set to the correct LED pin independent of which board is used.
    If you want to know what pin the on-board LED is connected to on your
    Arduino model, check the Technical Specs of your board at:
    https://www.arduino.cc/en/Main/Products

  created 2005
  by David A. Mellis
  modified 8 Feb 2010
  by Paul Stoffregen
  modified 11 Nov 2013
  by Scott Fitzgerald
  modified 9 Jan 2017
  by Arturo Guadalupi

  This example code is in the public domain.

  https://www.arduino.cc/en/Tutorial/BuiltInExamples/BlinkWithoutDelay
*/

// constants won't change. Used here to set a pin number:
const int ledPin =  LED_BUILTIN;// the number of the LED pin

// Variables will change:
int ledState = LOW;             // ledState used to set the LED

// Generally, you should use "unsigned long" for variables that hold time
// The value will quickly become too large for an int to store
unsigned long previousMillis = 0;        // will store last time LED was updated

// constants won't change:
const long interval = 1000;           // interval at which to blink (milliseconds)

void setup() {
  // set the digital pin as output:
  pinMode(ledPin, OUTPUT);
}

void loop() {
  // here is where you'd put code that needs to be running all the time.

  // check to see if it's time to blink the LED; that is, if the difference
  // between the current time and last time you blinked the LED is bigger than
  // the interval at which you want to blink the LED.
  unsigned long currentMillis = millis();

  if (currentMillis - previousMillis >= interval) {
    // save the last time you blinked the LED
    previousMillis = currentMillis;

    // if the LED is off turn it on and vice-versa:
    if (ledState == LOW) {
      ledState = HIGH;
    } else {
      ledState = LOW;
    }

    // set the LED with the ledState of the variable:
    digitalWrite(ledPin, ledState);
  }
}
```

### Debug Message

```plain
I don't know, the port couldn't be found
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2024-08-12:
On original esp32, this should have nothing to do with the IDE.  Since you are in linux, you should be able to see the device on a usb port by calling `lsusb` from a terminal. If you don't see "Silicon Labs CP210x UART Bridge" (or maybe a CH340) in the list, then you have a physical connection problem. In most cases, connection problems with esp32 is due to a problem with the usb cable.  Try a different cable, try a different usb port.

BimoSora2, 2024-08-15:
> On original esp32, this should have nothing to do with the IDE. Since you are in linux, you should be able to see the device on a usb port by calling `lsusb` from a terminal. If you don't see [...]

Hi guys, I have tried 3 cables and 4 esp32 all the same in Arduino IDE 1.18.9 the port is not read. When I downgrade the board to 3.0.3 the port is read, that's what I'm confused about

[image]
[image]

SuGlider, 2024-08-16:
>Hi guys, I have tried 3 cables and 4 esp32 all the same in Arduino IDE 1.18.9 the port is not read. When I downgrade the board to 3.0.3 the port is read, that's what I'm confused about

I'll try to test it using Linux.

BimoSora2, 2024-08-16:
> > Hi guys, I have tried 3 cables and 4 esp32 all the same in Arduino IDE 1.18.9 the port is not read. When I downgrade the board to 3.0.3 the port is read, that's what I'm confused about
> 
> I'll try to test it using Linux.

ok

lbernstone, 2024-08-22:
Please open a terminal and run `lsusb` when the machine is in a state where it does not recognize the port. It is just hard to understand how the version of arduino-esp32 could have any effect on this, as it is an OS-level/IDE issue.  If you switch to an Arduino AVR board (eg Arduino Uno), you should get the same port showing up (/dev/ttyUSB0 or ACM0).
You are certain this is an ESP32 Devkit V1 (maybe post a picture of the module markings)? What distro of linux is this?

BimoSora2, 2024-08-23:
> Please open a terminal and run `lsusb` when the machine is in a state where it does not recognize the port. It is just hard to understand how the version of arduino-esp32 could have any effect on [...]

there is a strange thing happening, the port is not being read so I can't use serial monitor on esp 1.8.9 but I can upload the code

Parrot OS Based Debian
[image]
[image]

closed 2025-03-12 by Jason2866