espressif/arduino-esp32 #10152: I would like to reopen the issue. Encountered this again. Please fix. @SuGlider

khryssmar, 2024-08-13:
I would like to reopen the issue. Encountered this again. Please fix. @SuGlider 
 
[image]
[image]

_Originally posted by @khryssmar in https://github.com/espressif/arduino-esp32/issues/9500#issuecomment-2284217892_

lbernstone, 2024-08-13:
Please fill out the issue details.  This looks like it is still correct in 2.0.17 and v2.x branch.  The change in v3.0 is intentional.

Jason2866, 2024-08-13:
Which issue in general? Some defaults are changed for Core 3.0.x as designed.
Since UART IOs can freely assigned to nearly any GPIO just change as you want.

SuGlider, 2024-08-13:
@khryssmar - this is intentional and it is documented in https://docs.espressif.com/projects/arduino-esp32/en/latest/migration_guides/2.x_to_3.0.html#id14

In case that the sketch needs to use other pins, `Serial1` and `Serial2` can change their pins by using `Serial1.begin(baud_rate, mode, rxPin, txPin)` or by `Serial2.setPins(rxPin, txPin)`.
Setting the pin to `-1` will force it to keep the default or to keep it unchanged.
Example: `Serial2.setPins(-1, 17);` will only change the TX2 pin to 17.

The problem with the former ESP32 `Serial2` pins is that those are used to control the PSRAM within the ESP32-WROVER module. In case that the sketch uses PSRAM, it would crash its execution.

closed 2024-08-13 by SuGlider

SuGlider, 2024-08-13:
For the records:

This pins change is valid for Arduino Core 3.x.x. But it was commited by mistake into 2.0.15 -- 2.0.17.
Explanation in https://github.com/espressif/arduino-esp32/issues/10152#issuecomment-2286991672

------

Arduino Core 2.0.15, 2.0.16 and 2.0.17 had a mistake using new pins.
Arduino Core 2.0.14 and lower use the original pins.

Arduino Core from `release/v2.x` branch uses the same pins as before:
Fixed with https://github.com/espressif/arduino-esp32/pull/9502

SuGlider, 2024-08-13:
Arduino Core branch `release/v2.x` will continue to receive new contributions as usual.
But it is very improbable that we will release a 2.0.18 version.

It is possible to create a `sketchboard` from a cloned branch, like `release/v2.x`
For that, please follow the instructions from 
https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html#windows-manual-installation