espressif/arduino-esp32 #10158: Program uploading issue in ESP32-S3 in custom 4 layer PCB

hardikofficial963, 2024-08-14:
### Board

ESP32-S3

### Device Description

ESP32-S3 based PCB  

### Hardware Configuration

TX and RX are connected to CH340 USB to UART chip.

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

80Mhz

### PSRAM enabled

yes

### Upload speed

115200

### Description

I am working on a PCB design based on the ESP32-S3. In our initial prototype, we used a 2-layer PCB, and it worked fine.

For the next iteration, we upgraded to a 4-layer PCB, where we added additional Power and GND planes. However, we're experiencing issues when uploading code to the device.

Could the problem be related to using the same capacitor and resistor values in the RC circuits on the Enable pin and IO0 pin that we used for the 2-layer PCB? 

Please advise on how we can address this issue. I am sharing the schematic below.

[image]

### Sketch

```cpp
void setup()
{

}

Void loop()
{

}
```

### Debug Message

```plain
Sketch uses 268405 bytes (20%) of program storage space. Maximum is 1310720 bytes.
Global variables use 18808 bytes (5%) of dynamic memory, leaving 308872 bytes for local variables. Maximum is 327680 bytes.
"C:\Users\hardi\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\4.5.1/esptool.exe" --chip esp32s3 --port "COM7" --baud 921600  --before default_reset --after hard_reset write_flash -e -z --flash_mode dio --flash_freq 80m --flash_size 4MB 0x0 "C:\Users\hardi\AppData\Local\Temp\arduino\sketches\15D75FF90C55FB9F0F99D7202D1A3FE4/BlinkBinary.ino.bootloader.bin" 0x8000 "C:\Users\hardi\AppData\Local\Temp\arduino\sketches\15D75FF90C55FB9F0F99D7202D1A3FE4/BlinkBinary.ino.partitions.bin" 0xe000 "C:\Users\hardi\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.16/tools/partitions/boot_app0.bin" 0x10000 "C:\Users\hardi\AppData\Local\Temp\arduino\sketches\15D75FF90C55FB9F0F99D7202D1A3FE4/BlinkBinary.ino.bin" 
esptool.py v4.5.1
Serial port COM7
Connecting...............................

A serial exception error occurred: Cannot configure port, something went wrong. Original message: PermissionError(13, 'Access is denied.', None, 5)
Note: This error originates from pySerial. It is likely not a problem with esptool, but with the hardware connection or drivers.
For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html
Failed uploading: uploading error: exit status 1
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2024-08-14:
Have you changed anything between 2 layer and 4 layer board? Have you tried entering download mode manually? I do not see IO0 being pulled up?

hardikofficial963, 2024-08-14:
I have not change anything in 4 layer, Yes I have tried entering download mode manually, it is working fine.

me-no-dev, 2024-08-14:
Do you have a pullup on IO0? And 10uF cap on IO0 or EN is a bit too much. 100nF is enough. 10K pullup is also OK

Parsaabasi, 2025-03-11:
Hi @hardikofficial963 
Do you still need assistance with this?

lbernstone, 2025-03-11:
https://docs.espressif.com/projects/esp-hardware-design-guidelines/en/latest/esp32s3/schematic-checklist.html#chip-power-up-and-reset-timing