espressif/arduino-esp32 #10161: HardwareSerial at 460800 BAUD not working

mettatec, 2024-08-14:
### Board

ESP32 Dev Module

### Device Description

Custom hardware

### Hardware Configuration

GPIO 16 and 17 used as UART HardwareSerial

### Version

v3.0.4

### IDE Name

Visual Studio Code

### Operating System

Windows 11

### Flash frequency

80MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

HardwareSerial at 460800 does not read data correctly when using ESP32-Arduino Core v3. 

### Sketch

```cpp
#include <HardwareSerial.h>

HardwareSerial MySerial(1);
const int MySerialRX = 16;
const int MySerialTX = 17;

uint8_t rBuffer[2048]; 

void setup() 
{
    MySerial.begin(230400, SERIAL_8N1, MySerialRX, MySerialTX);
    Serial.begin(115200);
}

void loop() 
{
    while (MySerial.available() > 0) {
        int s = MySerial.readBytes(rBuffer, sizeof(rBuffer));
          Serial.write(rBuffer, s);
    }
}
```

### Debug Message

```plain
ets Jul 29 2019 12:21:46
[... 235 lines cut ...]

---------------------------------
Correct Messages should be:
$GNGST,,0.0000,9189718,9189718,0.0000,3754429,3754429,3754429*73
$GNRMC,,V,,,,,,,,,,N,V*37
$GNGGA,,,,,,0,00,99.99,,,,,,*56
$GNGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99,1*33
$GNGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99,2*30
$GNGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99,3*31
$GNGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99,4*36
$GNGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99,5*37
$GPGSV,1,1,00,0*65
$GLGSV,1,1,00,0*79
$GAGSV,1,1,00,0*74
$GBGSV,1,1,00,0*77
$GQGSV,1,1,00,0*64
```

### Other Steps to Reproduce

Tried with ESP32-Arduino Core v2 and it works OK.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

mettatec, 2024-08-14:
Tried already. Same result

me-no-dev, 2024-08-14:
I know for certain that the baudrates work fine. You can try with running Serial at that baud instead of 115200. Is it possible that your GPS requires maybe different UART config? Parity or Stop bits?

mettatec, 2024-08-14:
Same code with esp32 core V2 works well.

me-no-dev, 2024-08-14:
@SuGlider PTAL

mettatec, 2024-08-15:
The program works OK at 230400 baud. Is there any change in the clock source for UART in new core v3?

SuGlider, 2024-08-15:
> The program works OK at 230400 baud. Is there any change in the clock source for UART in new core v3?

Yes, for 3.0.4 we changed the UART clock source to 2MHz (REF_TICK), which seems not to work correctly for 4060800.

There is a test for the baud rate. When it is higher than 1 Mbps, it uses this clock source.
I'll post a PR limiting it to 250 Kbps for the ESP32 and ESP32-S2, higher than that it shall use APB as clock source.

mettatec, 2024-08-15:
I did the chage from the PR post (core 3.0.4 in my local files) and it works OK now. Thanks!

PR #10172 (2024-08-16): Fix(UART): ESP32 | ESP32-S2 UART Clock Source
## Description of Change
Problem detected with ESP32 and ESP32-S2 when the baudrate goes to 460600 bps.

REF_TICK (2MHz) seem not to work properly.
Limiting the use of REF_TICK for up to 205 Kbps.

## Tests scenarios
ESP32 / S2.

```cpp
#include <HardwareSerial.h>

HardwareSerial MySerial(1);
const int MySerialRX = 16;
const int MySerialTX = 17;

uint8_t rBuffer[2048]; 

void setup() 
{
    MySerial.begin(460800, SERIAL_8N1, MySerialRX, MySerialTX);
    Serial.begin(115200);
}

void loop() 
{
    while (MySerial.available() > 0) {
        int s = M [... cut]
closed 2024-08-15 by mettatec