espressif/arduino-esp32 #10165: PPP Crash while im call PPP.end() and Call WiFi.begin(ssid,pass);

The3ven, 2024-08-15:
### Board

ESP32S3 Dev Module

### Device Description

im using esp32s3 with esp32-3.0.3 core, 16 mb internal memory, i have Quectel EC200CN atteched with gpio 4 and 5.

### Hardware Configuration

4 , 5 Quectel EC200UCN
SD card default pin

### Version

v3.0.3

### IDE Name

Arduino IDE

### Operating System

Windows 10 22H2 (OS Build 19045.4717)

### Flash frequency

QIO 80MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

my esp32s3 crash when im call WiFi.begin(ssid,pass); after PPP.end(); it crash with network _network_event_task task, idk how to fix this please help me.

### Sketch

```cpp
#include <PPP.h>
#include <WiFi.h>

String ssid = "ssid";
String pass = "pass";

int loop_cnt = 0;
bool quectelOffOnce = true;
bool quackOff = false;

#define PPP_MODEM_APN "AIRTELGPRS.COM" // For SimCard ISP
#define PPP_MODEM_PIN   NULL // or NULL

// EC200U copyed from WaveShare SIM7600 HW Flow Control
#define PPP_MODEM_RST     41 // as vishal sys it will aviliable in next rev
#define PPP_MODEM_RST_LOW true  //active HIGH
#define PPP_MODEM_TX      4
#define PPP_MODEM_RX      5
#define PPP_MODEM_RTS     -1
#define PPP_MODEM_CTS     -1
#define PPP_MODEM_FC      ESP_MODEM_FLOW_CONTROL_NONE
#define PPP_MODEM_MODEL   PPP_MODEM_GENERIC

void onEvent(arduino_event_id_t event, arduino_event_info_t info) {
    switch (event) {
    case ARDUINO_EVENT_PPP_START:
        Serial.println("PPP Started");
        break;
    case ARDUINO_EVENT_PPP_CONNECTED:
        Serial.println("PPP Connected");
        break;
    case ARDUINO_EVENT_PPP_GOT_IP:
        Serial.println("PPP Got IP");
        break;
    case ARDUINO_EVENT_PPP_LOST_IP:
        Serial.println("PPP Lost IP");
        break;
    case ARDUINO_EVENT_PPP_DISCONNECTED:
        Serial.println("PPP Disconnected");
        break;
    case ARDUINO_EVENT_PPP_STOP:
        Serial.println("PPP Stopped");
        break;
    default:
        break;
    }
}

void testClient(const char* host, uint16_t port) {
    NetworkClient client;
    if (!client.connect(host, port)) {
        Serial.println("Connection Failed");
        return;
    }
    client.printf("GET / HTTP/1.1\r\nHost: %s\r\n\r\n", host);
    while (client.connected() && !client.available());
    while (client.available()) {
        client.read();  //Serial.write(client.read());
    }

    Serial.println("Connection Success");
    client.stop();
}

void Quectelbegin() {
    // Listen for modem events
    Network.onEvent(onEvent);

    // Configure the modem
    PPP.setApn(PPP_MODEM_APN);
    PPP.setPin(PPP_MODEM_PIN);
    PPP.setResetPin(PPP_MODEM_RST, PPP_MODEM_RST_LOW, 300);
    PPP.setPins(PPP_MODEM_TX, PPP_MODEM_RX, PPP_MODEM_RTS, PPP_MODEM_CTS, PPP_MODEM_FC);

    Serial.println("Starting the modem. It might take a while!");
    PPP.begin(PPP_MODEM_MODEL);

    Serial.print("Manufacturer: ");
    Serial.println(PPP.cmd("AT+CGMI", 10000));
    Serial.print("Model: ");
    Serial.println(PPP.moduleName());
    Serial.print("IMEI: ");
    Serial.println(PPP.IMEI());

    bool attached = PPP.attached();
    if (!attached) {
        int i = 0;
        unsigned int s = millis();
        Serial.print("Waiting to connect to network");
        while (!attached && ((++i) < 600)) {
            Serial.print(".");
            delay(100);
            attached = PPP.attached();
        }
        Serial.print((millis() - s) / 1000.0, 1);
        Serial.println("s");
        attached = PPP.attached();
    }

    Serial.print("Attached: ");
    Serial.println(attached);
    Serial.print("State: ");
    Serial.println(PPP.radioState());
    if (attached) {
        Serial.print("Operator: ");
        Serial.println(PPP.operatorName());
        Serial.print("IMSI: ");
        Serial.println(PPP.IMSI());
        Serial.print("RSSI: ");
        Serial.println(PPP.RSSI());
        int ber = PPP.BER();
        if (ber > 0) {
            Serial.print("BER: ");
            Serial.println(ber);
            Serial.print("NetMode: ");
            Serial.println(PPP.networkMode());
        }

        Serial.println("Switching to data mode...");
        PPP.mode(ESP_MODEM_MODE_CMUX);  // Data and Command mixed mode
        if (!PPP.waitStatusBits(ESP_NETIF_CONNECTED_BIT, 1000)) {
            Serial.println("Failed to connect to internet!");
        }
        else {
            Serial.println("Connected to internet!");
        }
    }
    else {
        Serial.println("Failed to connect to network!");
    }
}

bool queConnected() {
    return PPP.connected();
}

String getIP() {
    return String(PPP.localIP().toString());
}

void setup()
{
    Serial.begin(115200);
    Quectelbegin(); // begin quectel to get connect to internet
    while (!queConnected()) // check if quectel connect to internet if not wait for it
    {
        // boo!
    }
    if (queConnected())
    {
        String IP = getIP();
        if (IP != "0.0.0.0")
        {
            Serial.println("PPPOS CONNECTED !");
            Serial.println("IP : " + IP);
        }
    }
}

void resetQuectel()
{
    if (quectelOffOnce)
    {
        if (PPP.connected() && WiFi.status() != WL_CONNECTED)
        {
            Serial.println("PPP.connected() : " + String(PPP.connected()));
            Serial.println("WiFi.status() == WL_DISCONNECTED : " + String(WiFi.status() == WL_DISCONNECTED));
            Serial.println("+PPP END #1");
            PPP.end();
            Serial.println("+PPP END #2");
        }

        Serial.println("PPP end here !");
        pinMode(PPP_MODEM_RST, OUTPUT);
        digitalWrite(PPP_MODEM_RST, true);
        delay(3000);
        digitalWrite(PPP_MODEM_RST, false);
        delay(100);
        Serial.println("Modem Reset Done");
        quectelOffOnce = false;
    }
}

void connectAndCheckWiFiStatus()
{
    if (WiFi.status() != WL_CONNECTED)
    {
        WiFi.begin(ssid, pass);
        delay(200);
        while (WiFi.status() != WL_CONNECTED)
        {
            Serial.print(".");
            delay(100);
        }
        if (WiFi.status() == WL_CONNECTED)
        {
            return;
        }
    }
}

void loop()
{
    if (!quackOff)
    {
        if (!queConnected())
        {
            Quectelbegin();
        }
        while (!queConnected())
        {
        }; // check if quectel connect to internet if not wait for it
    }
    delay(100);
    testClient("google.com", 80);
    if (loop_cnt >= 200)
    {
        if (!quackOff)
        {
            resetQuectel();
            quackOff = true;
        }
    }
    if (quackOff)
    {
        connectAndCheckWiFiStatus();
    }
    loop_cnt++;
    delay(2000);
}
```

### Debug Message

```plain
Core  1 register dump:
PC      : 0x00000000  PS      : 0x00060530  A0      : 0x82089248  A1      : 0x3fca8700  
A2      : 0x00000001  A3      : 0x3fca23c0  A4      : 0x3fca1600  A5      : 0x3fca8710  
A6      : 0x3fca87e0  A7      : 0x3fce9ff0  A8      : 0x82089219  A9      : 0x3fca86f0  
A10     : 0x3fca8710  A11     : 0x3fca8704  A12     : 0x3fca8728  A13     : 0x3fca87dc  
A14     : 0x00000032  A15     : 0x00000000  SAR     : 0x00000019  EXCCAUSE: 0x00000014  
EXCVADDR: 0x00000000  LBEG    : 0x40056f5c  LEND    : 0x40056f72  LCOUNT  : 0x00000000  

Backtrace: 0xfffffffd:0x3fca8700 0x42089245:0x3fca8800 0x40380e36:0x3fca8820

ELF file SHA256: c90df16c60dd61c6
```

### Other Steps to Reproduce

i have only windows.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2024-08-15:
please simplify the sketch to minimum necessary and decode the backtrace with ExceptionDecoder

The3ven, 2024-08-15:
this is minimum ketch code from my project, it just have wifi, ppp and a testclient function, all of them handled by flags

me-no-dev, 2024-08-15:
What is the exception reason? Can you simplify the sketch? Maybe also try this:
[image]

me-no-dev, 2024-08-15:
> his is minimum ketch code from my project

it contains a lot of calls that are not necessary to show that "PPP Crash while im call PPP.end() and Call WiFi.begin(ssid,pass);"

The3ven, 2024-08-15:
it like when PPP.end()  called it didnt end event task which is begin by NetworkEvent which is called by PPP.begin()

me-no-dev, 2024-08-15:
That sketch above is too complicated to display and replicate the issue. I am not sure how to say that better. Please simplify it to something like starting PPP, stopping PPP and starting WIFI. Code above has many ifs and tracks divferent things that are in no way connected to the issue.

The3ven, 2024-08-15:
> That sketch above is too complicated to display and replicate the issue. I am not sure how to say that better. Please simplify it to something like starting PPP, stopping PPP and starting WIFI. Code above has many ifs and tracks divferent things that are in no way connected to the issue.

i got it let me simplify it

The3ven, 2024-08-15:
`#include <PPP.h>
#include <WiFi.h>

String ssid = "ssid";
String pass = "pass";

int loop_cnt = 0;
bool quackOff = false;

#define PPP_MODEM_APN "AIRTELGPRS.COM" // For SimCard ISP
#define PPP_MODEM_PIN   NULL // or NULL

// EC200U copyed from WaveShare SIM7600 HW Flow Control
#define PPP_MODEM_RST     41 // as vishal sys it will aviliable in next rev
#define PPP_MODEM_RST_LOW true  //active HIGH
#define PPP_MODEM_TX      4
#define PPP_MODEM_RX      5
#define PPP_MODEM_RTS     -1
#define PPP_MODEM_CTS     -1
#define PPP_MODEM_FC      ESP_MODEM_FLOW_CONTROL_NONE
#define PPP_MODEM_MODEL   PPP_MODEM_GENERIC

void onEvent(arduino_event_id_t event, arduino_event_info_t info) {
    switch (event) {
    case ARDUINO_EVENT_PPP_START:
        Serial.println("PPP Started");
        break;
    case ARDUINO_EVENT_PPP_CONNECTED:
        Serial.println("PPP Connected");
        break;
    case ARDUINO_EVENT_PPP_GOT_IP:
        Serial.println("PPP Got IP");
        break;
    case ARDUINO_EVENT_PPP_LOST_IP:
        Serial.println("PPP Lost IP");
        break;
    case ARDUINO_EVENT_PPP_DISCONNECTED:
        Serial.println("PPP Disconnected");
        break;
    case ARDUINO_EVENT_PPP_STOP:
        Serial.println("PPP Stopped");
        break;
    default:
        break;
    }
}

void testClient(const char* host, uint16_t port) {
    NetworkClient client;
    if (!client.connect(host, port)) {
        Serial.println("Connection Failed");
        return;
    }
    client.printf("GET / HTTP/1.1\r\nHost: %s\r\n\r\n", host);
    while (client.connected() && !client.available());
    while (client.available()) {
        client.read();  //Serial.write(client.read());
    }

    Serial.println("Connection Success");
    client.stop();
}

void setup()
{
    Serial.begin(115200);
    Network.onEvent(onEvent);

    // Configure the modem
    PPP.setApn(PPP_MODEM_APN);
    PPP.setPin(PPP_MODEM_PIN);
    PPP.setResetPin(PPP_MODEM_RST, PPP_MODEM_RST_LOW, 300);
    PPP.setPins(PPP_MODEM_TX, PPP_MODEM_RX, PPP_MODEM_RTS, PPP_MODEM_CTS, PPP_MODEM_FC);

    Serial.println("Starting the modem. It might take a while!");
    PPP.begin(PPP_MODEM_MODEL);

    Serial.print("Manufacturer: ");
    Serial.println(PPP.cmd("AT+CGMI", 10000));
    Serial.print("Model: ");
    Serial.println(PPP.moduleName());
    Serial.print("IMEI: ");
    Serial.println(PPP.IMEI());

    bool attached = PPP.attached();
    if (!attached) {
        int i = 0;
        unsigned int s = millis();
        Serial.print("Waiting to connect to network");
        while (!attached && ((++i) < 600)) {
            Serial.print(".");
            delay(100);
            attached = PPP.attached();
        }
        Serial.print((millis() - s) / 1000.0, 1);
        Serial.println("s");
        attached = PPP.attached();
    }

    Serial.print("Attached: ");
    Serial.println(attached);
    Serial.print("State: ");
    Serial.println(PPP.radioState());
    if (attached) {
        Serial.print("Operator: ");
        Serial.println(PPP.operatorName());
        Serial.print("IMSI: ");
        Serial.println(PPP.IMSI());
        Serial.print("RSSI: ");
        Serial.println(PPP.RSSI());
        int ber = PPP.BER();
        if (ber > 0) {
            Serial.print("BER: ");
            Serial.println(ber);
            Serial.print("NetMode: ");
            Serial.println(PPP.networkMode());
        }

        Serial.println("Switching to data mode...");
        PPP.mode(ESP_MODEM_MODE_CMUX);  // Data and Command mixed mode
        if (!PPP.waitStatusBits(ESP_NETIF_CONNECTED_BIT, 1000)) {
            Serial.println("Failed to connect to internet!");
        }
        else {
            Serial.println("Connected to internet!");
        }
    }
    else {
        Serial.println("Failed to connect to network!");
    }
}

void loop()
{
    if (loop_cnt >= 200 && PPP.connected() && WiFi.status() != WL_CONNECTED)
    {
        PPP.end();
        pinMode(PPP_MODEM_RST, OUTPUT);
        digitalWrite(PPP_MODEM_RST, true);
        delay(3000);
        digitalWrite(PPP_MODEM_RST, false);
        delay(100);
        quackOff = true;
    }

    if (quackOff)
    {
        if (WiFi.status() != WL_CONNECTED)
        {
            WiFi.begin(ssid, pass);
            delay(200);
            while (WiFi.status() != WL_CONNECTED)
            {
                Serial.print(".");
                delay(100);
            }
        }
    }

    testClient("google.com", 80);
    loop_cnt++;
    delay(2000);
}`

Here we go

Jason2866, 2024-08-15:
Using the example sketch (modified for Lilygo T-SIMCAM with mPCIe SIM7600G LTE module) works fine with the S3
```

#include <PPP.h>

#define PPP_MODEM_APN "internet"
#define PPP_MODEM_PIN "0000"  // or NULL

// Lilygo T-SIMCAM with mPCIe SIM7600G LTE module
#define PPP_MODEM_RST     48
#define PPP_MODEM_RST_LOW false  //active HIGH
#define PPP_MODEM_TX      45
#define PPP_MODEM_RX      46
#define PPP_MODEM_RTS     -1
#define PPP_MODEM_CTS     -1
#define PPP_MODEM_FC      ESP_MODEM_FLOW_CONTROL_NONE
#define PPP_MODEM_MODEL   PPP_MODEM_SIM7600

// SIM800 basic module with just TX,RX and RST
// #define PPP_MODEM_RST     0
// #define PPP_MODEM_RST_LOW true //active LOW
// #define PPP_MODEM_TX      2
// #define PPP_MODEM_RX      19
// #define PPP_MODEM_RTS     -1
// #define PPP_MODEM_CTS     -1
// #define PPP_MODEM_FC      ESP_MODEM_FLOW_CONTROL_NONE
// #define PPP_MODEM_MODEL   PPP_MODEM_SIM800

void onEvent(arduino_event_id_t event, arduino_event_info_t info) {
  switch (event) {
    case ARDUINO_EVENT_PPP_START:        Serial.println("PPP Started"); break;
    case ARDUINO_EVENT_PPP_CONNECTED:    Serial.println("PPP Connected"); break;
    case ARDUINO_EVENT_PPP_GOT_IP:       Serial.println("PPP Got IP"); break;
    case ARDUINO_EVENT_PPP_LOST_IP:      Serial.println("PPP Lost IP"); break;
    case ARDUINO_EVENT_PPP_DISCONNECTED: Serial.println("PPP Disconnected"); break;
    case ARDUINO_EVENT_PPP_STOP:         Serial.println("PPP Stopped"); break;
    default:                             break;
  }
}

void testClient(const char *host, uint16_t port) {
  NetworkClient client;
  if (!client.connect(host, port)) {
    Serial.println("Connection Failed");
    return;
  }
  client.printf("GET / HTTP/1.1\r\nHost: %s\r\n\r\n", host);
  while (client.connected() && !client.available());
  while (client.available()) {
    client.read();  //Serial.write(client.read());
  }

  Serial.println("Connection Success");
  client.stop();
}

void setup() {
  Serial.begin(115200);

  // Listen for modem events
  Network.onEvent(onEvent);

  // Configure the modem
  PPP.setApn(PPP_MODEM_APN);
  PPP.setPin(PPP_MODEM_PIN);
  PPP.setResetPin(PPP_MODEM_RST, PPP_MODEM_RST_LOW);
  PPP.setPins(PPP_MODEM_TX, PPP_MODEM_RX, PPP_MODEM_RTS, PPP_MODEM_CTS, PPP_MODEM_FC);

  Serial.println("Starting the modem. It might take a while!");
  PPP.begin(PPP_MODEM_MODEL);

  Serial.print("Manufacturer: ");
  Serial.println(PPP.cmd("AT+CGMI", 10000));
  Serial.print("Model: ");
  Serial.println(PPP.moduleName());
  Serial.print("IMEI: ");
  Serial.println(PPP.IMEI());

  bool attached = PPP.attached();
  if (!attached) {
    int i = 0;
    unsigned int s = millis();
    Serial.print("Waiting to connect to network");
    while (!attached && ((++i) < 600)) {
      Serial.print(".");
      delay(100);
      attached = PPP.attached();
    }
    Serial.print((millis() - s) / 1000.0, 1);
    Serial.println("s");
    attached = PPP.attached();
  }

  Serial.print("Attached: ");
  Serial.println(attached);
  Serial.print("State: ");
  Serial.println(PPP.radioState());
  if (attached) {
    Serial.print("Operator: ");
    Serial.println(PPP.operatorName());
    Serial.print("IMSI: ");
    Serial.println(PPP.IMSI());
    Serial.print("RSSI: ");
    Serial.println(PPP.RSSI());
    int ber = PPP.BER();
    if (ber > 0) {
      Serial.print("BER: ");
      Serial.println(ber);
      Serial.print("NetMode: ");
      Serial.println(PPP.networkMode());
    }

    Serial.println("Switching to data mode...");
    PPP.mode(ESP_MODEM_MODE_CMUX);  // Data and Command mixed mode
    if (!PPP.waitStatusBits(ESP_NETIF_CONNECTED_BIT, 1000)) {
      Serial.println("Failed to connect to internet!");
    } else {
      Serial.println("Connected to internet!");
    }
  } else {
    Serial.println("Failed to connect to network!");
  }
}

void loop() {
  if (PPP.connected()) {
    testClient("google.com", 80);
  }
  delay(20000);
}

```

The3ven, 2024-08-15:
> Using the example sketch (modified for Lilygo T-SIMCAM with mPCIe SIM7600G LTE module) works fine with the S3
>
> ```
>
> #include <PPP.h>
>
> #define PPP_MODEM_APN "internet"
> #define PPP_MODEM_PIN "0000" [...]

PPP Standalone Work like a charm with my board too, but when im add WiFi with it, then i encounter crash, did you check PPP with WiFi ?

me-no-dev, 2024-08-15:
I tried your sketch and removed the unnecessary prints. Here is the result:
```
Starting the modem. It might take a while!
PPP Started
Waiting to connect to network..............................................................................................
Switching to data mode...
PPP Got IP
PPP Connected
Connected to internet!
Connection Success
Connection Success
Connection Success
Connection Success
Connection Success
Connection Success
Connection Success
Connection Success
Connection Success
Connection Success
Connection Success
Connection Success
Connection Success
Connection Success
Connection Success
Connection Success
Connection Success
Connection Success
Connection Success
Connection Success
PPP Disconnected
PPP Stopped
..........Connection Success
Connection Success
Connection Success
Connection Success

```

The3ven, 2024-08-15:
thanks a lot it seems my project have issue, i will check my project and get back to you, please don`t close this issue for now

The3ven, 2024-08-16:
i have add delay between PPP.end() and then turn LTE (Quectel EC200UCN) module off via gpio pins 
`
  pinMode(PPP_MODEM_RST, OUTPUT);
  digitalWrite(PPP_MODEM_RST, true);
  delay(3000);
  digitalWrite(PPP_MODEM_RST, false);
  delay(100);
`

like so, and then i wait for any of one Event from 

`
ARDUINO_EVENT_PPP_LOST_IP
ARDUINO_EVENT_PPP_DISCONNECTED
ARDUINO_EVENT_PPP_STOP
` 

then im turn wifi on and it got crashed,
one more thing i notice if i didnt turn wifi on it didnt crash and PPP.end() 
function executed successfully

here is crash backtrack and logs

`
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x4208bff5  PS      : 0x00060930  A0      : 0x820e704d  A1      : 0x3fcb9020  
A2      : 0x00000000  A3      : 0x3fcb90c4  A4      : 0x00000032  A5      : 0x3fcb903c  
A6      : 0x3fca1660  A7      : 0x00000000  A8      : 0x00000000  A9      : 0xbde60ddc  
A10     : 0x00000000  A11     : 0x3fcb90c4  A12     : 0x3fcb903c  A13     : 0x00000009  
A14     : 0x3fcb90b4  A15     : 0xff000000  SAR     : 0x0000001b  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x400570e8  LEND    : 0x400570f3  LCOUNT  : 0xffffffff  

Backtrace: 0x4208bff2:0x3fcb9020 0x420e704a:0x3fcb90a0 0x42050d07:0x3fcb91a0 0x4203217b:0x3fcb9230 0x42073635:0x3fcb92b0 0x4204d2d1:0x3fcb9300 0x40380e36:0x3fcb9360

ELF file SHA256: a600cda290b27b66
`

`
PC: 0x4208bff5: NetworkEvents::removeEvent(std::function , arduino_event_id_t) at C:\Users\HP\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.3\libraries\Network\src\NetworkEvents.cpp line 231
EXCVADDR: 0x00000000

Decoding stack results
0x4208bff2: NetworkEvents::removeEvent(std::function , arduino_event_id_t) at c:\users\hp\appdata\local\arduino15\packages\esp32\tools\esp-xs3\2302\xtensa-esp32s3-elf\include\c++\12.2.0\bits/std_function.h line 662
0x420e704a: PPPClass::end() at C:\Users\HP\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.3\libraries\PPP\src\PPP.cpp line 421
0x42050d07: quectelEnd(String, String, int) at F:\__FIRMWARE__\NDDB_DPMCU_Project/ORIGHT_Quectel.cpp line 538
0x4203217b: sync_successful_screen(bool) at F:\__FIRMWARE__\NDDB_DPMCU_Project/Menu_Utils.cpp line 1577
0x42073635: ftp_ppp_func(int) at F:\__FIRMWARE__\NDDB_DPMCU_Project/WiFi_FTP_Utils.cpp line 218
0x4204d2d1: main_task(void*) at F:\__FIRMWARE__\NDDB_DPMCU_Project/NDDB_DPMCU_Project.ino line 149
0x40380e36: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c line 162

`

i have also enabled verbose here 

`
[127329][V][NetworkEvents.cpp:119] checkForEvent(): Network Event: 46 - PPP_DISCONNECTED
[127339][V][PPP.cpp:101] _onPppArduinoEvent(): PPP Arduino Event 46: PPP_DISCONNECTED
[127347][V][NetworkEvents.cpp:119] checkForEvent(): Network Event: 44 - PPP_STOP
[127355][V][PPP.cpp:101] _onPppArduinoEvent(): PPP Arduino Event 44: PPP_STOP
[127377][V][NetworkInterface.cpp:264] destroyNetif(): Unregistered event handler
[127383][V][NetworkEvents.cpp:119] checkForEvent(): Network Event: 9 - WIFI_READY
[127386][V][STA.cpp:184] _onStaEvent(): STA Started
[127397][V][NetworkEvents.cpp:119] checkForEvent(): Network Event: 11 - STA_START
[127405][V][STA.cpp:110] _onStaArduinoEvent(): Arduino STA Event: 11 - STA_START
`

The3ven, 2024-08-16:
@me-no-dev @Jason2866 i found and killed this issue, it was in my code i have a variable which is being true from callback function which handles WiFi to be turn on as i know callbacks are work so fast and my WiFi get in connection mode before PPP.end() in middle of execution to kill own callback and that same callback is called from WiFi and when WiFi called it there is no such callback because PPP killed that and it got crashed. Thank you very much everyone for your valuable time to help me in this journey to get rid off from this issue, ❤ from India

closed 2024-08-16 by The3ven

barbiani, 2025-09-29:
I am facing the same problem and did not find a solution yet.

PPPos is connected, then wifi STA connects. WiFi STA event signals a task to cal PPP.end(). Crash.

What is the proper way to end PPP when wifi is connected again?

Thank you.