espressif/arduino-esp32 #10167: Inconsistant of component's hash files of arduino esp32 component dependencies (DSP, libsodium)

executer-uno, 2024-08-15:
### Board

ESP32, ESP32-S3 may be others

### Device Description

build system issue

### Hardware Configuration

bare HelloWorld project from espressif__arduino-esp32\examples\hello_world

### Version

v3.0.4

### IDE Name

Visual Studio Code with ESP-IDF extension v1.8.0 (+path https://github.com/espressif/vscode-esp-idf-extension/pull/1252)

### Operating System

Windows 10

### Flash frequency

no matter

### PSRAM enabled

no

### Upload speed

115200

### Description

Got consistent error messages on menuconfig, build or target change on some preconditions (clear project, sdkconfig updated).

Starting point is
Clear copy of espressif__arduino-esp32\examples\hello_world project with main\idf_component.yml one line commented out:
```
dependencies:
  espressif/arduino-esp32:
    version: "*"
    # override_path: "../../../"
    pre_release: false
```

- **:x: menuconfig gives an Error (see "Debug Message") + only on that first step lipsodium hash also appears in error test**
- :heavy_check_mark: espressif__esp-dsp\.component_hash deleted
- menuconfig opens fine
- **:x: build - Fails on same error (see "Debug Message")**
- :heavy_check_mark: espressif__esp-dsp\.component_hash deleted
- build - Ok (entire build, takes long time)
- menuconfig opens fine, change some parameters, save sdkconfig
- **:x: build - Fails on same error (see "Debug Message")**
- :heavy_check_mark: espressif__esp-dsp\.component_hash deleted
- build - Ok (entire rebuild, takes long time)
- main.cpp updated
- build - ok, changes only, fast
- **:x: change target to ESP32-S3 same error (see "Debug Message")**
- :heavy_check_mark: espressif__esp-dsp\.component_hash deleted
- change target to ESP32-S3 - OK
- **:x: menuconfig gives an Error (see "Debug Message")**
- :heavy_check_mark: espressif__esp-dsp\.component_hash deleted
- menuconfig opens fine
- **:x: build - Fails on same error (see "Debug Message")**
- :heavy_check_mark: espressif__esp-dsp\.component_hash deleted
- build - Ok (entire rebuild, takes long time)

### Sketch

```cpp
main\main.cpp

#include "Arduino.h"

void setup() {
  Serial.begin(115200);
}

void loop() {
  Serial.println("Hello world! Update!");
  delay(1000);
}
```
```

### Debug Message

```plain
---------------------------ERROR--------------------------
CMake Error at xxxxxxx/esp/v5.1/esp-idf/tools/cmake/build.cmake:540 (message):
  ERROR: Some components (espressif/esp-dsp) in the "managed_components"
  directory were modified on the disk since the last run of the CMake.
  Content of this directory is managed automatically.

  If you want to keep the changes, you can move the directory with the
-----------------------END OF ERROR-----------------------
---------------------------ERROR--------------------------
  component to the "components"directory of your project.

  I.E.  for "espressif__esp-dsp" run:

  mv
  xxxxxxx\VSC_Workspace\hello_world\managed_components\espressif__esp-dsp
  xxxxxxx\VSC_Workspace\hello_world\components\espressif__esp-dsp

  Or, if you want to discard the changes remove the ".component_hash" file
  from the component's directory.

  I.E.  for "espressif__esp-dsp" run:

  rm
  xxxxxxx\VSC_Workspace\hello_world\managed_components\espressif__esp-dsp\.component_hash

Call Stack (most recent call first):
  xxxxxxx/esp/v5.1/esp-idf/tools/cmake/project.cmake:604 (idf_build_process)
  CMakeLists.txt:8 (project)
-----------------------END OF ERROR-----------------------
```

### Other Steps to Reproduce

Going to check if it is not an issue with ESP-IDF VSC Extension patch from PR1252
https://github.com/espressif/vscode-esp-idf-extension/pull/1252

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2024-08-15:
We have those same issues in the lib-builder and go around them by deleting the `.component_hash` files before `idf.py build`. You can post an issue in the relevant component repositories, if you want to help bring more attention to the problem.

executer-uno, 2024-08-15:
Ok, Issue defenetly not depends on https://github.com/espressif/vscode-esp-idf-extension/pull/1252, exactly same behaviour on clear ESP-IDF extension v1.8.0

Jason2866, 2024-08-15:
There is a setting for avoiding the component manager issue. Set the environment variable
`IDF_COMPONENT_OVERWRITE_MANAGED_COMPONENTS` to `1` and it compiles.

executer-uno, 2024-08-15:
Thanks for pointing that env var
https://github.com/espressif/idf-component-manager/issues/40
looks like here is a problem's root.

I test it with 
```
set(ENV{IDF_COMPONENT_OVERWRITE_MANAGED_COMPONENTS} 1)
```
in root CMakeLists and it works ok now. sdkconfig updates leads to full rebuild, cpp sources updates leads to fast changes build.
Thanks a lot with that note, github search shows its rarely used and not obvious that idf-component-manager causes it.

closed 2024-08-15 by executer-uno