espressif/arduino-esp32 #10174: NetworkClientSecure not achieving parity for Ethernet and Wifi

waymond91, 2024-08-16:
### Board

ESP32-S3 No PSRAM

### Device Description

Adafuit Feather Esp32 S3 No PSRAM

### Hardware Configuration

Adafruit Ethernet Feather : Wiznet W5500

### Version

v3.0.4

### IDE Name

Platformio

### Operating System

macOS 14.3.1

### Flash frequency

Platformio Default/Defined

### PSRAM enabled

no

### Upload speed

115200

### Description

I am connecting to an ssl encrypted mqtt broker using NetworkClientSecure. Using WiFi, everything works as expected, however when I change the network interface to Ethernet, the device resets without a backtrace.

In version 2.X with WiFiClientSecure there was an option to "setDebugLevel" which was a big help for https stuff because it gave some feedback as to what the issue was, but Im not getting any feedback for mqtt on version 2 or 3.

This sketch follows the [EMQX esp32 example](https://docs.emqx.com/en/cloud/latest/connect_to_deployments/esp32.html) replacing WiFiClientSecure wth NetworkClientSecure.

I do have cert files for the client but Im just using `setInsecure()` for now:

```
#include <WiFi.h>
#include <PubSubClient.h>
#include <NetworkClientSecure.h>
#include <SPI.h>

// WiFi credentials
const char *ssid = "ssid";             // Replace with your WiFi name
const char *password = "password";   // Replace with your WiFi password

// MQTT Broker settings
const char *mqtt_broker = "our-self-signed-emqx-broker.com";
const char *mqtt_topic = "emqx/esp32";
const int mqtt_port = 18004;

NetworkClientSecure esp_client;
PubSubClient mqtt_client(esp_client);

void connectToWiFi();

void connectToMQTT();

void mqttCallback(char *topic, byte *payload, unsigned int length);

void setup() {
    Serial.begin(115200);
    connectToWiFi();

    // Set Root CA certificate
    esp_client.setInsecure();

    mqtt_client.setServer(mqtt_broker, mqtt_port);
    mqtt_client.setKeepAlive(60);
    mqtt_client.setCallback(mqttCallback);
    connectToMQTT();
}

void connectToWiFi() {
    WiFi.begin(ssid, password);
    Serial.print("Connecting to WiFi");
    while (WiFi.status() != WL_CONNECTED) {
        delay(500);
        Serial.print(".");
    }
    Serial.println("\nConnected to WiFi");
}

void connectToMQTT() {
    while (!mqtt_client.connected()) {
        String client_id = "esp32-client-" + String(WiFi.macAddress());
        Serial.printf("Connecting to MQTT Broker as %s...\n", client_id.c_str());
        if (mqtt_client.connect(client_id.c_str())) {
            Serial.println("Connected to MQTT broker");
            mqtt_client.subscribe(mqtt_topic);
            mqtt_client.publish(mqtt_topic, "Hi EMQX I'm ESP32 ^^");  // Publish message upon connection
        } else {
            Serial.print("Failed to connect to MQTT broker, rc=");
            Serial.print(mqtt_client.state());
            Serial.println(" Retrying in 5 seconds.");
            delay(5000);
        }
    }
}

void mqttCallback(char *topic, byte *payload, unsigned int length) {
    Serial.print("Message received on topic: ");
    Serial.println(topic);
    Serial.print("Message: ");
    for (unsigned int i = 0; i < length; i++) {
        Serial.print((char) payload[i]);
    }
    Serial.println("\n-----------------------");
}

void loop() {
    if (!mqtt_client.connected()) {
        connectToMQTT();
    }
    mqtt_client.loop();
}

```

And in the terminal: 
```
Connected to WiFi
Connecting to MQTT Broker as esp32-client-DC:54:75:C3:B6:00...
Connected to MQTT broker
Message received on topic: emqx/esp32
Message: Hi EMQX I'm ESP32 ^^
```

My platformio.ini:

```
[env:adafruit_feather_esp32s3_nopsram]
platform = https://github.com/pioarduino/platform-espressif32/releases/download/51.03.04/platform-espressif32.zip
board = adafruit_feather_esp32s3_nopsram
framework = arduino
board_build.partitions = min_spiffs.csv
lib_deps = 
    knolleary/PubSubClient@^2.8
    arduino-libraries/Ethernet@^2.0.2
    arduino-libraries/NTPClient@^3.2.1
monitor_filters = esp32_exception_decoder

monitor_speed = 115200

build_flags =
    -DCORE_DEBUG_LEVEL=5
```

### Sketch

```cpp
#include <Ethernet.h>
#include <PubSubClient.h>
#include <NetworkClientSecure.h>
#include <SPI.h>

// Ethernet settings
byte mac[] = { 0xDE, 0xAD, 0xBE, 0xEF, 0xFE, 0xED }; // Replace with your Ethernet MAC address

// MQTT Broker settings
const char *mqtt_broker = "our-self-signed-emqx-broker.com";
const char *mqtt_topic = "emqx/esp32";
const int mqtt_port = 18004;

// Ethernet and MQTT client initialization
NetworkClientSecure eth_client;
PubSubClient mqtt_client(eth_client);

// Function Declarations
void connectToEthernet();
void connectToMQTT();
void mqttCallback(char *topic, byte *payload, unsigned int length);

void setup() {
    Serial.begin(115200);
    connectToEthernet();

    eth_client.setInsecure();

    mqtt_client.setServer(mqtt_broker, mqtt_port);
    mqtt_client.setKeepAlive(60);
    mqtt_client.setCallback(mqttCallback);
    connectToMQTT();
}

void connectToEthernet() {
    Serial.println("Connecting to Ethernet");
    if (Ethernet.begin(mac) == 0) {
        Serial.println("Failed to configure Ethernet using DHCP");
        while (true); // Stop if DHCP failed
    }
    Serial.print("Connected to Ethernet with IP: ");
    Serial.println(Ethernet.localIP());
}

void connectToMQTT() {
    while (!mqtt_client.connected()) {
        String client_id = "esp32-client-" + String(Ethernet.localIP());
        Serial.printf("Connecting to MQTT Broker as %s...\n", client_id.c_str());
        if (mqtt_client.connect(client_id.c_str())) {
            Serial.println("Connected to MQTT broker");
            mqtt_client.subscribe(mqtt_topic);
            mqtt_client.publish(mqtt_topic, "Hi EMQX I'm ESP32 ^^");  // Publish message upon connection
        } else {
            Serial.print("Failed to connect to MQTT broker, rc=");
            Serial.print(mqtt_client.state());
            Serial.println(" Retrying in 5 seconds.");
            delay(5000);
        }
    }
}

void mqttCallback(char *topic, byte *payload, unsigned int length) {
    Serial.print("Message received on topic: ");
    Serial.println(topic);
    Serial.print("Message: ");
    for (unsigned int i = 0; i < length; i++) {
        Serial.print((char)payload[i]);
    }
    Serial.println("\n-----------------------");
}

void loop() {
    if (!mqtt_client.connected()) {
        connectToMQTT();
    }
    mqtt_client.loop();
}
```

### Debug Message

```plain
Reconnecting to /dev/cu.usbmodem14201 .
Please build project in debug configuration to get more details about an exception.
See https://docs.platformio.org/page/projectconf/build_configurations.html

         Connected!
Connecting to Ethernet
Connected to Ethernet with IP: 192.168.0.44
Connecting to MQTT Broker as esp32-client-738240704...
Disconnected (read failed: [Errno 6] Device not configured)
Reconnecting to /dev/cu.usbmodem14201 .
Please build project in debug configuration to get more details about an exception.
See https://docs.platformio.org/page/projectconf/build_configurations.html

         Connected!
Connecting to Ethernet
Connected to Ethernet with IP: 192.168.0.44
Connecting to MQTT Broker as esp32-client-738240704...
Disconnected (read failed: [Errno 6] Device not configured)
Reconnecting to /dev/cu.usbmodem14201 .
Please build project in debug configuration to get more details about an exception.
See https://docs.platformio.org/page/projectconf/build_configurations.html

         Connected!
Connecting to Ethernet
Connected to Ethernet with IP: 192.168.0.44
Connecting to MQTT Broker as esp32-client-738240704...
Disconnected (read failed: [Errno 6] Device not configured)
Reconnecting to /dev/cu.usbmodem14201 .
Please build project in debug configuration to get more details about an exception.
See https://docs.platformio.org/page/projectconf/build_configurations.html

         Connected!
Connecting to Ethernet
Connected to Ethernet with IP: 192.168.0.44
Connecting to MQTT Broker as esp32-client-738240704...
Disconnected (read failed: [Errno 6] Device not configured)
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2024-08-16:
Is there a reason you have Ethernet.h in your platformio.ini rather than using the library included with the core (and designed for Network.h)?

waymond91, 2024-08-16:
Wow great point - I just migrated to this platform for the networkclientsecure library, and I was just using that other ethernet library before because it ships with the adafruit feather.

Having some difficulty using ETH.h but Ill keep you posted.

waymond91, 2024-08-16:
Thank you for the help! I found another library that seemed to do it. Not sure why I couldnt get ETH.h to work.
For posterity:
```
#include <EthernetESP32.h>
#include <NetworkClientSecure.h>
#include <SPI.h>
#include <PubSubClient.h>

#define WIZNET_SCLK_PIN 29
#define WIZNET_MISO_PIN 30
#define WIZNET_MOSI_PIN 28
#define WIZNET_CS_PIN 10
#define WIZNET_RESET_PIN 26

W5500Driver driver(WIZNET_CS_PIN);
NetworkClientSecure client;
PubSubClient mqtt_client(client);

IPAddress ip(192, 168, 0, 177);

// MQTT Broker settings
const char *mqtt_broker = "flipbit.gps-mapping.com";
const char *mqtt_topic = "emqx/esp32";
const int mqtt_port = 18004;

// Function Declarations
void connectToEthernet();
void connectToMQTT();
void mqttCallback(char *topic, byte *payload, unsigned int length);

void setup() {
  Serial.begin(115200);
  delay(500);
  while (!Serial.available()){
    delay(100);
  }
 // SPI.begin(WIZNET_SCLK_PIN, WIZNET_MISO_PIN, WIZNET_MOSI_PIN);

  Ethernet.init(driver);

  // start the Ethernet connection:
  Serial.println("Initialize Ethernet with DHCP:");
  if (Ethernet.begin() == 0) {
    Serial.println("Failed to configure Ethernet using DHCP");
    // Check for Ethernet hardware present
    if (Ethernet.hardwareStatus() == EthernetNoHardware) {
      Serial.println("Ethernet shield was not found.  Sorry, can't run without hardware. :(");
      while (true) {
        Serial.println("fuck");
        delay(500); // do nothing, no point running without Ethernet hardware
      }
    }
    if (Ethernet.linkStatus() == LinkOFF) {
      Serial.println("Ethernet cable is not connected.");
    }
    // try to configure using IP address instead of DHCP:
    Ethernet.begin(ip);
  } else {
    Serial.print("  DHCP assigned IP ");
    Serial.println(Ethernet.localIP());
  }

  client.setInsecure();
  mqtt_client.setServer(mqtt_broker, mqtt_port);
  mqtt_client.setKeepAlive(60);
  mqtt_client.setCallback(mqttCallback);
  connectToMQTT();

}
void connectToMQTT() {
    while (!mqtt_client.connected()) {
        String client_id = "esp32-client-" + String(Ethernet.localIP());
        Serial.printf("Connecting to MQTT Broker as %s...\n", client_id.c_str());
        if (mqtt_client.connect(client_id.c_str())) {
            Serial.println("Connected to MQTT broker");
            mqtt_client.subscribe(mqtt_topic);
            mqtt_client.publish(mqtt_topic, "Hi EMQX I'm ESP32 ^^");  // Publish message upon connection
        } else {
            Serial.print("Failed to connect to MQTT broker, rc=");
            Serial.print(mqtt_client.state());
            Serial.println(" Retrying in 5 seconds.");
            delay(5000);
        }
    }
}

void mqttCallback(char *topic, byte *payload, unsigned int length) {
    Serial.print("Message received on topic: ");
    Serial.println(topic);
    Serial.print("Message: ");
    for (unsigned int i = 0; i < length; i++) {
        Serial.print((char)payload[i]);
    }
    Serial.println("\n-----------------------");
}

void loop() {
    if (!mqtt_client.connected()) {
        connectToMQTT();
    }
    mqtt_client.loop();
}
```
platform.ini

```
[env:adafruit_feather_esp32s3_nopsram]
platform = https://github.com/pioarduino/platform-espressif32/releases/download/51.03.04/platform-espressif32.zip
board = adafruit_feather_esp32s3_nopsram
framework = arduino
board_build.partitions = min_spiffs.csv
lib_deps = 
    knolleary/PubSubClient@^2.8
    https://github.com/Networking-for-Arduino/EthernetESP32
monitor_filters = esp32_exception_decoder
monitor_speed = 115200
```

closed 2024-08-16 by waymond91

Jason2866, 2024-08-18:
@waymond91 This is not needed and will make issues.
```
lib_deps =
       https://github.com/Networking-for-Arduino/EthernetESP32
```
The Arduino core has a working Network lib included. Check your setup and code.

waymond91, 2024-08-20:
@Jason2866 Thank you for the help. At the time of posting, I could not `#include <NetworkClientConfig>` withot using this as a dependency