espressif/arduino-esp32 #10184: Cant stablish mqtt connection in aws

ioviedodev, 2024-08-16:
### Board

ESP32 DEV KIT C

### Device Description

Just esp32 dev kit c and wifi connection

### Hardware Configuration

no connections

### Version

v3.0.4

### IDE Name

ECLIPSE

### Operating System

Windows 10

### Flash frequency

80 MH

### PSRAM enabled

yes

### Upload speed

115200

### Description

Using arduino-esp32 as IDF Component, i cant stablish mqtt connection the program is very simple i think that is something that i need to enable in sdkconfig. 

### Sketch

```cpp
//file: main.cpp
#ifndef ARDUINO_H
#include <Arduino.h>
#endif

#include "WiFi.h"
#include <WiFiClientSecure.h>
#include <MQTTClient.h>
#include <ArduinoJson.h>
//C8:F0:9E:69:43:F4

#define WIFI_SSID   "SSIDEXAMPLE"
#define WIFI_PASSWORD   "passwordExample"
#define THINGNAME   "44566873-21312132132132"
#define AWS_IOT_PUBLISH_TOPIC   "44566873-21312132132132"
#define AWS_IOT_SUBSCRIBE_TOPIC "44566873-21312132132132/room"
const char AWS_IOT_ENDPOINT[] = "23232323asd-ats.iot.us-east-1.amazonaws.com";

static const char AWS_CERT_CA[] = R"EOF(
-----BEGIN CERTIFICATE-----
MIIDQTCCAimgAwIBAgITBmyfz5m/jAo54vB4ikPmljZbyjANBgkqhkiG9w0BAQsF
ADA5MQswCQYDVQQGEwJVUzEPMA0GA1UEChMGQW1hem9uMRkwFwYDVQQDExBBbWF6
-----END CERTIFICATE-----
)EOF";

static const char AWS_DEV_CERT_CA[] = R"EOF(
-----BEGIN CERTIFICATE-----
[... 205 lines cut ...]
[ 96434][E][NetworkClientSecure.cpp:159] connect(): start_ssl_client: connect failed: -1
.[ 96541][E][NetworkClient.cpp:315] setSocketOption(): fail on 0, errno: 9, "Bad file number"
[115050][E][ssl_client.cpp:146] start_ssl_client(): socket error on fd 54, errno: 113, "Software caused connection abort"
[115062][E][NetworkClientSecure.cpp:159] connect(): start_ssl_client: connect failed: -1
.[115169][E][NetworkClient.cpp:315] setSocketOption(): fail on 0, errno: 9, "Bad file number"
[133678][E][ssl_client.cpp:146] start_ssl_client(): socket error on fd 54, errno: 113, "Software caused connection abort"
[133690][E][NetworkClientSecure.cpp:159] connect(): start_ssl_client: connect failed: -1
.[133797][E][NetworkClient.cpp:315] setSocketOption(): fail on 0, errno: 9, "Bad file number"
D (135540) wifi:eb is dhcp or dns sport = 31478, dport = 53
D (136540) wifi:eb is dhcp or dns sport = 31478, dport = 53
[153571][E][ssl_client.cpp:146] start_ssl_client(): socket error on fd 54, errno: 113, "Software caused connection abort"
[153582][E][NetworkClientSecure.cpp:159] connect(): start_ssl_client: connect failed: -1
.[153690][E][NetworkClient.cpp:315] setSocketOption(): fail on 0, errno: 9, "Bad file number"
[172199][E][ssl_client.cpp:146] start_ssl_client(): socket error on fd 54, errno: 113, "Software caused connection abort"
[172211][E][NetworkClientSecure.cpp:159] connect(): start_ssl_client: connect failed: -1
.[172318][E][NetworkClient.cpp:315] setSocketOption(): fail on 0, errno: 9, "Bad file number"
```

### Other Steps to Reproduce

.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

ioviedodev, 2024-08-16:
hi, i add the sdkconfig.
#
# Automatically generated file. DO NOT EDIT.
# Espressif IoT Development Framework (ESP-IDF) 5.1.4 Project Configuration
#
CONFIG_SOC_BROWNOUT_RESET_SUPPORTED="Not determined"
CONFIG_SOC_TWAI_BRP_DIV_SUPPORTED="Not determined"
CONFIG_SOC_DPORT_WORKAROUND="Not determined"
CONFIG_SOC_CAPS_ECO_VER_MAX=301
CONFIG_SOC_ADC_SUPPORTED=y
CONFIG_SOC_DAC_SUPPORTED=y
CONFIG_SOC_UART_SUPPORTED=y
CONFIG_SOC_MCPWM_SUPPORTED=y
CONFIG_SOC_GPTIMER_SUPPORTED=y
CONFIG_SOC_SDMMC_HOST_SUPPORTED=y
CONFIG_SOC_BT_SUPPORTED=y
CONFIG_SOC_PCNT_SUPPORTED=y
CONFIG_SOC_WIFI_SUPPORTED=y
CONFIG_SOC_SDIO_SLAVE_SUPPORTED=y
CONFIG_SOC_TWAI_SUPPORTED=y
CONFIG_SOC_EMAC_SUPPORTED=y
CONFIG_SOC_ULP_SUPPORTED=y
CONFIG_SOC_CCOMP_TIMER_SUPPORTED=y
CONFIG_SOC_RTC_FAST_MEM_SUPPORTED=y
CONFIG_SOC_RTC_SLOW_MEM_SUPPORTED=y
CONFIG_SOC_RTC_MEM_SUPPORTED=y
CONFIG_SOC_I2S_SUPPORTED=y
CONFIG_SOC_RMT_SUPPORTED=y
CONFIG_SOC_SDM_SUPPORTED=y
CONFIG_SOC_GPSPI_SUPPORTED=y
CONFIG_SOC_LEDC_SUPPORTED=y
CONFIG_SOC_I2C_SUPPORTED=y
CONFIG_SOC_SUPPORT_COEXISTENCE=y
CONFIG_SOC_AES_SUPPORTED=y
CONFIG_SOC_MPI_SUPPORTED=y
CONFIG_SOC_SHA_SUPPORTED=y
CONFIG_SOC_FLASH_ENC_SUPPORTED=y
CONFIG_SOC_SECURE_BOOT_SUPPORTED=y
CONFIG_SOC_TOUCH_SENSOR_SUPPORTED=y
CONFIG_SOC_BOD_SUPPORTED=y
CONFIG_SOC_ULP_FSM_SUPPORTED=y
CONFIG_SOC_DPORT_WORKAROUND_DIS_INTERRUPT_LVL=5
CONFIG_SOC_XTAL_SUPPORT_26M=y
CONFIG_SOC_XTAL_SUPPORT_40M=y
CONFIG_SOC_XTAL_SUPPORT_AUTO_DETECT=y
CONFIG_SOC_ADC_RTC_CTRL_SUPPORTED=y
CONFIG_SOC_ADC_DIG_CTRL_SUPPORTED=y
CONFIG_SOC_ADC_DMA_SUPPORTED=y
CONFIG_SOC_ADC_PERIPH_NUM=2
CONFIG_SOC_ADC_MAX_CHANNEL_NUM=10
CONFIG_SOC_ADC_ATTEN_NUM=4
CONFIG_SOC_ADC_DIGI_CONTROLLER_NUM=2
CONFIG_SOC_ADC_PATT_LEN_MAX=16
CONFIG_SOC_ADC_DIGI_MIN_BITWIDTH=9
CONFIG_SOC_ADC_DIGI_MAX_BITWIDTH=12
CONFIG_SOC_ADC_DIGI_RESULT_BYTES=2
CONFIG_SOC_ADC_DIGI_DATA_BYTES_PER_CONV=4
CONFIG_SOC_ADC_SAMPLE_FREQ_THRES_HIGH=2
CONFIG_SOC_ADC_SAMPLE_FREQ_THRES_LOW=20
CONFIG_SOC_ADC_RTC_MIN_BITWIDTH=9
CONFIG_SOC_ADC_RTC_MAX_BITWIDTH=12
CONFIG_SOC_SHARED_IDCACHE_SUPPORTED=y
CONFIG_SOC_IDCACHE_PER_CORE=y
CONFIG_SOC_CPU_CORES_NUM=2
CONFIG_SOC_CPU_INTR_NUM=32
CONFIG_SOC_CPU_HAS_FPU=y
CONFIG_SOC_CPU_BREAKPOINTS_NUM=2
CONFIG_SOC_CPU_WATCHPOINTS_NUM=2
CONFIG_SOC_CPU_WATCHPOINT_MAX_REGION_SIZE=64
CONFIG_SOC_DAC_CHAN_NUM=2
[... 2203 lines cut ...]
# CONFIG_ESP32_DEFAULT_PTHREAD_CORE_0 is not set
# CONFIG_ESP32_DEFAULT_PTHREAD_CORE_1 is not set
CONFIG_ESP32_PTHREAD_TASK_CORE_DEFAULT=-1
CONFIG_ESP32_PTHREAD_TASK_NAME_DEFAULT="pthread"
CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ABORTS=y
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_FAILS is not set
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ALLOWED is not set
# CONFIG_ESP32_ULP_COPROC_ENABLED is not set
CONFIG_SUPPRESS_SELECT_DEBUG_OUTPUT=y
CONFIG_SUPPORT_TERMIOS=y
CONFIG_SEMIHOSTFS_MAX_MOUNT_POINTS=1
CONFIG_MB_MASTER_TIMEOUT_MS_RESPOND=3000
CONFIG_MB_MASTER_DELAY_MS_CONVERT=200
CONFIG_MB_QUEUE_LENGTH=20
CONFIG_MB_SERIAL_TASK_STACK_SIZE=4096
CONFIG_MB_SERIAL_BUF_SIZE=256
CONFIG_MB_SERIAL_TASK_PRIO=10
CONFIG_MB_CONTROLLER_SLAVE_ID_SUPPORT=y
CONFIG_MB_CONTROLLER_SLAVE_ID=0x00112233
CONFIG_MB_CONTROLLER_NOTIFY_TIMEOUT=20
CONFIG_MB_CONTROLLER_NOTIFY_QUEUE_SIZE=20
CONFIG_MB_CONTROLLER_STACK_SIZE=4096
CONFIG_MB_EVENT_QUEUE_TIMEOUT=20
# CONFIG_MB_TIMER_PORT_ENABLED is not set
# End of deprecated options

lbernstone, 2024-08-16:
Those certificates are not valid pem files.  It's OK to obscure the contents of the private key & cert, but you should be posting a valid AWS CA. Does this code work with the pre-compiled libraries? Rather than posting the whole sdkconfig, please just indicate what you need to change from the arduino-esp32 defaults.

everslick, 2024-08-21:
BTW: Multi line source code should be enclosed in \`\`\` ... \`\`\` and **NOT** in \` ... \` for correct markdown.

TD-er, 2024-08-24:
Plase also check to see if your libraries call `.flush()` instead of `.clear()` as that has recently been changed here: https://github.com/espressif/arduino-esp32/pull/9453/files

IMHO (or may be IMnsHO) this PR should never have been done like this as it is actually breaking stuff all around causing lots of lost hours.

P-R-O-C-H-Y, 2024-11-14:
Hi @ioviedodev, I am closing this issue as there was no answers for a long time.
Feel free to reopen if your issue persists. Thanks

closed 2024-11-14 by P-R-O-C-H-Y