espressif/arduino-esp32 #10185: Add developer contributor guidelines to build and test our PR locally

mathieucarbou, 2024-08-16:
### Board

N/A

### Device Description

N/A

### Hardware Configuration

N/A

### Version

latest master (checkout manually)

### IDE Name

na

### Operating System

na

### Flash frequency

na

### PSRAM enabled

yes

### Upload speed

na

### Description

Hello,

I am preparing a PR to contribute middleware support (aka Expressif) to WebServer, plus some examples and middleware implementations (cors, security, logging). 

I didn't find any guidelines explaining the process to compile our changes and test the examples we write on a real board before submitting the PR.

Right now, I work around that by adding some platformio.ini files and using PIO to build the examples, but this is quite a hack.

What contributors should do to locally test and run the examples we write against the modifications we did ?

Thanks :-)

### Sketch

```cpp
na
```

### Debug Message

```plain
na
```

### Other Steps to Reproduce

na

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2024-08-16:
@lucasssvaz @P-R-O-C-H-Y any way this can be tested with wokwi?

Jason2866, 2024-08-16:
@mathieucarbou Thx, for bringing this up. After the big refactoring of the GH Actions (btw. great work imho) i do not manage to test my changes anymore before opening a PR ;-)

mathieucarbou, 2024-08-16:
> @mathieucarbou Thx, for bringing this up. After the big refactoring of the GH Actions (btw. great work imho) i do not manage to test my changes anymore before opening a PR ;-)

I've tried running `github/scripts/on-push.sh` locally, it succeeded, which proves it didn't consider my changes... 🤣

PR #10221 (2024-09-10): docs(contrib): Add instructions on how to test changes
## Description of Change

Add instructions on how to create and run tests.
Fix formatting.

## Tests scenarios

Built locally

## Related links

Closes #10185