espressif/arduino-esp32 #10189: VS Code Intellisense breaks

khryssmar, 2024-08-18:
### Board

ESP32-WROOM-32D

### Device Description

N/A

### Hardware Configuration

N/A

### Version

v3.0.4

### IDE Name

Arduino

### Operating System

Windows 10

### Flash frequency

80Mhz

### PSRAM enabled

yes

### Upload speed

921600

### Description

I dont know if the issue is for this repo but the VS  Code intellisense breaks when using the 3.0.4 version. I tried it in 2.0.17 and it is just fine. Seems like the paths the got included in the JSON file is not nearly the same number as that from the 2.0.17.

from 2.0.17:
[image]

from 3.0.4:
[image]
[image]

### Sketch

```cpp
N/A
```

### Debug Message

```plain
N/A
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Jason2866, 2024-08-18:
Since *.ino files are not c nor c++ intellisense does not work reliable. This is a known limitation of VSC. Can only be solved by switching of Intellisense.

khryssmar, 2024-08-18:
*.ino is treated as cpp. Also, it Intellisense breaks due to some include files that are not added to the include list.
[image]

Is there any reason why the previous version has the complete include list and newer version is not?

Jason2866, 2024-08-18:
This github is for issues regarding Arduino code. For issues and questions around VSC wrong place.

closed 2024-08-18 by Jason2866