espressif/arduino-esp32 #10195: SD card not working when using Wifi AP. Working fine when connecting to an existing network.

KobeOne2, 2024-08-18:
### Board

ESP32-S3

### Device Description

ESP32-S3-WROOM-1

### Hardware Configuration

#define SCK 18
#define MISO 19
#define MOSI 23
#define CS 5

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

80 MHz

### PSRAM enabled

no

### Upload speed

115200

### Description

It appears that the SD card only works when not using an access point:

When an access point is created the sd card stops working.

WiFi.softAP("FHD", "123456789");

The SD card functions well as long as no AP is used. Do you have any idea what's going on here?

### Sketch

```cpp
#include <WiFi.h> //Built-in #include <ESP32WebServer.h> //https://github.com/Pedroalbuquerque/ESP32WebServer download and place in your Libraries folder #include <ESPmDNS.h>

#include "CSS.h" //Includes headers of the web and de style file #include <SD.h> #include <SPI.h>

#define SCK 18 //48 #define MISO 19 //36 #define MOSI 23 //35 #define CS 5 //37

ESP32WebServer server(80);

#define servername "FHD33" //Define the name to your server...

bool SD_present = false; //Controls if the SD card is present or not

/********* SETUP **********/

void setup(void) {
Serial.begin(115200);
WiFi.softAP("FHD", "123456789"); //Network and password for the access point genereted by ESP32 ////
 if (!MDNS.begin(servername)) {
Serial.println("Error setting up MDNS responder!"); ESP.restart(); }

Serial.print("Initializing SD card...");

if (!SD.begin(CS)) { 
Serial.println("Card failed or not present, no SD Card data logging possible..."); 
SD_present = false; 
} 
else { 
Serial.println(F("Card initialised... file access enabled...")); 
SD_present = true; 
}

server.begin();

Serial.println("HTTP server started"); }

/********* LOOP **********/

void loop(void) { 
server.handleClient(); //Listen for client connections
}
```

### Debug Message

```plain
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13836
load:0x40080400,len:3608
entry 0x400805f0
Initializing SD card...Card failed or not present, no SD Card data logging possible...
HTTP server started
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2024-08-18:
Please try this [code](https://wokwi.com/projects/406601450804838401).  Pin 48 is a bad choice if using a dev board, as that is where the built-in LED is attached.
```
#include <WiFi.h>
#include <SD.h>

#define SCK  12 //default for S3
#define MISO 13
#define MOSI 11
#define CS   10

void setup(void) {
  Serial.begin(115200);
  WiFi.softAP("FHD", "123456789");
  Serial.print("Initializing SD card...");
  SPI.begin(SCK, MISO, MOSI, CS);
  if (!SD.begin(CS)) {
    Serial.println("SDCard failed");
  } else {
    Serial.println("SDCard initialised");
  }
}
```

KobeOne2, 2024-08-19:
Thanks a lot!

Indeed the default pins seem to work. Do you maybe know which pins should be avoided for the SD card so the Wifi works.

I actually made a mistake in the original post the pins I'm currently using are:
#define SCK 2
#define MISO 21
#define MOSI 45
#define CS 35

lbernstone, 2024-08-19:
I'm [unable to reproduce](https://wokwi.com/projects/406620419624344577)
Put a delay(100) in after you start the AP.  Turn core debugging level to verbose and see if you get more info in the console.

everslick, 2024-08-21:
If I had to guess, I'd also say power supply.

VojtechBartoska, 2024-08-21:
@P-R-O-C-H-Y Please help with triage, thanks

Jason2866, 2025-03-12:
Closing since not reproducable and outdated. Feel free to ask to reopen if the issue still exists with actual Arduino core

closed 2025-03-12 by Jason2866