espressif/arduino-esp32 #10196: Wrong pin number of SPI and I2C for LOLIN(Wemos) S2 MINI

peaechan, 2024-08-19:
### Board

LOLIN S2 MINI

### Device Description

LOLIN(Wemos) S2 MINI

### Hardware Configuration

GPIO36 & 35 connected to SCK and MOSI

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE 2.3.2

### Operating System

Windows 11

### Flash frequency

240MHz

### PSRAM enabled

yes

### Upload speed

921600

### Description

I have a small project using ePaper with LOLIN(Wemos) S2 mini. Initially, GPIO36 & 35 connected to SCK and MOSI using attached pinout diagram but it won't work. Then, I found that default SPI pins in arduino IDE are not correct. So, I have modified the pin number of RXTX, SPI and I2C in pins_arduino.h.

Original:
```
static const uint8_t TX = 39;
static const uint8_t RX = 37;

static const uint8_t SDA = 33;
static const uint8_t SCL = 35;

static const uint8_t SS = 12;
static const uint8_t MOSI = 11;
static const uint8_t MISO = 9;
static const uint8_t SCK = 7;
```
Modified:
```
static const uint8_t TX = 39;
static const uint8_t RX = 40;

static const uint8_t SDA = 8;
static const uint8_t SCL = 9;

static const uint8_t SS = 34;
static const uint8_t MOSI = 35;
static const uint8_t MISO = 37;
static const uint8_t SCK = 36;
```
[image]

### Sketch

```cpp
None
```

### Debug Message

```plain
None
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2024-08-19:
Not sure where you got that image from.  The [Wemos reference](https://www.wemos.cc/en/latest/s2/s2_mini.html) does not include those markings, so I'm going to say the board mfr is correct here.

peaechan, 2024-08-20:
FYI. The image is come from [here](https://www.studiopieters.nl/esp32-s2/).

Jason2866, 2024-08-20:
Clearly not branded as LOLIN(Wemos) S2 MIN. Yours is a StudioPieters board.

bartbergman, 2025-02-23:
Well, It's not a "Studio Pieters" board, this company doesn't produce any boards.
But it might be a clone from China that uses a different pin mapping apparently?
The changes that @peaechan made work perfectly for me as well, thanks!

closed 2025-03-12 by Jason2866