espressif/arduino-esp32 #10209: Interrupt use wrong pin on Arduino Nano ESP32

andreock, 2024-08-20:
### Board

Arduino Nano ESP32

### Device Description

Arduino Nano ESP32

### Hardware Configuration

Pin D5 is set in INPUT_PULLUP mode and connected to a push button to GND.

### Version

v3.0.4

### IDE Name

PlatformIO

### Operating System

Arch Linux

### Flash frequency

80Mhz

### PSRAM enabled

no

### Upload speed

460800

### Description

When i use the function attachInterrupt(digitalPinToGPIONumber(D5), change_state, CHANGE); the pin that handle interrupt is the D8 pin despite I use digitalPinToGPIONumber(D5). On the same board interrupt worked until about begin of August then stopped to work, I also rollback to 2.0.16 version of arduino-esp32 but still no works. Can be a problem of the board?

### Sketch

```cpp
#include <Arduino.h>

bool test = false;

void ARDUINO_ISR_ATTR change_state() {
  test = true;
}

void setup() {
  Serial.begin(9600);
  pinMode(D5, INPUT_PULLUP);
  attachInterrupt(digitalPinToGPIONumber(D5), change_state, CHANGE);
}

void loop() {
  if(test)
    Serial.println("Hello");
  Serial.println("World");
  delay(1000);
}
```

### Debug Message

```plain
No guru meditation error happened
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2024-08-21:
@pillo79 would this be the best way to attach the interrupt?

pillo79, 2024-08-21:
Hello @andreock, I suspect the problem is your use of the `digitalPinToGPIONumber` function. These should really be used only when directly writing to a CPU register or something like that; for Arduino APIs, this detail should be handled by the core for you, transparently, regardless of the [Pin Mapping](https://docs.arduino.cc/tutorials/nano-esp32/pin-setup) setting in the IDE. 

Can you simply try with `attachInterrupt(D5, ...)`?

andreock, 2024-08-21:
Sorry, I made a typo in the previous question, I use digitalPinToInterrupt not digitalPinToGPIONumber. By the way if I use attachInterrupt(D5, ...) with the current SDK(3.0.4) works. I was sure I had tried yesterday attachInterrupt(D5,...) without digitalPinToInterrupt(), Sorry for the useless issue.

closed 2024-08-21 by andreock