espressif/arduino-esp32 #10220: Esp Now stop working after random days and freeze the esp32

Armagedon13, 2024-08-22:
### Board

NodeMCU-32S

### Device Description

[image]

### Hardware Configuration

#ifndef ETH_PHY_TYPE
#define ETH_PHY_TYPE ETH_PHY_W5500
#define ETH_PHY_ADDR 1
#define ETH_PHY_CS 15
#define ETH_PHY_IRQ 4
#define ETH_PHY_RST 5
#endif

// SPI pins
#define ETH_SPI_SCK 14
#define ETH_SPI_MISO 12
#define ETH_SPI_MOSI 13

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE 2.3.2

### Operating System

Windows 11

### Flash frequency

80 MHz

### PSRAM enabled

yes

### Upload speed

921600

### Description

i have 2 Esp32 nodemcu, they connect each other with esp now that send 2 thins: 
The first is the heartbeat that check if the connection between the 2 esp exist and show a led status.
The second is the weight scale data that send it from ethernet udp to the esp32 and the esp32 send it over esp32 to the second one.
the program works really well, send all the data, and says the connection is stablished, but after 2 or more days the esp32 stops working, the led keep blinking like the connection exist, but the esp32 not response, i have a switch that you can change the esp32 to habilitate the OTA and wifi AP so you can program it. but the switch don't work is like it get stucked, the worst part is i put a Watchdog but i think it didin't work, i can't see the log of the problem because if i connect to usb start working normally, i change the power supply 3 times and i have a  capacitor filter, all this in 5v in VIN

### Sketch

```cpp
/*
Transmisor
*/

#ifndef ETH_PHY_TYPE
#define ETH_PHY_TYPE ETH_PHY_W5500
#define ETH_PHY_ADDR 1
#define ETH_PHY_CS 15
#define ETH_PHY_IRQ 4
#define ETH_PHY_RST 5
#endif

// SPI pins
[... 541 lines cut ...]
------------------------------------------
GPIO Info:
------------------------------------------
  GPIO : BUS_TYPE[bus/unit][chan]
  --------------------------------------  
     1 : UART_TX[0]
     3 : UART_RX[0]
     4 : ETH_IRQ[0]
     5 : ETH_RST[0]
    12 : SPI_MASTER_MISO[3]
    13 : SPI_MASTER_MOSI[3]
    14 : SPI_MASTER_SCK[3]
    15 : ETH_CS[0]
    19 : GPIO
    21 : GPIO
    22 : GPIO
    23 : GPIO
============ After Setup End =============
Heartbeat sent to Receiver
Connection to Receiver established
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Armagedon13, 2024-08-22:
@me-no-dev ok, im gonna edit it

Armagedon13, 2024-08-22:
> we have our own ESP_NOW library for Arduino

the one i made is wrong? and there is no way to know if they are connected each other, thats why i made the heartbeat for that reason, to get a feedback with leds.
I see the ESP example, and it's new, it's not too easy to set up.

Jason2866, 2024-08-22:
I had sometimes hangs when using the W5500 with Interrupt. Since changed to use the W5500 **without** Interrupt it works fine

seansimmons74, 2024-08-23:
Forgive me if this is unrelated but I noticed you are seeing a ~50k drop in free bytes/min free bytes.  I am also having this issue and it never seems to recover (eventually leading to the board crashing/hanging).  The first drop is 50k and then the rest are much smaller but it eventually causes my ESP32 to hang/perform slowly due to the continued drop in free heap.  I am making http POST requests and I noticed that SOMEWHERE (still trying to narrow it down) in the HTTPClient.cpp file around this area: `    if(!_client->connect(_host.c_str(), _port, _connectTimeout)) {` 
something happens causing my free heap to drop.  If this is irrelevant, please disregard.

me-no-dev, 2024-08-23:
@seansimmons74 it is irrelevant and should be fixed in 3.0.4

Armagedon13, 2024-08-23:
Is there any way to know the ESP NOW overflow? I tried sending more than the limit, the protocol stops working until the Esp32 is restarted, the size of the weight scale data never exceeds 70 bytes. I changed the heartbeat to only send one recognition byte, it does this every 5 seconds, maybe this overflows the protocol? I was testing if there are memory overflows but I don't see any errors.

me-no-dev, 2024-08-23:
It is very hard to say, given that you do not provide any logs :) why not look into our library and try to use that instead.

Armagedon13, 2024-08-23:
@me-no-dev well... i see the examples but i don't understand how to use it, to make this i search over internet and see a good tutorial [this](https://randomnerdtutorials.com/esp-now-esp32-arduino-ide/), i just wanna make a bidirectional communication like i made, but im gonna try to make it.

me-no-dev, 2024-08-23:
@P-R-O-C-H-Y will help you next week

lucasssvaz, 2024-08-23:
@Armagedon13 Maybe this helps: https://developer.espressif.com/blog/arduino-esp-now-lib/

seansimmons74, 2024-08-27:
> @seansimmons74 it is irrelevant and should be fixed in 3.0.4

@me-no-dev I'm still pretty new to all this but are you saying Arduino Core v3.0.4?  I'm using platform IO and I'm a bit confused on how to upgrade to that.  I see that even the latest espressif32 platform version is v6.8.0 and that only supports v2.0.17 of Arduino.  Can you point me in the right direction to get this stuff updated to the proper version, please?  Also, can you drop a link to the patch that you say fixes the problem in 3.0.4, please?  Thank you for your help.

Jason2866, 2024-08-27:
@seansimmons74  The community fork pioarduino supports actual core 3.0.4 with Platformio 
https://github.com/pioarduino/platform-espressif32

seansimmons74, 2024-08-27:
> @seansimmons74 The community fork pioarduino supports actual core 3.0.4 with Platformio https://github.com/pioarduino/platform-espressif32

@Jason2866 Ok I managed to update and am running 3.0.4.  Also verified that the HTTPClient and WiFi libraries updated along with it (Build shows dependencies @ 3.0.4).  I am still having the issue I mentioned before.  Any other advice would be appreciated.  If I need to create a proper issue somewhere, can you let me know where the right place might be?

me-no-dev, 2024-08-28:
@seansimmons74 please open a new issue with minimal example to reproduce and good explanation on what and how it's happening.

Jason2866, 2025-03-12:
Closing since outdated. Feel free to ask to reopen if the issue still exists with actual Arduino core.

closed 2025-03-12 by Jason2866