espressif/arduino-esp32 #10227: Path from PC used to compile code included in compiled firmware WHY?

JustTryingToGetSomeWorkDone, 2024-08-23:
### Board

ESP32-C3-mini-1 on custom prototype board

### Device Description

connected and in use:
128x32 OLED with SSD1306 via i2c on pins 0 and 1
1 momentary switch on pin 4

connected but unused:
DS3231 module is powered and connected to i2c, but currently unused
there is a mosfet circuit on the board that is currently unused on pin 10

### Hardware Configuration

USB CDC is connected and being used for serial communication and programming

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE 2.3.2 with ESP32 core 3.0.4

### Operating System

Windows 10

### Flash frequency

80 MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

I'm making 2 devices. One is an LED dimmer, the other is a controller for the dimmer. I'm attempting to use ESPNOW to communicate via unencrypted broadcasts. This is very much a work in progress with the aim being get something sending messages, change the message, then upgrade the interface and functionality incrementally. I started with the example code: ESP_NOW Broadcast Master, to which I have added some libraries for a button I use to navigate on this early prototype, and an OLED which I plan to eventually upgrade to something else. I'm currently at the change the message stage on the controller device. Previous tests at sending the example code messages worked fine. Even when I swapped out all the serial prints to display prints, no problem.

I'm attempting to transmit a struct that contains information about whether or not the LEDs are connected, their duty cycles in % of 100, stuff like that. I'm just beginning this step, so after a little hassle with pointers and casting to an unsigned integer, it compiles and looks ok. All I want it to do at this point is when I click a button, the value of the duty cycle should be incremented by 10 and displayed. If it goes over 100, it loops back to 0. Double click and it sends the message. However, from the get go it displays things I do not expect. Other parts of the UI show up in the place of the duty cycle which should be an int. If I click, it shows a frowny face, which I'm quite sure isn't in my code. If I click again and again and again, it starts showing the path to the wire.h library on the PC that I used to compile the firmware: /user/blah/AppData/Local/Arduino15/Package/esp32... something along those lines. It's like it's scrolling when it should be incrementing the initial value of 5 by 10.

So MY main issue and question is: why on earth is it including my PC paths in the firmware? I get that this seems like a buffer issue or memory/stack issue. I'll figure that part out, but having private data included in the compiled and flashed firmware seems like a HUGE privacy issue. Is this the fault of the Arduino IDE or the ESP32 core?

### Sketch

```cpp
#define DEBUGCODE 

#ifdef DEBUGCODE
#include <Arduino.h>
#endif

#include <Wire.h> // regular wire library for esp32
#include <Adafruit_SSD1306.h> // library for the oled I'm using
#include <Adafruit_GFX.h> // library for the oled I'm using
#include <BfButton.h> // https://github.com/mickey9801/ButtonFever 
#include <Preferences.h> // standard esp NVS saving library
#include <DS3231-RTC.h> //  https://github.com/hasenradball/DS3231-RTC
#include "ESP32_NOW.h"
#include "WiFi.h"

#include <esp_mac.h>  // For the MAC2STR and MACSTR macros

[... 257 lines cut ...]
  myMessage.LED3duty = 5;
  myMessage.LED4duty = 5;
  myMessage.factoryReset = false;

}

void loop() {
  btn.read();
  OLED();
}
```

### Debug Message

```plain
No error messages
```

### Other Steps to Reproduce

I don't have any other information about reproducing this issue. I suppose you should know if you're including local PC paths in compiled firmware that does not require such information.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

JustTryingToGetSomeWorkDone, 2024-08-23:
Well, never look behind the curtain, I guess. I started to wonder if this was due to my error reporting settings or something, so I set the Core Debug Level to None, recompiled, and checked the resulting .elf in Notepad++. The file is riddled with paths from my computer right there in plain text. Why is this necessary? Which settings can I deactivate to make sure this type of information doesn't get sent to the world if I decide to share my work?

The fact that I've never heard about this is what I find the most disturbing. I'm surely not the first to look inside an elf file. Should there not at least be some warning that this can happen.

Disabling the JTAG debugger has no effect either. I'm starting to think there's so way to remove this from the file. Considering the ESP will never need this information, it seems like a huge breach of privacy to me. If it hadn't popped up on my OLED I would have never even checked.

JustTryingToGetSomeWorkDone, 2024-08-23:
I have determined this is the fault of the Arduino IDE. Compiling the basic blink example sketch on non-ESP boards results in the same behavior. Closing as it does not belong here. I apologize for jumping the gun.

closed 2024-08-23 by JustTryingToGetSomeWorkDone

me-no-dev, 2024-08-23:
This is not dues to the IDE, but due to GCC. When you compile code, such information is included in the result in order to be able to trace code and errors. When we print logs from out Arduino, we strip that part of paths and do not print it, but if you do something like `Serial.println(__FILE__);` you will get the full path of the file you added that line into.