espressif/arduino-esp32 #10228: Set the correct reset sequence after flashing for CDC targets

peterdragun, 2024-08-23:
### Board

all boards with OTG (ESP32-S2, ESP32-S3)

### Device Description

devkit connected using native USB and OTG mode enabled

### Hardware Configuration

no other HW required

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE

### Operating System

macOS, but can be reproduced anywhere

### Flash frequency

N/A

### PSRAM enabled

yes

### Upload speed

N/A

### Description

When flashing using USB-OTG esptool is not able to reset the target. The default behavior is that esptool tries to reset the target to run the application. Because this is not available when connected using OTG esptool fails with an error that this was not done because it was not able to reset the target. Arduino gets this message and appends ''uploading error" even though uploading was successful and the only issue is that the chip wasn't reset.

There are two ways how this can be improved from the Arduino side:
1. Pass the correct `--after` argument when CDC is selected
2. Catch the error message and adjust the output accordingly - instead of 'Failed to upload' to something like 'Failed to reset...'

### Sketch

```cpp
any sketch, e.g. Blink example
```

### Debug Message

```plain
Warning: ESP32-S2 (revision v0.0) chip was placed into download mode using GPIO0.
esptool.py can not exit the download mode over USB. To run the app, reset the chip manually.
To suppress this note, set --after option to 'no_reset'.
Failed uploading: uploading error: exit status 1
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2024-08-23:
S2 should work. Doesn't it? This is when not manually entering DownloadMode, but using esptool to reset into download?

peterdragun, 2024-08-23:
Only if the previous application enabled the OTG port. If you are running just e.g. blink the port won't show up, so you need to switch to bootloader manually.

me-no-dev, 2024-08-29:
I used the following sketch to test and got different results.
- ESP32-S2 works as expected. It flashes and boots back into the firmware.
- ESP32-S3 flashes fine, but does not reboot into the firmware. No error from `esptool` is given, instead connecting the IDE back to the port fails.

Test sketch (CDC on Boot is enabled):
```cpp
void setup() {
  Serial.begin(115200);
}

void loop() {
  while (Serial.available()) {
    size_t l = Serial.available();
    uint8_t b[l];
    l = Serial.read(b, l);
    Serial.write(b, l);
  }
}
```

Error from the IDE for S3:
[image]

peterdragun, 2024-09-02:
I don't think you are using the correct ports on your devkits. For S2 this is easy, you just need a devkit (e.g. ESP32-S2-DevKitC-1) with two ports, usually, one is marked UART and the other one is USB.  And connect to the USB port. For S3 it is similar but by default, USB Serial/JTAG peripheral is connected to that port - so you need to change it in eFuse or by strapping the pin.

That way you should get the warning/error from esptool because there is no way to reset the target using a USB-CDC port.

me-no-dev, 2024-09-02:
Of course I am using the proper USB ports and in proper mode (OTG not JTAG/CDC) and CDC_ON_BOOT set to 1. The same as all our users. If I have to set an efuse, then we are going outside of the regular Arduino use case.

Both chips tested with TinyUSB booted and esptool commanding reboot to Download + flash.

closed 2024-10-14 by peterdragun