espressif/arduino-esp32 #10229: i2c scanner example  outputs dummy values or error codes, there is

jeo-apps, 2024-08-23:
### Board

esp32

### Device Description

esp32 wroom3d devkit v1

### Hardware Configuration

gpio pins 21 connected to sda and 22 to scl, issue occured during testing of mcp3424 adc module

### Version

v3.0.4

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

80 mhz

### PSRAM enabled

no

### Upload speed

115200

### Description

 I got error mesages at serial monitor  look like below, tested different modules both esp32 and adc

### Sketch

```cpp
#include "Wire.h"

void setup() {
  Serial.begin(115200);
  Wire.begin();
}

void loop() {
  byte error, address;
  int nDevices = 0;

  delay(5000);

  Serial.println("Scanning for I2C devices ...");
  for (address = 0x01; address < 0x7f; address++) {
    Wire.beginTransmission(address);
    error = Wire.endTransmission();
    if (error == 0) {
      Serial.printf("I2C device found at address 0x%02X\n", address);
      nDevices++;
    } else if (error != 2) {
      Serial.printf("Error %d at address 0x%02X\n", error, address);
    }
  }
  if (nDevices == 0) {
    Serial.println("No I2C devices found");
  }
}
```

### Debug Message

```plain
12:04:44.353 -> Scanning for I2C devices ...
12:04:44.353 -> I2C device found at address 0x04
12:04:44.353 -> Error 5 at address 0x05
12:04:44.353 -> I2C device found at address 0x68
12:04:49.382 -> Scanning for I2C devices ...
12:04:49.382 -> Error 5 at address 0x06
12:04:49.382 -> I2C device found at address 0x68
12:04:54.388 -> Scanning for I2C devices ...
12:04:54.388 -> I2C device found at address 0x04
12:04:54.388 -> Error 5 at address 0x05
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2024-08-23:
does it still show the bogus addresses when the MCP is not connected?

Jason2866, 2025-03-12:
closing since outdated. Feel free to ask to reopen if the issue is still there with actual Arduino core