espressif/arduino-esp32 #10253: Clarify the way LittleFS is meant to be used with partition schemes.

EternityForest, 2024-08-27:
### Related area

Documentation/UI

### Hardware specification

Board independent

### Is your feature request related to a problem?

LittleFS appears to work perfectly when used with a SPIFFS partition layout, but the fact that this works is not obvious, and I'm not exactly sure whether this is supported behavior or not, as I assume it's just using the wrong partition type and then ignoring the metadata flags.

For a new user this could potentially cause some confusion.

### Describe the solution you'd like

Either add LittleFS-specific menu options, or rename them to something "files" or "user data" if the partition type is meant to be ignored, or maybe just add a comment to littlefs example code, if this is in fact how it's meant to be used.

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

me-no-dev, 2024-08-28:
LittleFS is only recently supported as partition type and is in fact no different than SPIFFS partition (from internal point of view), so so far we have always used SPIFFS to kinda signify that the partition if used for storage. We could look into forcing to check for LittleFS type, but then we might break a lot of existing code (people seems to have moved away from using SPIFFS mostly)

@P-R-O-C-H-Y PTAL

EternityForest, 2024-08-28:
Yeah, I'm not a fan of breaking changes myself, and everything works great
now.

The current implementation follows the rule of least surprise, it works
exactly as I was hoping it would before I tried it, so I've got no
complaints about the actual functionality.

closed 2025-03-12 by Jason2866