espressif/arduino-esp32 #10266: ESP32-S3 WIFI connect too slow on V3

Nezaemmy, 2024-08-30:
### Board

ESP32-S3 DevKitC-1

### Device Description

ESP32-S3 DevKitC-1

### Hardware Configuration

NONE

### Version

v3.0.4

### IDE Name

Arduino IDE

### Operating System

Windows 11

### Flash frequency

80Mhz

### PSRAM enabled

no

### Upload speed

921600

### Description

I have 2 devices, esp32 and esp32-s3, on V2 all devices connect fast to the internet. 
on V3 esp32 connect first while esp32-s3 sometimes takes more than 2 minutes to join the AP

### Sketch

```cpp
Any sketck make the esp32-s3 connecting to the Access point
```

### Debug Message

```plain
none
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Jason2866, 2024-08-30:
Please provide a sketch. None of my esp32-s3 has issues to connect to an AP with core 3.0.4.

atanisoft, 2024-08-31:
Similarly all of my ESP32-S3 devices connect in under 15 seconds

closed 2024-09-02 by Nezaemmy