espressif/arduino-esp32 #10273: Need an interface to modify the Vendor (manufacturer) name and Product name of a USB HID device

Ccccraz, 2024-08-31:
### Related area

usb, usb hid

### Hardware specification

ESP32S3

### Is your feature request related to a problem?

As a newcomer to HID device development, I have recently been attempting to develop a custom HID project. Based on the existing functionality, I am able to complete the development of most features. However, I noticed that the Vendor name and Product name recognized in my operating system are "Espressif Systems" and "TinyUSB HID" respectively. I would like to be able to modify these two strings.

### Describe the solution you'd like

- As in esp-idf, by defining it in the following manner:
```cpp
const char* hid_string_descriptor[5] = {
    // array of pointer to string descriptors
    (char[]){0x09, 0x04},  // 0: is supported language is English (0x0409)
    "TinyUSB",             // 1: Manufacturer
    "TinyUSB Device",      // 2: Product
    "123456",              // 3: Serials, should use chip ID
    "Example HID interface",  // 4: HID
};
```
- Provide methods similar to the following:
```cpp
void set_vendor_name(char* name, size_t size);
void set_device_name(char* name, size_t size);
```

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

me-no-dev, 2024-08-31:
Here they are: https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/USB.h#L52-L89
```cpp
USB.manufacturerName("Your Name");
...
USB.begin();
```

closed 2024-08-31 by me-no-dev

Ccccraz, 2024-08-31:
Thank you very much for your answer!!!🎉