espressif/arduino-esp32 #10275: Add "ARDUINO_PARTITION_%s" define for use in  #ifdef  statements

Rob58329, 2024-09-01:
### Related area

added options

### Hardware specification

Basic NodeMCU_32S

### Is your feature request related to a problem?

https://github.com/HamzaHajeir/ESP32-Arduino-Builds/blob/master/tools/platformio-build-esp32.py has what appears to be a defined variable of form: " ARDUINO_PARTITION_%s " as shown below:

```
CPPDEFINES=[
     ...
        (...
        "ARDUINO_PARTITION_%s" % basename(env.BoardConfig().get(
            "build.partitions", "default.csv")).replace(".csv", "").replace("-", "_")
    ]

```
However https://github.com/espressif/arduino-esp32/blob/master/tools/platformio-build.py does not seem to create this "ARDUINO_PARTITION_%s" define.

Is it possible to add it?

[I have an ARDUINO-IDE project where I would like to change the compiled/used software parts dependant on which partition scheme is selected - and for this I need a "ARDUINO_PARTITION_%s" type define which I can use in a `#ifdef ARDUINO_PARTITION_abc` type statement.]

### Describe the solution you'd like

file https://github.com/espressif/arduino-esp32/blob/master/tools/platformio-build.py updated as detailed above.

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

Jason2866, 2024-09-01:
I don't get the use case. With Platformio you can set the partition scheme you want easily with the command `board_build.partitions = <every partition you want>` in platformio.ini

Jason2866, 2024-09-01:
Anyways about usefull or not it is there. https://github.com/espressif/esp32-arduino-libs/blob/1a029d52a9cbc5863b00d0316124d79be1de2922/esp32s2/platformio-build.py#L418

closed 2024-09-01 by Jason2866

Rob58329, 2024-09-01:
@Jason2866: Many thanks for your comments.

>The referenced python script is ONLY used with Platformio.

Ah, that's a pity, as I would like this definition to be available for use in a `#ifdef `statement in the ARDUINO-IDE.

>I don't get the use case.

The "Partition Scheme" setting (eg. using the "ESP32 Dev Module" board) allows you to select differing amounts of APP and SPIFFS space.  As my software is rather large, I want to be able to change what I compile (the features I include) depending on the various spaces available.  But for this I need a `#ifdef ARDUINO_PARTITION_abc `type statement (so depending on selection I can change what's included).

PS.  The link you included is actually for the ESP32s2, however the ESP32(original) https://github.com/espressif/esp32-arduino-libs/blob/1a029d52a9cbc5863b00d0316124d79be1de2922/esp32/platformio-build.py has similar code. 

However as you clarified that this file is only used in PlatformIO, and not in the ARDUINO-IDE, then I would still appreciate some similar functionality in the ARDUINO-IDE if this is possible.

Jason2866, 2024-09-01:
> However as you clarified that this file is only used in PlatformIO, and not in the ARDUINO-IDE, then I would still appreciate some similar functionality in the ARDUINO-IDE if this is possible.

No, it is ArduinoIDE ;-)  If you have a big Arduino project you should consider to switch to the Platformio fork pioarduino. 
https://github.com/pioarduino/platform-espressif32

me-no-dev, 2024-09-01:
It is available in Arduino IDE: https://github.com/espressif/arduino-esp32/blob/master/platform.txt#L147 `-DARDUINO_PARTITION_{build.partitions}`

Rob58329, 2024-09-27:
@me-no-dev - Very many thanks for this info!

If anyone want's to know how to do this, an example (using board="ESP32 Dev Module") is:

```
#if ARDUINO_PARTITION_default // from boards.txt: "esp32.menu.PartitionScheme.default=Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS)"
  Serial.println("ARDUINO_PARTITION_default being used");
  // #error
#elif ARDUINO_PARTITION_no_ota // from boards.txt: "esp32.menu.PartitionScheme.no_ota=No OTA (2MB APP/2MB SPIFFS)"
  Serial.println("ARDUINO_PARTITION_no_ota being used");
  // #error
#else
  #error "wrong Partition Scheme has been selected"
#endif
```