espressif/arduino-esp32 #10278: Support of ESP32-P4

VojtechBartoska, 2024-09-02:


VojtechBartoska, 2024-10-21:
Hello,

table updated, most of peripherals are already supported.

mutatrum, 2024-10-23:
The ESP32-P4 has 16 or 32MB PSRAM built in. Is that picked up automatically if configured?

me-no-dev, 2024-10-23:
@mutatrum yes. Just enable PSRAM and it will pick it up

Y1hsiaochunnn, 2024-10-31:
Hi @lucasssvaz ,I was wondering about the example program to drive es8311 using ESP32-P4 on Arduino and will it support ESP-ADF's numerous voice frames?

me-no-dev, 2024-10-31:
@Y1hsiaochunnn we do not import ADF, so if you want to use one of it's features, maybe you should use Arduino as IDF component.
Here is an example program with 8311 driver:
[ESP32P4_EV_I2S.zip](https://github.com/user-attachments/files/17585378/ESP32P4_EV_I2S.zip)

Y1hsiaochunnn, 2024-10-31:
Hi, @me-no-dev ,I tested the program and it worked fine. My ESP32-S3 SDK v3.0.7 tests also worked fine (pointer definition is a little different). I think this code is ready to be published in the sample application.

me-no-dev, 2025-01-28:
@hitecSmartHome it is. Same for all chips with PSRAM

hitecSmartHome, 2025-01-28:
> @hitecSmartHome it is. Same for all chips with PSRAM

Well, we have an esp wrover e with 8mb psram and it does not integrated into the memory map. We have to use allocators and the esp does not report it as one chunck

me-no-dev, 2025-01-28:
what version of the core and how are you checking?

marekm72, 2025-05-08:
Is the Parallel TX unit supported?  Not listed in the above table.
https://docs.espressif.com/projects/esp-idf/en/latest/esp32p4/api-reference/peripherals/parlio/parlio_tx.html

lbernstone, 2025-05-09:
> Is the Parallel TX unit supported? Not listed in the above table. [docs.espressif.com/projects/esp-idf/en/latest/esp32p4/api-reference/peripherals/parlio/parlio_tx.html](https://docs.espressif.com/projects/esp-idf/en/latest/esp32p4/api-reference/peripherals/parlio/parlio_tx.html)

There is no native OO/C++ code, but the includes & archives are there for the IDF driver, so you should be able to use it.

lyusupov, 2025-08-17:
Is there any progress with:
* Bluetooth LE  ?  - _build of basic examples fails now_
* ADC calibration ? -  _gives an error message on ``analogReadMilliVolts()``_

boomvalt, 2025-12-24:
> > Will ESP-NOW support be coming to the P4?
> 
> Our library will support ESP-NOW for P4 as soon as ESP-NOW is supported in ESP-Hosted.

Thank you for the reply. Is there an information on when it will be supported in EDP-Hosted ?

lucasssvaz, 2025-12-30:
@boomvalt I don't know the project timeline but there is already an issue open for it:

https://github.com/espressif/esp-hosted-mcu/issues/34

closed 2026-04-17 by me-no-dev