espressif/arduino-esp32 #10288: NetworkClientRxBuffer::clear() may not always clear

TD-er, 2024-09-04:
### Board

Any

### Device Description

-

### Hardware Configuration

-

### Version

latest master (checkout manually)

### IDE Name

PlatformIO

### Operating System

Windows 11

### Flash frequency

Any

### PSRAM enabled

yes

### Upload speed

115200

### Description

See current implementation of `clear()`:
https://github.com/espressif/arduino-esp32/blob/e70f4d3819500151af6a0adda561f09f647b3fc7/libraries/Network/src/NetworkClient.cpp#L149-L155

and `fillBuffer()`:
https://github.com/espressif/arduino-esp32/blob/e70f4d3819500151af6a0adda561f09f647b3fc7/libraries/Network/src/NetworkClient.cpp#L63-L88

When the rx buffer is full or there is more data available than would fit in the buffer, the buffer is not cleared.

Suggested code change:
```c++
  void clear() {
    if (r_available()) {
      _pos = _fill;
      while(fillBuffer())
       _pos = _fill;        
    }
    _pos = 0;
    _fill = 0;
  }
```

Only drawback I can think of with my suggested change is when there is a continuous amount of data to be cleared.
However this would in the current situation also have lead to issues.

### Sketch

```cpp
-
```

### Debug Message

```plain
-
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2024-09-11:
OK, I get what you mean, but I think that `_pos = _fill;` is wrong, because it will cause `fillBuffer()` to return `0`. Instead it should do the following to make sure that each iteration the buffer is marked empty
```cpp
void clear() {
  if (r_available()) {
    _pos = 0;
    _fill = 0;
    while(fillBuffer()){
      _pos = 0;
      _fill = 0;
    }
  }
  _pos = 0;
  _fill = 0;
}
```

me-no-dev, 2024-09-11:
ah my mistake. i read it wrong... yes. your suggestion makes sense

PR #10331 (2024-09-13): fix(networking): NetworkClientRxBuffer::clear() may not always clear
## Description of Change

See discussion in issue #10288

## Related links

Fixes: #10288
commit 8a87df3b (2024-09-13): NetworkClientRxBuffer::clear() may not always clear (#10288) (#10331)

Fixes: #10288
closed 2024-09-13 by me-no-dev