espressif/arduino-esp32 #10295: [Feature] Display in the console the network identifier to which the card is connected, in addition to the role

lboue, 2024-09-05:
### Related area

OpenThread

### Hardware specification

ESP32-C6, ESP32-H2

### Is your feature request related to a problem?

Hello,

I'm using this example: [RouterNode.ino](https://github.com/espressif/arduino-esp32/blob/master/libraries/OpenThread/examples/SimpleThreadNetwork/RouterNode/RouterNode.ino) to test the connection to my existing Thread network.

I want to make sure that my board is connected to the right network (PANID). 
Is there any way of accessing this information and displaying it in the console?

Regards

### Describe the solution you'd like

Display in the console the network identifier to which the card is connected, in addition to the role.

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

P-R-O-C-H-Y, 2024-09-05:
@SuGlider Can you please take a look?

lboue, 2024-09-05:
I've managed to display all the information using the otPrintNetworkInformation method, but the formatting seems incorrect. What could be the cause?
```c++
void loop() {
  Serial.print("Thread Node State: ");
  Serial.println(otGetStringDeviceRole());
  // Prints information about the current Thread network
  // https://github.com/espressif/arduino-esp32/blob/master/libraries/OpenThread/helper_functions.md
  Serial.println("Thread NetworkInformation: ");
  Serial.println("---------------------------");
  otPrintNetworkInformation(Serial);
  Serial.println("---------------------------");
  delay(5000);
}
```

Here is the result:

```
Thread Node State: Router
Thread NetworkInformation: 
---------------------------
Thread Setup:
Node State: 	25
Network Name: 	0x988f
Channel: 	a748dbb0ecc9****
Pan ID: 	7226b9ec8288c183a90c8ee53f******
Ext Pan ID: 	fd4f:21ef:e564:0:4b49:e030:bedf:****
fda7:48db:b0ec:0:0:ff:fe00:6000
fda7:48db:b0ec:0:500e:4336:b06a:****
fe80:0:0:0:1089:7be5:b795:***
Network Key: 	ff33:40:fda7:48db:b0ec:0:0:1
ff32:40:fda7:48db:b0ec:0:0:1
ff02:0:0:0:0:0:0:2
ff03:0:0:0:0:0:0:2
ff02:0:0:0:0:0:0:1
ff03:0:0:0:0:0:0:1
ff03:0:0:0:0:0:0:fc
Node IP Addresses are:
router
Node Multicast Addresses are:
MyHome**
---------------------------
```

lboue, 2024-09-05:
To compare, I manually entered the CLI commands with the example [SimpleCLI.ino](https://github.com/espressif/arduino-esp32/blob/master/libraries/OpenThread/examples/SimpleCLI/SimpleCLI.ino)

```
ESP-ROM:esp32c6-20220919
OpenThread CLI started - type 'help' for a list of commands.
...
ot> thread start
Done

ot> state
router
Done

ot> networkname
MyHome36
Done

ot> channel
25
Done

ot> panid
0x988f
Done

ot> extpanid
a748dbb0ecc9****
Done

ot> networkkey
7226b9ec8288c183a**************
Done

ot> ipaddr
fda7:48db:b0ec:0:0:ff:fe00:c400
fd4f:21ef:e564:0:2cdb:4175:afaa:c784
fda7:48db:b0ec:0:ca44:cb34:8d30:9b1d
fe80:0:0:0:6c66:9661:dce4:65c2
Done

ot> ipmaddr
ff33:40:fda7:48db:b0ec:0:0:1
ff32:40:fda7:48db:b0ec:0:0:1
ff02:0:0:0:0:0:0:2
ff03:0:0:0:0:0:0:2
ff02:0:0:0:0:0:0:1
ff03:0:0:0:0:0:0:1
ff03:0:0:0:0:0:0:fc
Done
```

SuGlider, 2024-09-05:
@lboue - Yes, there are two ways to obtain the PANID of the current connected network.

1) Using OpenThread API in C:
`otPanId otLinkGetPanId( otInstance *aInstance );`

2) Using the OpenThread CLI helper `otGetRespCmd(const char *cmd, char * resp);`
``` cpp
  char resp[32];
  otGetRespCmd("panid", resp);
``` 

It seems that there is a synchronization problem with the CLI method. 
Something may have changed in the lastest Arduino Core.
I need to investigate and fix it.

### Anyway, the first method works fine.

Example: 

``` cpp
#include "OThreadCLI.h"
#include "OThreadCLI_Util.h"

// Leader node shall use the same Network Key and channel
#define CLI_NETWORK_KEY    "dataset networkkey 00112233445566778899aabbccddeeff"
#define CLI_NETWORK_CHANEL "dataset channel 24"

void setup() {
  Serial.begin(115200);
  OThreadCLI.begin(false);  // No AutoStart - fresh start
  Serial.println("Setting up OpenThread Node as Router/Child");
  Serial.println("Make sure the Leader Node is already running");

  OThreadCLI.println("dataset clear");
  OThreadCLI.println(CLI_NETWORK_KEY);
  OThreadCLI.println(CLI_NETWORK_CHANEL);
  OThreadCLI.println("dataset commit active");
  OThreadCLI.println("ifconfig up");
  OThreadCLI.println("thread start");

  // wait for the node to enter in the router state
  uint32_t timeout = millis() + 90000; // waits 90 seconds to 
  bool nodeIsReady = true; 
  while (otGetDeviceRole() != OT_ROLE_CHILD && otGetDeviceRole() != OT_ROLE_ROUTER) {
    Serial.print(".");
    if (millis() > timeout) {
      Serial.println("\r\n\t===> Timeout! Failed.");
      nodeIsReady = false;
      break;
    }
    delay(500);
  }

  if (nodeIsReady) {
    // print the PanID using 2 methods

    // CLI
    char resp[32];
    if (otGetRespCmd("panid", resp)) {
      Serial.printf("\r\nPanID[using CLI]: %s\r\n", resp);
    } else {
      Serial.printf("\r\nPanID[using CLI]: FAILED!\r\n");
    }

    // OpenThread API
    Serial.printf("PanID[using OT API]: %x\r\n", (uint16_t) otLinkGetPanId(esp_openthread_get_instance()));
  }
}

void loop() {
}
```

SuGlider, 2024-09-05:
I found out that the CLI sends a "Done" for a previous command and this causes a shift with the CLI responses.
I'll find a way to fix that and send a PR.

As I wrote above, `otPanId otLinkGetPanId( otInstance *aInstance );` shall help you to get the PanID information.
`otPanId` is a `uint16_t` value.

lboue, 2024-09-05:
> I found out that the CLI sends a "Done" for a previous command and this causes a shift with the CLI responses. I'll find a way to fix that and send a PR.
>
> As I wrote above, `otPanId otLinkGetPanId( [...]

Thank you very much. I'll give it a try.

SuGlider, 2024-09-05:
> I've managed to display all the information using the [`otPrintNetworkInformation`](https://github.com/espressif/arduino-esp32/blob/master/libraries/OpenThread/src/OThreadCLI_Util.cpp) method, but [...]

OK.... I got the issue.

The code executes many `OThreadCLI.println("CMD...");` in order to make it start as Child/Router.
Each of those `OThreadCLI.println()` will get a `Done` as response, but those will be buffered and it will be read by `otGetRespCmd(CMD, char *resp)` in the wrong sequence.

In order to fix it, the code shall be this:

``` cpp
#include "OThreadCLI.h"
#include "OThreadCLI_Util.h"

// Leader node shall use the same Network Key and channel
#define CLI_NETWORK_KEY    "dataset networkkey 00112233445566778899aabbccddeeff"
#define CLI_NETWORK_CHANEL "dataset channel 24"
bool otStatus = true;

void setup() {
  Serial.begin(115200);
  OThreadCLI.begin(false);  // No AutoStart - fresh start
  Serial.println("Setting up OpenThread Node as Router/Child");
  Serial.println("Make sure the Leader Node is already running");

  char resp[256];
  otStatus &= otGetRespCmd("dataset clear", resp);
  otStatus &= otGetRespCmd(CLI_NETWORK_KEY, resp);
  otStatus &= otGetRespCmd(CLI_NETWORK_CHANEL, resp);
  otStatus &= otGetRespCmd("dataset commit active", resp);
  otStatus &= otGetRespCmd("ifconfig up", resp);
  otStatus &= otGetRespCmd("thread start", resp);

  if (!otStatus) {
    Serial.println("\r\n\t===> Failed starting Thread Network!");
    return;
  }
  // wait for the node to enter in the router state
  uint32_t timeout = millis() + 90000; // waits 90 seconds to
  while (otGetDeviceRole() != OT_ROLE_CHILD && otGetDeviceRole() != OT_ROLE_ROUTER) {
    Serial.print(".");
    if (millis() > timeout) {
      Serial.println("\r\n\t===> Timeout! Failed.");
      otStatus = false;
      break;
    }
    delay(500);
  }

  if (otStatus) {
    // print the PanID using 2 methods

    // CLI
    if (otGetRespCmd("panid", resp)) {
      Serial.printf("\r\nPanID[using CLI]: %s\r\n", resp);
    } else {
      Serial.printf("\r\nPanID[using CLI]: FAILED!\r\n");
    }

    // OpenThread API
    Serial.printf("PanID[using OT API]: 0x%x\r\n", (uint16_t) otLinkGetPanId(esp_openthread_get_instance()));
  }
  Serial.println("\r\n");
}

void loop() {
  if (otStatus) {
  Serial.println("Thread NetworkInformation: ");
  Serial.println("---------------------------");
  otPrintNetworkInformation(Serial);
  Serial.println("---------------------------");
  } else {
    Serial.println("Some OpenThread operation has failed...");
  }
  delay(10000);
}
```

### Ouput
```
ESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:2
load:0x4086c410,len:0xcf8
load:0x4086e610,len:0x2e44
load:0x40875728,len:0x113c
entry 0x4086c410
Setting up OpenThread Node as Router/Child
Make sure the Leader Node is already running
.
PanID[using CLI]: 0x7393

PanID[using OT API]: 0x7393

Thread NetworkInformation: 
---------------------------
Thread Setup:
Node State: 	router
Network Name: 	OpenThread-7393
Channel: 	24
Pan ID: 	0x7393
Ext Pan ID: 	d77d766c73225b38
Network Key: 	00112233445566778899aabbccddeeff
Node IP Addresses are:
fd84:d563:f6b6:988e:0:ff:fe00:e400
fd84:d563:f6b6:988e:a653:978e:e102:ec6b
fe80:0:0:0:5804:96be:4317:2572
Node Multicast Addresses are:
ff33:40:fd84:d563:f6b6:988e:0:1
ff32:40:fd84:d563:f6b6:988e:0:1
ff02:0:0:0:0:0:0:2
ff03:0:0:0:0:0:0:2
ff02:0:0:0:0:0:0:1
ff03:0:0:0:0:0:0:1
ff03:0:0:0:0:0:0:fc
---------------------------
```

SuGlider, 2024-09-05:
I'll add this example to the Arduino Core to help future users.

PR #10299 (2024-09-09): OpenThread Example Improvement
## Description of Change

Adds a new example that mixes OpenThread native API with different Arduino OThreadCLI API calls.
New example is `ExtendedRouterNode.ino`.

Adds OpenThread Native API calls to the current Router and Leader Examples.

Fixes the Library `keywords.txt` file that didn't use TAB to separate fields.

## Tests scenarios
Tested using two ESP32-C6 one for the Leader and the other for the Child/Router.
Used the modified examples.

## Related links
Closes #10295 
Related to #9486
closed 2024-09-09 by me-no-dev