espressif/arduino-esp32 #10297: setToneChannel is declared but not implemented in v3.0.4

vicatcu, 2024-09-05:
### Board

Heltec Wi-Fi LoRa 32

### Device Description

Custom hardware

### Hardware Configuration

Irrelevant

### Version

v3.0.4

### IDE Name

Arduino IDE

### Operating System

Ubuntu 22.04

### Flash frequency

Irrelevant

### PSRAM enabled

no

### Upload speed

Irrelevant

### Description

Compiling a sketch that uses setToneChannel fails with a linker error like `undefined reference to `_Z14setToneChannelh'`. 

`grep -r setToneChannel .` within the folder `~/.arduino15/packages/esp32/hardware/esp32/3.0.4` reveals a declaration for setToneChannel, but no imiplementation.

### Sketch

```cpp
void setup() {
  setToneChannel(0);  
}

void loop() {

}
```

### Debug Message

```plain
Irrelevant, compile time issue.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2024-09-06:
@vicatcu - Please try the PR #10305

vicatcu, 2024-09-06:
I can confirm my sketch builds in v3.0.4 if I replace Tone.h with the one in this commit...

closed 2024-09-09 by me-no-dev