espressif/arduino-esp32 #10301: hostByName(): DNS Failed for 'xyz.co.in' with error '-54' in NetworkManager.cpp:131

demirschmid61, 2024-09-06:
### Board

ESP32 Dev Board

### Device Description

Dev Board

### Hardware Configuration

No

### Version

v3.0.2

### IDE Name

PlatformIO

### Operating System

Windows10

### Flash frequency

40Mhz

### PSRAM enabled

yes

### Upload speed

115200

### Description

Hello sir, 

I have migrated and using Arduino-esp32 **version 3.0.2** on platformio and facing the issue of DNS Failed from NetworkManager.cpp.

When I connect to Wi-Fi for the first time, it gets connected by fetching the proper IP address of the particular domain. 
When the Wi-Fi is turned OFF and ON again, the NetworkManager.cpp library does not fetch the IP address of the domain provided; it shows DNS failed.

The logs are shown below.

/***********************When WiFi Connected first-time *******************/
14:01:39.549 > [MO] info (main.cpp:1055): WiFi Connected
14:01:41.546 > [MO] info (main.cpp:4554): [  6246][E][NetworkManager.cpp:83] hostByName(): Clearing DNS cache
14:01:41.878 > [  6578][E][NetworkManager.cpp:123] hostByName(): DNS found IPv4 13.210.161.8

********************Now wifi is turned OFF and ON again ******************
14:03:41.301 >  WiFi connected  **(wifi connected here)**
14:03:46.079 > [130777][E][NetworkManager.cpp:131] hostByName(): DNS Failed for 'ocpp.chqtest.net' with error '-54'
14:03:46.089 > [130787][E][NetworkClient.cpp:242] connect(): connect on fd 48, errno: 118, "Host is unreachable"
14:03:51.099 > [135799][E][NetworkManager.cpp:131] hostByName(): DNS Failed for 'ocpp.chqtest.net' with error '-54'
14:03:51.113 > [135808][E][NetworkClient.cpp:242] connect(): connect on fd 48, errno: 118, "Host is unreachable"
14:03:56.121 > [140820][E][NetworkManager.cpp:83] hostByName(): Clearing DNS cache
14:04:03.127 > [147826][E][NetworkManager.cpp:131] hostByName(): DNS Failed for 'ocpp.chqtest.net' with error '-54'
14:04:20.153 > [164852][E][NetworkManager.cpp:131] hostByName(): DNS Failed for 'ocpp.chqtest.net' with error '-54'
*****************************************************************************

Could you please check and let me know where I should check the problem or should I upgrade the Arduinoesp32 library version version?

### Sketch

```cpp
.
```

### Debug Message

```plain
.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

demirschmid61, 2024-09-07:
@JAndrassy 
Perfect, sir. 
If I use the IP address instead of the domain, it connects perfectly when the Wi-Fi is turned back on.

demirschmid61, 2024-09-07:
@JAndrassy 
One more observation, whenever code try to get IP address of the domain

**[130777][E][NetworkManager.cpp:131] hostByName(): DNS Failed for 'ocpp.chqtest.net' with error '-54'** 

Other loop execution is also stop for a while.

me-no-dev, 2024-09-09:
Please update your core to 3.0.4. There were some fixes for DNS in environments with IPv6 around

dipankarkhartad, 2024-09-10:
@me-no-dev Thank you very much sir for your reply.

I'll update the core to 3.0.4 and let you know the observations.

demirschmid61, 2024-09-10:
@me-no-dev Thank you very much sir for your replay,

I'll just update the core to 3.0.4 and let you know the observations.

closed 2025-03-12 by Jason2866