espressif/arduino-esp32 #10303: ESP32C2什么时候能支持Arduino IDE呢

chen70854022, 2024-09-06:
### Related area

ESP32C2

### Hardware specification

ESP32C2

### Is your feature request related to a problem?

了解到ESP32C2的价格相比C3优势很大，但是发现好像Arduino并不支持C2，请问什么时候能支持呢

### Describe the solution you'd like

可以在Arduino中使用C2

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

SuGlider, 2024-09-06:
It is already in the IDE, but it is hidden.
In order to make it visible, you can change the `boards.txt` file in the line 48
https://github.com/espressif/arduino-esp32/blob/master/boards.txt#L48

change it to `esp32c2.hide=false` or remove the line.

VojtechBartoska, 2024-09-12:
Any updates on this @chen70854022? Is it still valid? thanks

closed 2025-03-12 by Jason2866