espressif/arduino-esp32 #10307: ESP32 {S2, S3} CDC and HID stall when using simultaneously

ChrGri, 2024-09-07:
### Board

ESP32 S3 dev and S2 mini

### Device Description

-

### Hardware Configuration

USB-OTG port used

### Version

latest master (checkout manually)

### IDE Name

PlatformIO

### Operating System

Win 11

### Flash frequency

-

### PSRAM enabled

yes

### Upload speed

-

### Description

This issue was found during investigation of [this ticket](https://github.com/espressif/arduino-esp32/issues/9582). I've opened this ticket, to clean up the issue tracking.

The issue is, that when sending signals over serial CDC and HID at the same time, at least one of the channels will stall after a while and therefore isn't sending anymore data to the PC.

### The steps to reproduce this issue: 
1) take ESP32 S2 mini or ESP32 S3 (other boards might behave similar)
2) upload the software with simultaneous CDC and HID output from here
https://github.com/ChrGri/DIY-Sim-Racing-FFB-Pedal/tree/develop/Validation/Joystick_Serial_test
3) restart the ESP
4) windows game controller output app to check the change in HID output
5) open a serial monitor 
6) the ESPs output will stall now after a while, if not, please reduce the delay
https://github.com/ChrGri/DIY-Sim-Racing-FFB-Pedal/blob/e2798eca295c65b02d2448eeabd65c69a27d08bd/Validation/Joystick_Serial_test/src/Joystick_Serial_test.ino#L54 

The expected behaviour is that CDC and HID output streams are stable and don't collapse.

### Sketch

```cpp
See link above.
```

### Debug Message

```plain
-
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Jason2866, 2024-09-07:
Your Platformio setup is mixing versions. Bad idea. If you want to use core 3.0.4 with Platformio you should consider using pioarduino. https://github.com/pioarduino/platform-espressif32

ChrGri, 2024-09-07:
Ideally I would want to use core version 2.x

Jason2866, 2024-09-07:
Development is done with actual core. So if anything is not working with core 2.0.x as wanted/expected it will not be changed/fixed in core 2.0.x.

ChrGri, 2024-09-07:
Can you tell me where the versions are mixed?

I've checked the platformio file and it only references espressif32, see
https://github.com/ChrGri/DIY-Sim-Racing-FFB-Pedal/blob/a62915e6e5cbf0ae2353c9a2a32ef5b7ad33ea9d/Validation/Joystick_Serial_test/platformio.ini#L17

Jason2866, 2024-09-07:
https://github.com/ChrGri/DIY-Sim-Racing-FFB-Pedal/blob/develop/ESP32_master/platformio.ini

ChrGri, 2024-09-08:
> https://github.com/ChrGri/DIY-Sim-Racing-FFB-Pedal/blob/develop/ESP32_master/platformio.ini

But that's a completely different project!?
So why is CDC + HID stalling with this?
https://github.com/ChrGri/DIY-Sim-Racing-FFB-Pedal/tree/develop/Validation/Joystick_Serial_test

Jason2866, 2024-09-09:
So you are using core 2.0.x. Not core 3.0.4 as you wrote in the issue template.
For core 2.0.x is no development and bug fixing done anymore.

ChrGri, 2024-09-09:
> So you are using core 2.0.x. Not core 3.0.4 as you wrote in the issue template. For core 2.0.x is no development and bug fixing done anymore.

No. The issue happens on core 3 too. Please try the toy example from the top.

SuGlider, 2024-09-10:
@ChrGri -  Could you please test it using Arduino 3.0.0-RC1 version? This uses the last stable TinyUSB version that I have found.

ChrGri, 2024-09-11:
> @ChrGri - Could you please test it using Arduino 3.0.0-RC1 version? This uses the last stable TinyUSB version that I have found.

@SuGlider **3.1.0-RC1?**
[image]

ChrGri, 2024-09-27:
> @ChrGri - Could you please test it using Arduino 3.0.0-RC1 version? This uses the last stable TinyUSB version that I have found.

Tried it, but it won't even compile
[image]

What am I doing wrong?

TD-er, 2024-09-28:
Did you remove the packages, platforms and cache folder from your .platformio directory?
Just to make sure you're not dealing with older files being present when changing platform definitions.

ChrGri, 2024-09-28:
> Did you remove the packages, platforms and cache folder from your .platformio directory? Just to make sure you're not dealing with older files being present when changing platform definitions.
> 
> Edit: Oh you're building for Arduino 3.0.x Maybe you should try [pioarduino](https://github.com/pioarduino/platform-espressif32) as platformio has no support for the latest Arduino 3.0.x/ESP-IDF5.x

Thank you, I tried with `platform = https://github.com/pioarduino/platform-espressif32/releases/download/51.03.05/platform-espressif32.zip`. The output HID + CDC stall similarly.

Jason2866, 2024-09-28:
This setup
```
platform = https://github.com/pioarduino/platform-espressif32/releases/download/51.03.05/platform-espressif32.zip and
https://github.com/pioarduino/platform-espressif32/releases/download/53.03.10-rc1/platform-espressif32.zip
```
is mixing different source and libs. Result will be unpredictable

ChrGri, 2024-09-28:
@Jason2866 
`platform = https://github.com/pioarduino/platform-espressif32.git#Arduino/IDF53` is set. Its a bit better now, yet still stall after a few seconds, see attached video

https://github.com/user-attachments/assets/4557d660-3b0b-48ea-a4d0-ed29621da584

Jason2866, 2024-12-22:
A small example where the issue happens would be good for bug hunting.
Your project is big and has much code which can generate side effects and / or issues.
There is a low chance that any Arduino developer has the time to do this.

ChrGri, 2024-12-22:
Hello @Jason2866, the "small example" is referenced in the first message, see "The steps to reproduce this issue:"

Jason2866, 2024-12-22:
@ChrGri Thx, overlooked. Is there a possibility to test when not using Windows? Dont use Windows and afaik most developers here don't use too.

ChrGri, 2024-12-22:
@Jason2866 I've tried it on Windows 11 and Mac M1. On both machines the serial output (from CH343) stalls after a few cycles. Adding `Serial.begin(115200)`, seems to increase the time until stall.

ChrGri, 2024-12-23:
@Jason2866 The UART serial output is caused by a missing `Serial.begin()`, see above. Unfortunately, this doesn't help in CDC mode. In CDC the output still stalls after a few seconds.

kbastronomics, 2025-01-16:
I am experiencing this same issue,  not using TinyUSB but using schnoog/Joystick_ESP32S2@^0.9.4 instead.
my build file is very basic using what platformio uses with no external references

```
Processing lolin_s2_mini_release (platform: espressif32; board: lolin_s2_mini; framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/lolin_s2_mini.html
PLATFORM: Espressif 32 (6.9.0) > WEMOS LOLIN S2 Mini
HARDWARE: ESP32S2 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 3.20017.0 (2.0.17) 
 - tool-esptoolpy @ 1.40501.0 (4.5.1) 
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5 
 - toolchain-xtensa-esp32s2 @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 44 compatible libraries
Scanning dependencies...
Dependency Graph
|-- Adafruit TCA8418 @ 1.0.2
|-- PCA95x5 @ 0.1.3
|-- TM16xx LEDs and Buttons @ 0.6.2404
|-- Adafruit GFX Library @ 1.11.11
|-- Adafruit BusIO @ 1.17.0
|-- arduino-timer @ 3.0.1
|-- Adafruit ADS1X15 @ 2.5.0
|-- Joystick_ESP32S2 @ 0.9.4
|-- ESP32Encoder @ 0.11.7
|-- DCS-BIOS @ 0.3.10
Building in release mode
Retrieving maximum program size .pio\build\lolin_s2_mini_release\firmware.elf
Checking size .pio\build\lolin_s2_mini_release\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]   8.8% (used 28896 bytes from 327680 bytes)
Flash: [===       ]  25.0% (used 327790 bytes from 1310720 bytes)
============================================================================= [SUCCESS] Took 5.45 seconds =============================================================================

Environment            Status    Duration
---------------------  --------  ------------
lolin_s2_mini_release  SUCCESS   00:00:05.451
```

my platformio.ini is much simpler

```
[env:lolin_s2_mini_release]
platform = espressif32
board = lolin_s2_mini
framework = arduino
monitor_port = COM8
monitor_speed = 250000
monitor_echo = on
lib_deps = 
	adafruit/Adafruit TCA8418@^1.0.2
	hideakitai/PCA95x5@^0.1.3
	maxint-rd/TM16xx LEDs and Buttons@^0.6.2404
	adafruit/Adafruit GFX Library@^1.11.10
	adafruit/Adafruit BusIO@^1.14.5
	contrem/arduino-timer@^3.0.1
	adafruit/Adafruit ADS1X15@^2.5.0
	schnoog/Joystick_ESP32S2@^0.9.4
	madhephaestus/ESP32Encoder@^0.11.7
build_flags = 
	-DARDUINO_USB_CDC_ON_BOOT=1
```

HID works find on it's own,   CDC Serial works find on it own.
Then I enable and Serial communications at all withing 10 seconds HID always stops functioning.

only way to get it back is to reboot the device.

My Serial device (output from DCS-BIOS) is a large amount of simulator data at 250K baud.
I have no control over that data.

I was going to try tinyUSB thinking it would be a better solution but since this is happening in more than one packing doing HID devices, I would tend to believe the issue is not Joystick_ESP32S2 nor is it tinyUSB either

kbastronomics, 2025-01-16:
One thing I just tried was the not send HID messages with every button press but to do it one time in loop()
while this did actually make my Pots run much faster and smoother,  i  still withing a short time of connecting the CDC port, I loose HID

michal2229, 2025-01-27:
I experience the same problem on ESP32-S2 [with this code](https://github.com/michal2229/g27_emu/blob/f29b3be16839aa2cbe9a91dc9e130aea1449b7e1/arduino/g27_emu/g27_emu.ino#L650). CDC + HID works for a short time, then it stalls. Disabling CDC makes HID work flawlessly.

kbastronomics, 2025-08-09:
The only work around until the Arduino TinyUSB is brought to a current release is to add TinyUSB to your sketch manually and replace the base one.  this works for us.

under platformIO though I can not get it to compile.  it keeps trying to include the base one and the one added to the project and not mater which option I use I can not get around it doing this

kbastronomics, 2025-08-11:
Yes,  I added the Adafruit_TinyUSB_Arduino to my project

created some devices
// Create USB CDC (Serial) and HID instances
Adafruit_USBD_CDC usb_serial;
Adafruit_USBD_HID usb_hid;

in setup added
  TinyUSBDevice.begin();

ruenes1, 2025-08-15:
That worked, thanks! I followed the Adafruit tutorials and was able to create a multiple HID device (keyboard+mouse) with robust CDC communication.

ChrGri, 2025-09-06:
@SuGlider there is a reliability issue with the current HID and CDC communication. Can the solution proposed [here](https://github.com/espressif/arduino-esp32/issues/10307#issuecomment-3169562392) be integrated into the core?

BR

SuGlider, 2025-09-15:
@ChrGri - This is the status summary per Arduino Core Version:
#### Using this [code as example](https://github.com/espressif/arduino-esp32/blob/master/libraries/USB/examples/Gamepad/Gamepad.ino) 
Arduino IDE Settings: 
**USB CDC On Boot: "Enabled"**  --- enables TinyUSB CDC
**USB Mode: "USB-OTG (TinyUSB)"** ---enables TinyUSB HID
**Core Debug Level: "Info"** -- enables debug output using TinyUSB CDC

The sketch starts USB CDC with `Serial.begin(115200)` when **USB CDC On Boot is "Enabled"**
It also starts USB HID Game Controller with `Gamepad.begin(); USB.begin();`
Testing Gamepad with https://hardwaretester.com/gamepad

**Note: The gamepad will only start whenever the Serial Monitor is open using the CDC port, otherwise it will be blocked waiting.**

-----------------------------------
#### Testing Sketch:

``` cpp
#ifndef ARDUINO_USB_MODE
#error This ESP32 SoC has no Native USB interface
#elif ARDUINO_USB_MODE == 1
#warning This sketch should be used when USB is in OTG mode
void setup() {}
void loop() {}
#else
#include "USB.h"
#include "USBHIDGamepad.h"
USBHIDGamepad Gamepad;

const int buttonPin = 0;
int previousButtonState = HIGH;

void setup() {
  // Open Serial Monitor with the CDC port in order to start the sketch
  Serial.begin(115200);
  Serial.setDebugOutput(true);
  while(!Serial) delay(250);
  delay(1000);

  pinMode(buttonPin, INPUT_PULLUP);
  Gamepad.begin();
  USB.begin();

  Serial.println("\n==================\nUSB Gamepad Testing\n==================\n");
  Serial.println("Press BOOT Button to activate the USB gamepad.");
  Serial.println("Longer press will change the affected button and controls.");
  Serial.println("Shorter press/release just activates the button and controls.");
}

void loop() {
  static uint8_t padID = 0;
  static long lastPress = 0;

  int buttonState = digitalRead(buttonPin);
  if (buttonState != previousButtonState) {
    if (buttonState == LOW) {                         // BOOT Button pressed
      Gamepad.pressButton(padID);                     // Buttons 1 to 32
      Gamepad.leftStick(padID << 3, padID << 3);      // X Axis, Y Axis
      Gamepad.rightStick(-(padID << 2), padID << 2);  // Z Axis, Z Rotation
      Gamepad.leftTrigger(padID << 4);                // X Rotation
      Gamepad.rightTrigger(-(padID << 4));            // Y Rotation
      Gamepad.hat((padID & 0x7) + 1);                 // Point of View Hat
      Serial.println();
      log_i("Pressed PadID [%d]", padID);
      lastPress = millis();
    } else {
      Gamepad.releaseButton(padID);
      Gamepad.leftStick(0, 0);
      Gamepad.rightStick(0, 0);
      Gamepad.leftTrigger(0);
      Gamepad.rightTrigger(0);
      Gamepad.hat(HAT_CENTER);
      Serial.println();
      log_i("Released PadID [%d]\n", padID);
      if (millis() - lastPress > 300) {
        padID = (padID + 1) & 0x1F;
        Serial.println();
        log_i("Changed padID to %d\n", padID);
      }
    }
  }
  previousButtonState = buttonState;
  Serial.print(".");
  delay(100);
}
#endif /* ARDUINO_USB_MODE */
```

#### Expected CDC Output:
This shall happen along with Gamepad HID output that can be viewed within https://hardwaretester.com/gamepad
```
==================
USB Gamepad Testing
==================

Press BOOT Button to activate the USB gamepad.
Longer press will change the affected button and controls.
Shorter press/release just activates the button and controls.
..............
[  3164][I][Gamepad.ino:44] loop(): Pressed PadID [0]
..
[  3370][I][Gamepad.ino:54] loop(): Released PadID [0]

............
[  4576][I][Gamepad.ino:44] loop(): Pressed PadID [0]
..
[  4782][I][Gamepad.ino:54] loop(): Released PadID [0]

......
[  5388][I][Gamepad.ino:44] loop(): Pressed PadID [0]
..........
[  6394][I][Gamepad.ino:54] loop(): Released PadID [0]

[  6395][I][Gamepad.ino:58] loop(): Changed padID to 1

.........
[  7301][I][Gamepad.ino:44] loop(): Pressed PadID [1]
.
[  7407][I][Gamepad.ino:54] loop(): Released PadID [1]

...........
[  8513][I][Gamepad.ino:44] loop(): Pressed PadID [1]
...............
[ 10019][I][Gamepad.ino:54] loop(): Released PadID [1]

[ 10020][I][Gamepad.ino:58] loop(): Changed padID to 2

......

```

Checking devices using Windows Device Manager.

### Arduino Core 3.0.7:
After upload, a new new Serial Port (UB CDC) is shown and a new Game Controlled is listed in the Windows Device Manager.
It seems to work fine. When Serial Monitor is open with the CDC COM port, it displays the correct output.
When the BOOT button is pressed, it is possible to see the Gamepad in action with https://hardwaretester.com/gamepad
After testing for a while (couple minutes), I couldn't see any stalling.
Device Name: TinyUSB HID | Vendor: 303a | Product: 1001

### Arduino Core 3.1.3:
Exactly the same as with Core 3.0.7. Seems to work fine.

### Arduino Core 3.2.1:
Exactly the same as with Core 3.0.7. Seems to work fine.
Note: USB Game device is enumerated and visible as soon as the ESP32-S3 is on, but it doesn't receive any event until the Serial Monitor window is open.

### Arduino Core 3.3.0:
Exactly the same as with Core 3.2.1, including enumerating USB HID right after booting. It seems to work fine.

SuGlider, 2025-09-15:
@ChrGri - This issue seems to be related to the fact that there are 2 tasks running when an Arduino Sketch is being executed:
1- The setup()/loop() task that runs the sketch itself.
2- The TinyUSB task that process all USB events.

The problem seems to be related to the fact that the Arduino Sketch Task never stops running until it gets stuck with too may USB output packets from CDC and HID. Adding the `delya(1)` right after writing to the USB CDC solves the issue because it force the RTOS Sheduler to change the execution to the USB dirver that sends/flushes the CDC data to the Host.

At this time, I'd say that the work around to fix it is to Write the CDC and delay for 1 ms in case all the execution is within a single task, like setup/loop task.

I'll analyse it futher to check if there is somthing that can be done to solve it for good.

SuGlider, 2025-09-28:
Debug News: The sketch hangs due to `USBCDC::write()` that waits for ever in a `while()` loop and never triggger WDT (Arduino Default Configuration). Changing the `while()` to exit on certain timeout avoids hanging `loop_task` but it doesn't solve the issue because after first CDC problem, it will delay <timeout_ms>  to execute each subsequence `CDC Write()`, delaying the whole sketch execution.

The root cause is due to TinyUSB loosing some internal synch and never sending USB CDC 64 bytes out.
Investigation continues.

kbastronomics, 2025-09-29:
was there some change made between the version of TinyUSB in the Arduino build and the stand alone version of TinyUSB?
i ask since this problem does not exist int he standaone version of TinyUSB?

SuGlider, 2025-09-29:
After going a bit deeper into USB in general and into HID/CDC:
https://beyondlogic.org/usbnutshell/usb4.shtml
https://en.wikipedia.org/wiki/USB_human_interface_device_class

"Each USB HID interface communicates with the host using either a control pipe or an interrupt pipe. Isochronous and bulk pipes are not used in HID class devices. Both IN and OUT control transfers are required for enumeration; only an IN interrupt transfer is required for HID reports. OUT interrupt transfers are optional in HID-class devices."

This is a very important consideration. There is pooling interval negotiated by the HID device (ESP32-S3) and the USB Host (PC/Windows). This interval is limited to 1 transfer of 64 bytes every millisecond for Full Speed (ESP32-S3 USB 2.0) - limiting the transfer rate up to 64 KB/s. It seems that TinyUSB flushes each HID report alone every 1 ms. 

By other hand USB CDC communicates using bulk pipes. Which means that it will try to send data whenever possible, when there is no Interrupt, Control or Isochronous high priority pending communication.

A bulk pipe can send up to 19 times 64 bytes, every 1 millisecond, limiting the it to about 1.2MB/s (Full Speed).

Looking into the TinyUSB Code, it seems to try best effort, pushing data whenever requested by the Host (Interrupt - HID) or whenever possible (bulk - CDC). It doesn't coordinate clearly Composite USB devices, which makes it hard for the Host to make it work. The Host may just **DROP** the device in case it doesn't receive the data in the way the Host expects it.

It ends up to the USB Device to coordinate the Data transfer in a Composite Device (CDC + HID).
Adding a `delay(2)` right after writing the CDC message, gives the Host time to process and request the data for HID and CDC.
When the Arduino application just pushes as fast as possible both CDC and HID, it seems that some problem with Device <-> Host communication happens.

If the Arduino application only sends HID data to the Host (not in Composite mode), it seems to work correctly.
The same if the Arduino application only sends CDC data to the Host.

The issue occurs when it is running in USB Device Composite mode. It seems to be a possible problem in the Windows USB Driver or in the way how the data is sent to the Host. 

Anyway, it doesn't seem clear how the ESP32 Arduino Layer can help it.

Investigation continues...

TD-er, 2025-09-29:
Hmm I wonder why this is running in the Arduino RTOS task and not as a separate RTOS task.
This way you can have quite strict 1 msec RTOS timings and not relying on `delay()` which is merely guaranteeing a minimal wait time of whatever argument is given.

TD-er, 2025-09-29:
Oh and just another thing which I found out while working on this USB camera back then...
The encoding over USB cables (thus the low level electrical layer) uses NRZI (or NRZ ?), which means it can only send so many 0's before it needs to send a 1 again.
The encoding is that a level change is a 1 and remaining at the same level is a 0.
This does have an effect on the maximum band width you can use, which isn't a constant but depends on the data you're sending.

To illustrate this, consider the USB camera I was developing for a 3D CNC geometrical metrology machine.
The pixel data was already being de-Bayered on the camera using an FPGA. So you can consider the image to be "monochrome" (which is important). The pixel data was sent as raw data. So black is 0x00 and white is 0xFF.
Most of the image was black, so you'll see the problem already... Every 6th bit there was an extra '1' inserted and thus effectively limiting the bandwidth by 1/6th.
Simple trick is to invert the data, as most of the data was black.
However then the problem is when processing larger bright areas.
So the trick I came up with was to XOR the data with 0xA5 (or 0x5A), which statistically made it near impossible to have more than six 0's in a row (due to noise of the sensor)
This way I could achieve near the max. bandwidth.

USB3 (as well as PCI-express and 10G Ethernet) does have a more elegant encoding, where they use a fixed ratio of padded 1's. (Manchester encoding using 4b/5b, 8b/10b or even 64b/66b).

This poor choice for a variable NRZI encoding in early USB standards may also be the cause of USB devices running out of sync or suddenly getting disconnected by the USB host.