espressif/arduino-esp32 #10309: error: 'ledcSetup'

LuizGuilhermeSilvaa, 2024-09-09:
### Board

ESP32 - WROOM 32 DEVKITV1

### Device Description

It's the esp32 with the cp210x driver chip

### Hardware Configuration

It's not connected to anything

### Version

v3.0.4

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

80mhz

### PSRAM enabled

yes

### Upload speed

921600

### Description

I can't use pwm because of a bug in the esp32 library.

"Compilation error: 'ledcSetup' was not declared in this scope"

### Sketch

```cpp
C:\Users\Guilherme\AppData\Local\Temp\.arduinoIDE-unsaved202488-11868-4gnakt.nmbmy\sketch_sep8c\sketch_sep8c.ino: In function 'void setup()':
C:\Users\Guilherme\AppData\Local\Temp\.arduinoIDE-unsaved202488-11868-4gnakt.nmbmy\sketch_sep8c\sketch_sep8c.ino:8:3: error: 'ledcSetup' was not declared in this scope
    8 |   ledcSetup(pwmChannel, freq, resolution);
      |   ^~~~~~~~~
C:\Users\Guilherme\AppData\Local\Temp\.arduinoIDE-unsaved202488-11868-4gnakt.nmbmy\sketch_sep8c\sketch_sep8c.ino:11:3: error: 'ledcAttachPin' was not declared in this scope; did you mean 'ledcAttach'?
   11 |   ledcAttachPin(ledPin, pwmChannel);
      |   ^~~~~~~~~~~~~
      |   ledcAttach

exit status 1

Compilation error: 'ledcSetup' was not declared in this scope
```

### Debug Message

```plain
Compilation error: 'ledcSetup' was not declared in this scope
```

### Other Steps to Reproduce

I was trying to use a "dabble" library, but right after compiling it gives an error.

I came to the conclusion that it was a bug in the library because I uninstalled it and installed another Arduino library, and even though the code didn't pass because it was incompatible, I just compiled it and there was no error.

[image]

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2024-09-09:
the ledc API has changed between v2 and v3.  If you want the old behavior, use v2.0.17
https://docs.espressif.com/projects/arduino-esp32/en/latest/migration_guides/2.x_to_3.0.html#ledc

closed 2024-09-13 by LuizGuilhermeSilvaa

Allankioni, 2025-02-18:
use this link to get the solution 
https://forum.sunfounder.com/t/please-help-compilation-error-ledcsetup-was-not-declared-in-this-scope/1382