espressif/arduino-esp32 #10315: Build Fail when transitioning from v2.0.14 to v3.0.4

henriberisha, 2024-09-10:
### Board

ESP32-WROOM-DA Module

### Device Description

https://www.lilygo.cc/products/t-sim7000g?srsltid=AfmBOoqWVM6LCwcBcCR1J7TSFsP0KJIWFpFk6GePpn4ogoGLV9jPQ-s0

### Hardware Configuration

[image]

### Version

v3.0.4

### IDE Name

Visual Studio Code with Arduino Extension installed

### Operating System

Ubuntu 22.04

### Flash frequency

80MHz

### PSRAM enabled

yes

### Upload speed

921600

### Description

I am not able to build and compile the sketch when using Board version: v3.0.4. however, i am able to build, install and run the sketch when using version: v2.0.14. I have tried other boards versions 3.0.x

I get include errors for libraries such as "esp_wifi.h" 

I would like to build my future projects on the newer board versions, and currently i am not able to do that. 

I am using only VSC with the Arduino extension to verify, build, and flash the sketches into the board. 

I look forward to your answers. Thank you very much
[image]

### Sketch

```cpp
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_wifi.h"
#include "esp_log.h"
#include "nvs_flash.h"
#include "esp_event.h"

#define DEFAULT_SCAN_LIST_SIZE 10

// Start Wi-Fi scan
wifi_scan_config_t scan_config = {
    .ssid = 0,
    .bssid = 0,
    .channel = 0,
    .show_hidden = true,
    .scan_type = WIFI_SCAN_TYPE_ACTIVE, // Active scan
    .scan_time = {
        .active = {
            .min = 100, // Minimum time (ms) to dwell on each channel
            .max = 300  // Maximum time (ms) to dwell on each channel
        }}};

// Simulated setup function
void setup()
{
    // Initialize NVS
    esp_err_t ret = nvs_flash_init();
    if (ret == ESP_ERR_NVS_NO_FREE_PAGES || ret == ESP_ERR_NVS_NEW_VERSION_FOUND)
    {
        ESP_ERROR_CHECK(nvs_flash_erase());
        ret = nvs_flash_init();
    }
    ESP_ERROR_CHECK(ret);

    // Initialize Wi-Fi stack
    ESP_ERROR_CHECK(esp_netif_init());
    ESP_ERROR_CHECK(esp_event_loop_create_default());
    esp_netif_create_default_wifi_sta();

    wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
    ESP_ERROR_CHECK(esp_wifi_init(&cfg));

    // Set Wi-Fi to station mode
    ESP_ERROR_CHECK(esp_wifi_set_mode(WIFI_MODE_STA));
    ESP_ERROR_CHECK(esp_wifi_start());
}

// Simulated loop function
void loop()
{

    // Start scanning (blocking)
    ESP_ERROR_CHECK(esp_wifi_scan_start(&scan_config, true));

    // Get the number of APs found
    uint16_t number = DEFAULT_SCAN_LIST_SIZE;
    wifi_ap_record_t ap_info[DEFAULT_SCAN_LIST_SIZE];
    ESP_ERROR_CHECK(esp_wifi_scan_get_ap_num(&number));
    ESP_ERROR_CHECK(esp_wifi_scan_get_ap_records(&number, ap_info));

    // Print the AP information
    for (int i = 0; i < number; i++)
    {
        printf("SSID: %s, RSSI: %d\n", ap_info[i].ssid, ap_info[i].rssi);
    }

    // Delay to simulate behavior of `loop()` running periodically
    vTaskDelay(pdMS_TO_TICKS(1000)); // Wait for 1 second before rescanning
}

void app_main(void)
{
    // Simulate `setup()`
    setup();

    // Simulate `loop()` with an infinite loop
    while (true)
    {
        loop(); // Call `loop()` repeatedly
    }
}
```

### Debug Message

```plain
Build problems related to including libraries
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2024-09-10:
Not sure what the arduino extension for vscode is, but it is not supported here. We can help you with [platformio](https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html#installing-using-platformio) in vscode.

henriberisha, 2024-09-10:
Thank you, 
I have never used platformio before, but happily to start using it. Would you be able to tell me, what the `platformio.ini` will look like for the above sketch?

lbernstone, 2024-09-10:
This forum is for issues with the code in the repo, not tutorials.
https://github.com/pioarduino/platform-espressif32
https://docs.platformio.org/en/latest/platforms/espressif32.html

closed 2024-09-10 by henriberisha