espressif/arduino-esp32 #10322: AP Configuration Fails after updating from 3.0.1 to 3.0.2 and Later

Bolbolito, 2024-09-10:
### Board

Lolin ESP32 S2 Mini

### Device Description

None

### Hardware Configuration

Not Applicable

### Version

v3.0.4

### IDE Name

Arduino IDE

### Operating System

Windows 11

### Flash frequency

40Mhz

### PSRAM enabled

yes

### Upload speed

115200

### Description

If updated to 3.0.2 the AP cannot be configured it just returns the old or default settings when calling the initializeOTA() function in my code and logs shows that AP cannot be configured. Works fine with 3.0.1.

### Sketch

```cpp
#include <WiFi.h>
#include <esp_now.h>
#include "esp_wifi.h"

oid setup() {
  Serial.begin(115200);

  setCpuFrequencyMhz(80);

  WiFi.mode(WIFI_STA);
  WiFi.disconnect();
  esp_wifi_set_protocol(WIFI_IF_STA, WIFI_PROTOCOL_LR);

  delay(1000);

  initESPNow();
}

void initializeOTA() {
  isUpdating = true;
  esp_now_deinit();
  delay(1000);

  String macAddress = WiFi.macAddress();
  macAddress.replace(":", "");  // Remove colons from MAC address

  // Define the SSID using the full MAC address
  String ssid = macAddress;

  // Define the password using the first 5 characters of the MAC address
  String password = "TEST-" + macAddress.substring(0, 5);

  // Print the SSID and password to the serial monitor
  Serial.print("SSID: ");
  Serial.println(ssid);
  Serial.print("Password: ");
  Serial.println(password);

  esp_wifi_set_protocol(WIFI_IF_STA, WIFI_PROTOCOL_11B | WIFI_PROTOCOL_11G | WIFI_PROTOCOL_11N);

  WiFi.mode(WIFI_AP);
  
  esp_wifi_set_protocol(WIFI_IF_AP, WIFI_PROTOCOL_11B | WIFI_PROTOCOL_11G | WIFI_PROTOCOL_11N);

  // Initialize the WiFi in AP mode with the custom SSID and password
  WiFi.softAP(ssid.c_str(), password.c_str());

  // Configure the static IP address
  if (!WiFi.softAPConfig(local_IP, gateway, subnet)) {
    Serial.println("Failed to configure AP");
  }

  // Print the IP address of the AP
  Serial.print("AP IP address: ");
  Serial.println(WiFi.softAPIP());
}
```

### Debug Message

```plain
16:10:57.763 -> SSID: XXXXXXXXXXXX
16:10:57.763 -> Password: XXXXX-XXXXXX
16:10:58.767 -> Failed to configure AP
16:10:59.756 -> AP IP address: 192.168.4.1
```

### Other Steps to Reproduce

Compiled using my two other pc with the same result when version is 3.0.2 or later.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2024-09-10:
the code above does not compile, please fix it. You are mixing IDF and Arduino calls. It's not rally clear what you are trying to do.

me-no-dev, 2024-09-10:
with the sketch above, AP starts normally on my end. Here is the debug log:
```
[   819][V][NetworkEvents.cpp:119] checkForEvent(): Network Event: 9 - WIFI_READY
[   918][W][STA.cpp:537] disconnect(): STA already disconnected.
[   920][V][STA.cpp:184] _onStaEvent(): STA Started
[   929][V][NetworkEvents.cpp:119] checkForEvent(): Network Event: 11 - STA_START
[   936][V][STA.cpp:110] _onStaArduinoEvent(): Arduino STA Event: 11 - STA_START
[   947][V][STA.cpp:188] _onStaEvent(): STA Stopped
[   959][V][NetworkEvents.cpp:119] checkForEvent(): Network Event: 12 - STA_STOP
[   966][V][STA.cpp:110] _onStaArduinoEvent(): Arduino STA Event: 12 - STA_STOP
[  1011][V][STA.cpp:184] _onStaEvent(): STA Started
[  1016][V][NetworkEvents.cpp:119] checkForEvent(): Network Event: 11 - STA_START
[  1024][V][STA.cpp:110] _onStaArduinoEvent(): Arduino STA Event: 11 - STA_START
SSID: 30AEA467C570
Password: TEST-30AEA
[  4012][V][STA.cpp:188] _onStaEvent(): STA Stopped
[  4019][V][NetworkEvents.cpp:119] checkForEvent(): Network Event: 12 - STA_STOP
[  4020][V][STA.cpp:184] _onStaEvent(): STA Started
[  4031][V][STA.cpp:110] _onStaArduinoEvent(): Arduino STA Event: 12 - STA_STOP
[  4038][V][NetworkEvents.cpp:119] checkForEvent(): Network Event: 11 - STA_START
[  4045][V][STA.cpp:110] _onStaArduinoEvent(): Arduino STA Event: 11 - STA_START
[  4053][V][STA.cpp:110] _onStaArduinoEvent(): Arduino STA Event: 11 - STA_START
[  4060][V][STA.cpp:188] _onStaEvent(): STA Stopped
[  4066][V][NetworkEvents.cpp:119] checkForEvent(): Network Event: 12 - STA_STOP
[  4068][V][AP.cpp:106] _onApEvent(): AP Started
[  4077][V][NetworkEvents.cpp:119] checkForEvent(): Network Event: 19 - AP_START
[  4085][V][AP.cpp:88] _onApArduinoEvent(): Arduino AP Event: 19 - AP_START
[  4109][V][AP.cpp:110] _onApEvent(): AP Stopped
[  4120][V][NetworkEvents.cpp:119] checkForEvent(): Network Event: 20 - AP_STOP
[  4128][V][AP.cpp:88] _onApArduinoEvent(): Arduino AP Event: 20 - AP_STOP
[  4620][W][WiFiGeneric.cpp:639] setTxPower(): Neither AP or STA has been started
[  4622][V][AP.cpp:106] _onApEvent(): AP Started
[  4632][V][NetworkEvents.cpp:119] checkForEvent(): Network Event: 19 - AP_START
[  4639][V][AP.cpp:88] _onApArduinoEvent(): Arduino AP Event: 19 - AP_START
[  4647][V][NetworkInterface.cpp:431] config(): SoftAP: 192.168.2.1 | Gateway: 192.168.2.1 | DHCP Start: 0.0.0.0 | Netmask: 255.255.255.0
[  4660][V][NetworkInterface.cpp:488] config(): DHCP Server Range: 192.168.2.2 to 192.168.2.12
AP IP address: 192.168.2.1
```

Bolbolito, 2024-09-11:
> with the sketch above, AP starts normally on my end. Here is the debug log:
>
> ```
> [   819][V][NetworkEvents.cpp:119] checkForEvent(): Network Event: 9 - WIFI_READY
> [   918][W][STA.cpp:537] [...]

Thank you. I have found the cause of the problem and fixed it by moving the initializeOTA calling to the loop. For some reason on 3.0.2 and later you can't trigger the initializeOTA directly via the onDataRecv function for esp-now anymore.

void onDataRecv(const esp_now_recv_info *info, const uint8_t *incomingData,
                int len) {
  if (len >= 4 && memcmp(incomingData, "otaupdate", 9) == 0) {
    shouldInitializeOTA = true;
  }
}

closed 2024-09-11 by me-no-dev