espressif/arduino-esp32 #10326: Not all BLE characteristics are discovered

kakopappa, 2024-09-11:
### Board

ESP32 S3

### Device Description

ESP32-S3-WROOM-1 dev board

### Hardware Configuration

nothing attached

### Version

v3.0.3

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

240

### PSRAM enabled

no

### Upload speed

921600

### Description

Hello,

I noticed not all BLE characteristics are discovered when there are over 8. 

[image]

### Sketch

```cpp
/*
    Based on Neil Kolban example for IDF: https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/tests/BLE%20Tests/SampleServer.cpp
    Ported to Arduino ESP32 by Evandro Copercini
    updates by chegewara
*/

#include <BLEDevice.h>
#include <BLEUtils.h>
#include <BLEServer.h>

// See the following for generating UUIDs:
// https://www.uuidgenerator.net/

#define SERVICE_UUID        "4fafc201-1fb5-459e-8fcc-c5c9c331914b"

#define CHARACTERISTIC_UUID_1  "beb5483e-36e1-0001-b7f5-ea07361b26a8"
#define CHARACTERISTIC_UUID_2  "beb5483e-36e1-0002-b7f5-ea07361b26a8"
#define CHARACTERISTIC_UUID_3  "beb5483e-36e1-0003-b7f5-ea07361b26a8"
#define CHARACTERISTIC_UUID_4  "beb5483e-36e1-0004-b7f5-ea07361b26a8"
#define CHARACTERISTIC_UUID_5  "beb5483e-36e1-0005-b7f5-ea07361b26a8"
#define CHARACTERISTIC_UUID_6  "beb5483e-36e1-0006-b7f5-ea07361b26a8"
#define CHARACTERISTIC_UUID_7  "beb5483e-36e1-0007-b7f5-ea07361b26a8"
#define CHARACTERISTIC_UUID_8  "beb5483e-36e1-0008-b7f5-ea07361b26a8"
[... 496 lines cut ...]
[  8008][V][BLECharacteristic.cpp:191] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_MTU_EVT
[  8018][V][BLECharacteristic.cpp:444] handleGATTServerEvent(): << handleGATTServerEvent
[  8025][V][BLECharacteristic.cpp:191] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_MTU_EVT
[  8036][V][BLECharacteristic.cpp:444] handleGATTServerEvent(): << handleGATTServerEvent
[  8044][V][BLECharacteristic.cpp:191] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_MTU_EVT
[  8053][V][BLECharacteristic.cpp:444] handleGATTServerEvent(): << handleGATTServerEvent
[  8061][V][BLECharacteristic.cpp:191] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_MTU_EVT
[  8071][V][BLECharacteristic.cpp:444] handleGATTServerEvent(): << handleGATTServerEvent
[  8079][V][BLECharacteristic.cpp:191] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_MTU_EVT
[  8088][V][BLECharacteristic.cpp:444] handleGATTServerEvent(): << handleGATTServerEvent
[  8096][V][BLECharacteristic.cpp:191] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_MTU_EVT
[  8106][V][BLECharacteristic.cpp:444] handleGATTServerEvent(): << handleGATTServerEvent
[  8114][V][BLECharacteristic.cpp:191] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_MTU_EVT
[  8123][V][BLECharacteristic.cpp:444] handleGATTServerEvent(): << handleGATTServerEvent
[  8131][V][BLEServer.cpp:275] handleGATTServerEvent(): << handleGATTServerEvent
```

### Other Steps to Reproduce

1. Flash the sketch
2. Use nRF Connect app to discover BLE devices. Click connect and see the characteristics. Not all are visible.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

kakopappa, 2024-09-12:
Hello @SuGlider Thanks for taking a look.

flashed the same code on ESP32-C3. I am seeing the same issue. 

[image]

I wrote the same code using NimBLE library. nRF Connect apps shows 10 characteristics. so I think it's likely something wrong with the ESP32 BLE library

[image]

```

#include <NimBLEDevice.h>

static NimBLEServer* pServer;

#define CHARACTERISTIC_UUID_1 "beb5483e-36e1-0001-b7f5-ea07361b26a8"
#define CHARACTERISTIC_UUID_2 "beb5483e-36e1-0002-b7f5-ea07361b26a8"
#define CHARACTERISTIC_UUID_3 "beb5483e-36e1-0003-b7f5-ea07361b26a8"
#define CHARACTERISTIC_UUID_4 "beb5483e-36e1-0004-b7f5-ea07361b26a8"
#define CHARACTERISTIC_UUID_5 "beb5483e-36e1-0005-b7f5-ea07361b26a8"
#define CHARACTERISTIC_UUID_6 "beb5483e-36e1-0006-b7f5-ea07361b26a8"
#define CHARACTERISTIC_UUID_7 "beb5483e-36e1-0007-b7f5-ea07361b26a8"
#define CHARACTERISTIC_UUID_8 "beb5483e-36e1-0008-b7f5-ea07361b26a8"
#define CHARACTERISTIC_UUID_9 "beb5483e-36e1-0009-b7f5-ea07361b26a8"
#define CHARACTERISTIC_UUID_10 "beb5483e-36e1-0010-b7f5-ea07361b26a8"

void setup() {
    Serial.begin(115200);
    Serial.println("Starting NimBLE Server");

    /** sets device name */
    NimBLEDevice::init("PROV-123456");
    NimBLEDevice::setPower(ESP_PWR_LVL_P9);
    NimBLEDevice::setSecurityAuth(/*BLE_SM_PAIR_AUTHREQ_BOND | BLE_SM_PAIR_AUTHREQ_MITM |*/ BLE_SM_PAIR_AUTHREQ_SC);

    pServer = NimBLEDevice::createServer();
 
    NimBLEService* pDeadService = pServer->createService("DEAD");
    NimBLECharacteristic* pBeefCharacteristic1 = pDeadService->createCharacteristic(CHARACTERISTIC_UUID_1, NIMBLE_PROPERTY::READ | NIMBLE_PROPERTY::WRITE);
    NimBLECharacteristic* pBeefCharacteristic2 = pDeadService->createCharacteristic(CHARACTERISTIC_UUID_2, NIMBLE_PROPERTY::READ | NIMBLE_PROPERTY::WRITE);
    NimBLECharacteristic* pBeefCharacteristic3 = pDeadService->createCharacteristic(CHARACTERISTIC_UUID_3, NIMBLE_PROPERTY::READ | NIMBLE_PROPERTY::WRITE);
    NimBLECharacteristic* pBeefCharacteristic4 = pDeadService->createCharacteristic(CHARACTERISTIC_UUID_4, NIMBLE_PROPERTY::READ | NIMBLE_PROPERTY::WRITE);
    NimBLECharacteristic* pBeefCharacteristic5 = pDeadService->createCharacteristic(CHARACTERISTIC_UUID_5, NIMBLE_PROPERTY::READ | NIMBLE_PROPERTY::WRITE);
    NimBLECharacteristic* pBeefCharacteristic6 = pDeadService->createCharacteristic(CHARACTERISTIC_UUID_6, NIMBLE_PROPERTY::READ | NIMBLE_PROPERTY::WRITE);
    NimBLECharacteristic* pBeefCharacteristic7 = pDeadService->createCharacteristic(CHARACTERISTIC_UUID_7, NIMBLE_PROPERTY::READ | NIMBLE_PROPERTY::WRITE);
    NimBLECharacteristic* pBeefCharacteristic8 = pDeadService->createCharacteristic(CHARACTERISTIC_UUID_8, NIMBLE_PROPERTY::READ | NIMBLE_PROPERTY::WRITE);
    NimBLECharacteristic* pBeefCharacteristic9 = pDeadService->createCharacteristic(CHARACTERISTIC_UUID_9, NIMBLE_PROPERTY::READ | NIMBLE_PROPERTY::WRITE);
    NimBLECharacteristic* pBeefCharacteristic10 = pDeadService->createCharacteristic(CHARACTERISTIC_UUID_10, NIMBLE_PROPERTY::READ | NIMBLE_PROPERTY::WRITE);

   
    /** Start the services when finished creating all Characteristics and Descriptors */
    pDeadService->start(); 

    NimBLEAdvertising* pAdvertising = NimBLEDevice::getAdvertising();
    pAdvertising->addServiceUUID(pDeadService->getUUID());
    pAdvertising->setScanResponse(true);
    pAdvertising->start();

    Serial.println("Advertising Started");
}

void loop() {
  delay(2000);
}
```

SuGlider, 2024-09-12:
Thanks for the information. I'll take a look into Bluedroid configuration and into ESP32 Arduino BLE Library.

SuGlider, 2025-02-11:
@lucasssvaz - PTAL -- it may worth testing with the new NimBLE library.

lucasssvaz, 2025-08-30:
We recently moved from Bluedroid to NimBLE. Could you please test it with the newest version ?