espressif/arduino-esp32 #10329: SD card can read or write exsist file, but any create (file/dir) crash the system.

developer-ken, 2024-09-11:
### Board

ESP32-PICO-V3-02

### Device Description

Custom made PCB with a ESP32-PICO-V3-02 and a TF card slot. Since both reading and writing exsisting files works fine, I don't think it is a hardware problem.

### IDE Name

PlatformIO

### Operating System

Windows11

### Description

Open an exists file on sd card for writing, reading or appending works just fine. But when I want to create some file/dir, just crashed.

For example, when "/test1.txt" does not exists, it crashes. When it exists, code works fine and content is actually written to the file.

### Sketch

```cpp
DEBUG.print("Build ");   //DEBUG is a Serial
  DEBUG.print(__DATE__);
  DEBUG.print(" ");
  DEBUG.print(__TIME__);
  DEBUG.println();
  DEBUG.println("===================================");

  DEBUG.println("Mounting SD card...");
  SPI.begin(SD_SCLK, SD_MISO, SD_MOSI, SD_CS);
  if (SD.begin(SD_CS, SPI, 24000000))                              //Line 124
  {
    DEBUG.println("SD card mounted:");
    DEBUG.printf("- Type: %d\n", (int)SD.cardType());
    DEBUG.printf("- Size: %lu\n", SD.cardSize() / 1024);
    DEBUG.printf("- Used: %lu\n", SD.usedBytes() / 1024);
  }
  else
  {
    DEBUG.println("SD card mount failed.");
    LED_ERR_ON;
    LED_SAP_ON;
    SETSTATE(LED_BLINK, REC);
    while (1)
      delay(1000);
  }

  // auto testf = SD.open("/test.txt", FILE_APPEND);
  // testf.println("Hey,Hello, World!");
  // testf.close();
  DEBUG.flush();
  SD_WriteFile("/test1.txt", "Hahaha");
  DEBUG.flush();
```

### Debug Message

Crash:
```plain
22:43:27.306 > Build Sep 11 2024 22:43:05
22:43:27.306 > ===================================
22:43:27.306 > Mounting SD card...
22:43:27.306 > Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
22:43:27.306 >
22:43:27.306 > Core  1 register dump:
22:43:27.306 > PC      : 0x4008f963  PS      : 0x00060a33  A0      : 0x8008fec6  A1      : 0x3ffb1fd0
22:43:27.306 > A2      : 0x3f800014  A3      : 0x3f800014  A4      : 0x3f80695c  A5      : 0x00000002
22:43:27.306 > A6      : 0x00060820  A7      : 0x00000001  A8      : 0xee22a700  A9      : 0xfffffffc
22:43:27.306 > A10     : 0x3f806954  A11     : 0x0000693f  A12     : 0x3f800038  A13     : 0x3f800074  
22:43:27.350 > A14     : 0x3f80001c  A15     : 0xaea23da8  SAR     : 0x0000000b  EXCCAUSE: 0x0000001c
22:43:27.350 > EXCVADDR: 0xee22a704  LBEG    : 0x40088000  LEND    : 0x4008800b  LCOUNT  : 0x00000000
22:43:27.350 >
22:43:27.350 >
22:43:27.350 > Backtrace: 0x4008f960:0x3ffb1fd0 0x4008fec3:0x3ffb1ff0 0x40090000:0x3ffb2010 0x40084114:0x3ffb2030 0x400841e3:0x3ffb2060 0x400e6b15:0x3ffb20b0 0x400e7af8:0x3ffb20d0 0x400d3bf7:0x3ffb21b0 0x400d2c46:0x3ffb21f0 0x400d1fe2:0x3ffb2220 0x400d749a:0x3ffb2290
22:43:27.641 >   #0  0x4008f960 in offset_to_block at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/heap_tlsf_block_functions.h:133
22:43:27.641 >       (inlined by) block_next at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/heap_tlsf_block_functions.h:145
22:43:27.641 >       (inlined by) block_link_next at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/heap_tlsf_block_functions.h:153
22:43:27.641 >       (inlined by) block_mark_as_free at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/heap_tlsf_block_functions.h:161
22:43:27.641 >       (inlined by) block_split at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/heap_tlsf.c:293
22:43:27.641 >       (inlined by) block_trim_free at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/heap_tlsf.c:374
22:43:27.641 >       (inlined by) block_prepare_used at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/heap_tlsf.c:454
22:43:27.641 >       (inlined by) tlsf_malloc at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/heap_tlsf.c:851
22:43:27.641 >   #1  0x4008fec3 in multi_heap_malloc_impl at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap.c:187
22:43:27.641 >   #2  0x40090000 in multi_heap_malloc at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap_poisoning.c:234
22:43:27.641 >       (inlined by) multi_heap_malloc at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap_poisoning.c:223
22:43:27.641 >   #3  0x40084114 in heap_caps_malloc_base at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/heap_caps.c:175
22:43:27.641 >       (inlined by) heap_caps_malloc_base at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/heap_caps.c:120
22:43:27.641 >   #4  0x400841e3 in heap_caps_malloc_prefer at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/heap_caps.c:290
22:43:27.641 >   #5  0x400e6b15 in ff_memalloc at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/fatfs/port/freertos/ffsystem.c:20
22:43:27.641 >   #6  0x400e7af8 in esp_vfs_fat_register at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/fatfs/vfs/vfs_fat.c:163
22:43:27.641 >   #7  0x400d3bf7 in sdcard_mount(unsigned char, char const*, unsigned char, bool) at C:/Users/dengb/.platformio/packages/framework-arduinoespressif32/libraries/SD/src/sd_diskio.cpp:795
22:43:27.641 >   #8  0x400d2c46 in fs::SDFS::begin(unsigned char, SPIClass&, unsigned int, char const*, unsigned char, bool) at C:/Users/dengb/.platformio/packages/framework-arduinoespressif32/libraries/SD/src/SD.cpp:38       
22:43:27.641 >   #9  0x400d1fe2 in setup() at src/main.cpp:124
22:43:27.641 >   #10 0x400d749a in loopTask(void*) at C:/Users/dengb/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:42
```

No crash:

```
22:45:59.459 > Build Sep 11 2024 22:45:42
22:45:59.459 > ===================================
22:45:59.459 > Mounting SD card...
22:45:59.579 > SD card mounted:
22:45:59.579 > - Type: 3
22:45:59.579 > - Size: 7563264
22:45:59.579 > - Used: 160
```

```
PACKAGES:
 - framework-arduinoespressif32 @ 3.20016.0 (2.0.16)
 - tool-esptoolpy @ 1.40501.0 (4.5.1)
 - tool-mkfatfs @ 2.0.1
 - tool-mklittlefs @ 1.203.210628 (2.3)
 - tool-mkspiffs @ 2.230.0 (2.30)
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5

Dependency Graph
|-- OneWire @ 2.3.8
|-- IniFile @ 1.3.0
|-- FS @ 2.0.0
|-- SD @ 2.0.0
|-- SD_MMC @ 2.0.0
|-- M1820TempSensor
|-- SPI @ 2.0.0
```

developer-ken, 2024-09-11:
Thanks a lot for your effort helping. 
I've changed the code a little bit and get a more related stack trace.
https://github.com/developer-ken/SCIRecorderFirmware/blob/e5017fb963ef7481c5ad1c354b4d5019cdc0e817/src/main.cpp#L116
```
23:42:10.801 > Backtrace: 0x4008f855:0x3ffb2010 0x4008fec3:0x3ffb2030 0x40090000:0x3ffb2050 0x40084114:0x3ffb2070 0x400841e3:0x3ffb20a0 0x400e6899:0x3ffb20f0 0x400e6fe4:0x3ffb2110 0x400e07b6:0x3ffb2140 0x400ee4ce:0x3ffb2160 0x400d25c2:0x3ffb21f0 0x400d1e1d:0x3ffb2220 0x400d721e:0x3ffb2290
23:42:10.801 >   #0  0x4008f855 in block_locate_free at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/heap_tlsf.c:439
23:42:10.801 >       (inlined by) tlsf_malloc at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/heap_tlsf.c:850
23:42:10.801 >   #1  0x4008fec3 in multi_heap_malloc_impl at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap.c:187
23:42:10.801 >   #2  0x40090000 in multi_heap_malloc at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap_poisoning.c:234
23:42:10.801 >       (inlined by) multi_heap_malloc at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap_poisoning.c:223
23:42:10.801 >   #3  0x40084114 in heap_caps_malloc_base at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/heap_caps.c:175
23:42:10.801 >       (inlined by) heap_caps_malloc_base at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/heap_caps.c:120
23:42:10.801 >   #4  0x400841e3 in heap_caps_malloc_prefer at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/heap_caps.c:290
23:42:10.801 >   #5  0x400e6899 in ff_memalloc at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/fatfs/port/freertos/ffsystem.c:20
23:42:10.801 >   #6  0x400e6fe4 in vfs_fat_opendir at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/fatfs/vfs/vfs_fat.c:733
23:42:10.801 >   #7  0x400e07b6 in esp_vfs_opendir at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/vfs/vfs.c:659 (discriminator 3)
23:42:10.801 >   #8  0x400ee4ce in VFSImpl::open(char const*, char const*, bool) at C:/Users/dengb/.platformio/packages/framework-arduinoespressif32/libraries/FS/src/vfs_api.cpp:54
23:42:10.801 >   #9  0x400d25c2 in fs::FS::open(char const*, char const*, bool) at C:/Users/dengb/.platformio/packages/framework-arduinoespressif32/libraries/FS/src/FS.cpp:234
23:42:10.801 >   #10 0x400d1e1d in setup() at src/main.cpp:116
23:42:10.801 >   #11 0x400d721e in loopTask(void*) at C:/Users/dengb/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:42
```

If those information is not useful, I may write a minimal schetch to reproduce the problem at weekend.

developer-ken, 2024-09-12:
Thanks for your info. Disabling psram works.
With psram **enabled** , I add a line to print `ESP.getMaxAllocHeap()` :
```
18:14:36.748 > Build Sep 12 2024 18:14:08
18:14:36.748 > ===================================
18:14:36.748 > Mounting SD card...
18:14:36.762 > SD card mounted:
18:14:36.762 > - Type: 3
18:14:36.762 > - Size: 7761920
18:14:36.762 > - Used: 96
18:14:36.762 > Heap: 114676
18:14:36.809 > Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
...Same crash stack...
```
I disabled psram by
```
build_unflags =
    -DBOARD_HAS_PSRAM
```
and it works just fine.
```
18:20:42.824 > ===================================
18:20:42.824 > Mounting SD card...
18:20:43.057 > SD card mounted:
18:20:43.057 > - Type: 3
18:20:43.057 > - Size: 7761920
18:20:43.057 > - Used: 96
18:20:43.057 > Heap: 110580
```

What can be the problem here?

lbernstone, 2024-09-12:
Something with how the psram is implemented in pico.  @P-R-O-C-H-Y @SuGlider  Do you have a pico-v3-02 device available to test this?

me-no-dev, 2024-09-12:
@developer-ken which pins are you using?

lbernstone, 2024-09-12:
> @developer-ken which pins are you using?

Not seeing 16/17

```
#define LED_ERR    13
#define LED_SAP    14
#define LED_REC    27

#define SD_CS      8
#define SD_MISO    9
#define SD_MOSI    7
#define SD_SCLK    10

#define RS485_RX   19
#define RS485_TX   22
#define RS485_TNOW 21

#define IRDA_TX    33
#define IRDA_RX    25
#define IRDA_nTNOW 26

#define TEMPSENSOR 4
```

P-R-O-C-H-Y, 2024-09-13:
I have got only pico-v3-zero here without PSRAM unfortunately.

developer-ken, 2024-09-13:
I double checked and find out I used the pin connected internally to psram for SD card SPI. 
I've already made a PCB with the wrong pin config and have to use the chip without psram.
Maybe change the BOM to use pico-v3-zero, as I have to disable psram anyway.

Its my fault not checking pin configuration carefully before odering a PCB. It's confusing me that ESP32-PICO chips have those internally connected pins wired out.

Anyway, thanks for you guys helping solving this problem. I will close this issue.

closed 2024-09-13 by developer-ken