espressif/arduino-esp32 #10333: ESP32-C6 failed on Arduino IDE initial SD card, but it works on ESP-IDF

Sail-211010, 2024-09-12:
### Board

ESP32-C6

### Device Description

My own board

### Hardware Configuration

NO

### Version

v3.0.2

### IDE Name

ArduinoIDE

### Operating System

Windows

### Flash frequency

80

### PSRAM enabled

no

### Upload speed

115200

### Description

I conducted SD card test on my board. Under IDF, when I enabled.format_if_mount_failed = true, all three SD cards in my hand (16GB/32GB/32GB) could be recognized normally, but when I implemented it on Arduino, Only one SD card (16GB) can be recognized, and the other two cannot be recognized,

Here is my program

### Sketch

```cpp
#include "Arduino.h"
#include <SPI.h>
#include "SD.h"
#include "FS.h"

#define EXAMPLE_PIN_NUM_MISO 5
#define EXAMPLE_PIN_NUM_MOSI 6
#define EXAMPLE_PIN_NUM_SCLK 7
#define SD_CS 4
SPI.begin(EXAMPLE_PIN_NUM_SCLK,EXAMPLE_PIN_NUM_MISO,EXAMPLE_PIN_NUM_MOSI);
if (SD.begin(SD_CS)) {
  printf("SD card initialization successful!\r\n");
} else {
  printf("SD card initialization failed!\r\n");
}

uint8_t cardType = SD.cardType();
if(cardType == CARD_NONE){
    printf("No SD card attached\r\n");
    return;
}
```

### Debug Message

```plain
SD card initialization failed!
No SD card attached
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

P-R-O-C-H-Y, 2024-09-17:
@Sail-211010 Can you try it with setting `format_if_empty` parameter to true?
`SD.begin(SD_CS, SPI, 4000000, "/sd", 5, true);`

Sail-211010, 2024-09-23:
@P-R-O-C-H-Y It doesn't work. You tested it

Jason2866, 2025-03-12:
Closing since issue is not reproducable with actual core

closed 2025-03-12 by Jason2866