espressif/arduino-esp32 #10340: UART error on console: `uart: tout_thresh = 3 > maximum value = 1`

hitecSmartHome, 2024-09-13:
### Board

ESP32-Wrover

### Device Description

using Serial1

### Hardware Configuration

using Serial1

### Version

latest master (checkout manually)

### IDE Name

PlatformIO

### Operating System

Windows10

### Flash frequency

80

### PSRAM enabled

yes

### Upload speed

115200

### Description

I have upgraded to Arduino V3 and IDF 5.3.0.
Started to get this error message on my console `uart: tout_thresh = 3 > maximum value = 1`

### Sketch

```cpp
// MBUS_RX_TIMEOUT / (MBUS_BAUD / 10) = millisec
// 3 / (115200 / 10) = 2.6 ms
// 4 / (115200 / 10) = 3.4 ms
// 5 / (115200 / 10) = 4.3 ms
// 6 / (115200 / 10) = 5.2 ms
#define MBUS_RX_TIMEOUT 3
#define MBUS_BAUD 115200
#define MBUS_RX 35
#define MBUS_TX 32
#define MBUS_RTS 33
#define PACKET_TRIGGER_ONLY_ON_TIMEOUT true

Serial1.begin(MBUS_BAUD, SERIAL_8N1, MBUS_RX, MBUS_TX);
Serial1.setPins(-1, -1, -1, MBUS_RTS);
Serial1.setMode(UART_MODE_RS485_HALF_DUPLEX);
Serial1.setRxTimeout(MBUS_RX_TIMEOUT);

Serial1.onReceive(
    std::bind(&Modbus::handlePacket, this),
    PACKET_TRIGGER_ONLY_ON_TIMEOUT
);
Serial1.onReceiveError(
    std::bind(&Modbus::handleReceiveError, this, std::placeholders::_1)
);
```

### Debug Message

```plain
I (4304) uart: ESP_INTR_FLAG_IRAM flag not set while CONFIG_UART_ISR_IN_IRAM is enabled, flag updated

I (4313) uart: queue free spaces: 20

E (4317) uart: tout_thresh = 3 > maximum value = 1
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Jason2866, 2024-09-13:
Branch master is **not** the recommended branch when using IDF 5.3.x
Use branch https://github.com/espressif/arduino-esp32/tree/release/v3.1.x

hitecSmartHome, 2024-09-18:
Thank you. I ended up decreasing the symbol_timeout to 1 since this value is 1 byte time. In our case this is the best option. Thank you very much!

closed 2024-09-18 by hitecSmartHome