espressif/arduino-esp32 #10341: FS crash on read and or write.

hitecSmartHome, 2024-09-13:
### Board

ESP32-Wrover

### Device Description

not related

### Hardware Configuration

not related

### Version

latest master (checkout manually)

### IDE Name

PlatformIO

### Operating System

Windows10

### Flash frequency

80

### PSRAM enabled

yes

### Upload speed

115200

### Description

We are experiencing occasional crashes in FS.h.
It seems to us it is likely a dangling pointer or double free.

### dependencies.lock file

```
esp_littlefs:
    component_hash: 50909b249137952e7f0891b3b6f6d9a0275e1830423b9c06a9ea2b32d0357e43
    dependencies:
    - name: idf
      version: '>=4.3'
    source:
      git: https://github.com/joltwallet/esp_littlefs.git
      path: .
      type: git
    version: 3e5e7a11b7f06515a1f93873b6fe5a9efe88338b
```

### Sketch

```cpp
int FileSystem::read(const char* path, JsonDocument& doc) {
    if (xSemaphoreTake(fileMutex, FILE_SEM_WAIT) == pdTRUE) {
        File file = userFS.open(path);
        if (!file) {
            xSemaphoreGive(fileMutex);
            return -1;
        }
        DeserializationError error = deserializeJson(doc, file);
        file.close();
        if (error) {
            xSemaphoreGive(fileMutex);
[... 396 lines cut ...]
    Locker locked(graphLock);
    if( !locked ){ return false; }

    JsonDocument graphs;

    int size = db.read(GRAPHS_PATH,graphs); // Crash is coming from here. See the function above.
    if( size <= 0 ){ return false; }
    if( !graphs.containsKey(type) ){ return false; }
    
    JsonArray graphsArray = graphs[type].as<JsonArray>();
    for(JsonObject graph : graphsArray ){
        if( graph["id"].as<std::string>() == id ){
            JsonArray savedData = graph["data"].as<JsonArray>();
            return graphData.set(savedData);
        }
    }
    return false;
}
```

Again. All my File access is protected with mutexes and it can happen both with read and write operations. The crash is always comes from the same location.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2024-09-13:
Please edit the info to show that you are using v2.0.17

hitecSmartHome, 2024-09-13:
> Please edit the info to show that you are using v2.0.17

Done, sorry for that.

> What exactly are you mutexing? 

The file open call desperately

> It looks like this may be an attempt to read from the disk while the cache is disabled. Do you have code that may be setting critical mode while this is running in parallel?

I have no critical section at all.
I do not disable any caching.
An other task heavily using UART1 hovewer.
I have set malloc to allocate from psram firstly and I try to do everything in PSram because the internal ram is insufficient.

> Just for testing, can you try reading the file completely into psram and then deserializing from there?

The problem is that this crash is really random and can occour at any other file open call.
It's not just this `get(const char* type, std::string id, JsonArray& graphData)` method.
It happened on other regular read and write calls. This coredump was the latest I could save.
I can load the file into psram in this method but it isn't garanteed that this json method would cause it again.

lbernstone, 2024-09-13:
Well, something is disappearing/unavailable when littlefs tries to stat that file.  Since you know what you have modified from a more standard config, that seems like a good place to start.  What CONFIG settings have you changed? Have you set an [internal reserve](https://docs.espressif.com/projects/esp-idf/en/release-v5.1/esp32/api-reference/kconfig.html?highlight=config_spiram#config-spiram-malloc-reserve-internal)?

lbernstone, 2024-09-13:
Also, how many file handles have you set when you init littlefs?

hitecSmartHome, 2024-09-13:
These settings are on default but will check specifically. I tried to tweak only the psram related stuffs in menuconfig. I can paste the whole sdkconfig in here if it helps

hitecSmartHome, 2024-09-14:
Oh gos. Is it possible that Iam exceeded the max open files and it crashes instead of fail? If 10 is the max open files it is very much possible that more than that amount opened the same file at the same time

hitecSmartHome, 2024-09-14:
```cpp
bool LittleFSFS::begin(bool formatOnFail, const char *basePath, uint8_t maxOpenFiles, const char *partitionLabel) {

  if (partitionLabel_) {
    free(partitionLabel_);
    partitionLabel_ = NULL;
  }

  if (partitionLabel) {
    partitionLabel_ = strdup(partitionLabel);
  }

  if (esp_littlefs_mounted(partitionLabel_)) {
    log_w("LittleFS Already Mounted!");
    return true;
  }

  esp_vfs_littlefs_conf_t conf = {
    .base_path = basePath,
    .partition_label = partitionLabel_,
    .partition = NULL,
    .format_if_mount_failed = false,
    .read_only = false,
    .dont_mount = false,
    .grow_on_mount = true
  };

  esp_err_t err = esp_vfs_littlefs_register(&conf);
  if (err == ESP_FAIL && formatOnFail) {
    if (format()) {
      err = esp_vfs_littlefs_register(&conf);
    }
  }
  if (err != ESP_OK) {
    log_e("Mounting LittleFS failed! Error: %d", err);
    return false;
  }
  _impl->mountpoint(basePath);
  return true;
}
```

`maxOpenFiles` doesn't even used

Jason2866, 2025-03-12:
Please open an issue in IDF github if the issue still exists

closed 2025-03-12 by Jason2866